code-iai/iai_kinect2 #100: Question about IR projector and resulting point cloud

Tabjones, 2015-05-28:
Hi, this is more a question than an issue, but i think it was worth it to share with you.

I found out that sometimes covering the IR projector with a 'semi-transparent' tape could improve the resulting point cloud quality. Take a look a this picture i made:
[image]

Left pic has a tape covering the IR projector, right hasn't. As you can see border of objects are much more defined in the left image, especially around the pot handle and the pattern right border.
However, covering the projector, sensibly reduces IR range: left image has no floor and almost no objects on the background.

I was wondering why my uncovered point cloud has such high level of noise around the borders. and yours hasn't!

  * Do you guys have covered the projector?
  * Is it because i have made a poor calibration ?
  * Is it becasue i have some strange/bad illumination in the room ?

any ideas ?
Thank you

jeppewalther, 2015-05-28:
Try to have a look at the corresponding IR image. My guess is that the sensor is blinding itself. In my setup I have completely covered one of the three IR lamps. This makes it possibly to move the sensor closer to objects. Before blinding itself.

Tabjones, 2015-05-28:
Here's the the corresponding depth colored images (from registration viewer)
[image]

I'm not sure what you mean by `blinding itself` !

wiedemeyer, 2015-05-28:
If something is shiny/reflective, the light from the ir emitter can blind the ir sensor, because the reflected light is too strong.
To see this you should look at the ir image. `rosrun image_view image_view image:=/kinect2/ir/image`

Tabjones, 2015-05-28:
Ok, now i understood, here's the ir images:
[image]

Covered one is just darker, but i don't see any artifacts in any of them, apart from that little shiny object in the top left corner, which i've no idea what it might be.

jeppewalther, 2015-06-03:
Interesting i don't see any burnout issues. Edges on the  black device to the left seems more defined in image with a covered lens though. 

Out of curiosity.
what kind of tape are you using?. 
How close to objects can you get with it before you get burnout issues?.

Tabjones, 2015-06-03:
This tape:
[image]

As you can see it is "semi transparent", we call it "paper-tape", but i have no idea how it might be called in other countries!

Anyway i also though of removing the external casing, like this:
[image]

And see how the projector behaves.
I'll keep you posted if you are interested

Tabjones, 2015-06-04:
I have removed the casing, but images and point clouds are much more noisier and burned out, so I put it back on! (Guess it was there for a reason!!!)

Anyway with the paper tape i can slightly come closer to objects before burning out, something like 35-40 cm, downside is that i can't see anything past 2 meters or maybe less.

jeppewalther, 2015-06-06:
Thanks .. in danish it is malertape (Painters tape). I will give it at try .. So far my best results have been from a total blind off of one of the three lamps. 
[image]

Tabjones, 2015-06-06:
Oh cool! Thanks for sharing, i'll give it a try too

closed 2015-10-26 by wiedemeyer