code-iai/iai_kinect2 #101: The Xf86VidMode library and headers were not found

SamTew, 2015-05-28:
when i am running ./install_ubuntu.sh, it show this error.
CMake Error at CMakeLists.txt:228 (message):
  The Xf86VidMode library and headers were not found

I am using ubuntu 12.04 (precise) and ros hydro.
Hope anyone can help me. Thanks.

wiedemeyer, 2015-05-29:
Hi, did you installed the dependencies for libfreenect2 as mentioned in the README?

SamTew, 2015-05-29:
ROS Hydro/Indigo
OpenCV
PCL
Eigen (optional)
OpenCL (optional)
libfreenect2

do u mean this one? I am confirm i have installed ROS Hydro and OpenCV. For PCL, is it come by default? If not, can u show me some guideline? Sorry as i am newbie.

wiedemeyer, 2015-05-29:
No, I meant the dependencies mentioned in the READMEs from iai_kinect2 and libfreenect2.
Those are for 14.04 but might also work for 12.04:
`sudo apt-get install -y build-essential libturbojpeg libtool autoconf libudev-dev cmake mesa-common-dev freeglut3-dev libxrandr-dev doxygen libxi-dev libopencv-dev automake`

Anyway, this is a problem related to libfreenect2, so it would be best, if you look on there GitHub for similar issues and maybe open a new issue there.

SamTew, 2015-05-29:
Yup. I did install. Alright. I try to ask there. Thanks a lot.

closed 2015-06-03 by wiedemeyer