code-iai/iai_kinect2 #102: OpenCV 3.0 Issues

lakehanne, 2015-05-28:
This is a follow-up on issue [95] (https://github.com/code-iai/iai_kinect2/issues/95) which I raised couple of days ago. 
I have opencv 3.00 on my Ubuntu trusty and I sourced indigo which I am using as well as `rosdep init` and `rosdep update`. But I find that some member functions could not be rightly conbverted as the following shows: 

>lex@LEX~/catkin_ws$ catkin_make -DCMAKE_BUILD_TYPE="Release"
Base path: /home/lex/catkin_ws
Source space: /home/lex/catkin_ws/src
Build space: /home/lex/catkin_ws/build
Devel space: /home/lex/catkin_ws/devel
Install space: /home/lex/catkin_ws/install
####
#### Running command: "cmake /home/lex/catkin_ws/src -DCMAKE_BUILD_TYPE=Release -DCATKIN_DEVEL_PREFIX=/home/lex/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/lex/catkin_ws/install -G Unix Makefiles" in "/home/lex/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/lex/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo
-- This workspace overlays: /opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/lex/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.14
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 5 packages in topological order:
-- ~~  - iai_kinect2 (metapackage)
-- ~~  - depth_registration
-- ~~  - kinect2_bridge
-- ~~  - kinect2_calibration
-- ~~  - registration_viewer
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'iai_kinect2'
-- ==> add_subdirectory(iai_kinect2/iai_kinect2)
-- +++ processing catkin package: 'depth_registration'
-- ==> add_subdirectory(iai_kinect2/depth_registration)
-- Eigen found (include: /usr/include/eigen3)
-- Could NOT find OpenCL (missing:  OPENCL_INCLUDE_DIRS) 
-- CPU based depth registration enabled
-- OpenCL based depth registration disabled
-- +++ processing catkin package: 'kinect2_bridge'
-- ==> add_subdirectory(iai_kinect2/kinect2_bridge)
-- Using these message generators: gencpp;genlisp;genpy
-- +++ processing catkin package: 'kinect2_calibration'
-- ==> add_subdirectory(iai_kinect2/kinect2_calibration)
-- Using these message generators: gencpp;genlisp;genpy
-- +++ processing catkin package: 'registration_viewer'
-- ==> add_subdirectory(iai_kinect2/registration_viewer)
-- Using these message generators: gencpp;genlisp;genpy
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- looking for PCL_COMMON
-- looking for PCL_KDTREE
-- looking for PCL_OCTREE
-- looking for PCL_SEARCH
-- looking for PCL_SURFACE
-- looking for PCL_GEOMETRY
-- looking for PCL_SAMPLE_CONSENSUS
-- looking for PCL_FILTERS
-- looking for PCL_FEATURES
-- looking for PCL_SEGMENTATION
-- looking for PCL_IO
-- looking for PCL_REGISTRATION
-- looking for PCL_KEYPOINTS
-- looking for PCL_RECOGNITION
-- looking for PCL_VISUALIZATION
-- looking for PCL_PEOPLE
-- looking for PCL_OUTOFCORE
-- looking for PCL_TRACKING
-- looking for PCL_APPS
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lex/catkin_ws/build
####
#### Running command: "make -j8 -l8" in "/home/lex/catkin_ws/build"
####
[ 16%] [ 33%] [ 50%] Building CXX object iai_kinect2/kinect2_calibration/CMakeFiles/kinect2_calibration.dir/src/kinect2_calibration.cpp.o
Building CXX object iai_kinect2/depth_registration/CMakeFiles/depth_registration.dir/src/depth_registration.cpp.o
Building CXX object iai_kinect2/depth_registration/CMakeFiles/depth_registration.dir/src/depth_registration_cpu.cpp.o
[ 66%] Building CXX object iai_kinect2/registration_viewer/CMakeFiles/viewer.dir/src/viewer.cpp.o
Linking CXX shared library /home/lex/catkin_ws/devel/lib/libdepth_registration.so
[ 66%] Built target depth_registration
[ 83%] [100%] Building CXX object iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o
Building CXX object iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/src/kinect2_bridge.cpp.o
/home/lex/catkin_ws/src/iai_kinect2/kinect2_calibration/src/kinect2_calibration.cpp: In member function ‘void CameraCalibration::calibrateExtrinsics()’:
/home/lex/catkin_ws/src/iai_kinect2/kinect2_calibration/src/kinect2_calibration.cpp:682:56: error: cannot convert ‘const cv::TermCriteria’ to ‘int’ for argument ‘13’ to ‘double cv::stereoCalibrate(cv::InputArrayOfArrays, cv::InputArrayOfArrays, cv::InputArrayOfArrays, cv::InputOutputArray, cv::InputOutputArray, cv::InputOutputArray, cv::InputOutputArray, cv::Size, cv::OutputArray, cv::OutputArray, cv::OutputArray, cv::OutputArray, int, cv::TermCriteria)’
                                 cv::CALIB_FIX_INTRINSIC);
                                                        ^
/home/lex/catkin_ws/src/iai_kinect2/kinect2_calibration/src/kinect2_calibration.cpp: In member function ‘void DepthCalibration::getPlane(size_t, cv::Mat&, double&) const’:
/home/lex/catkin_ws/src/iai_kinect2/kinect2_calibration/src/kinect2_calibration.cpp:969:138: error: ‘ITERATIVE’ is not a member of ‘cv’
     cv::solvePnPRansac(board, points[index], cameraMatrix, distortion, rvec, translation, false, 300, 0.05, board.size(), cv::noArray(), cv::ITERATIVE);
                                                                                                                                          ^
make[2]: *** [iai_kinect2/kinect2_calibration/CMakeFiles/kinect2_calibration.dir/src/kinect2_calibration.cpp.o] Error 1
make[1]: *** [iai_kinect2/kinect2_calibration/CMakeFiles/kinect2_calibration.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:52:0,
                 from /usr/include/pcl-1.7/pcl/visualization/cloud_viewer.h:39,
                 from /home/lex/catkin_ws/src/iai_kinect2/registration_viewer/src/viewer.cpp:32:
/usr/include/pcl-1.7/pcl/visualization/area_picking_event.h: In member function ‘bool pcl::visualization::AreaPickingEvent::getPointsIndices(std::vector<int>&) const’:
/usr/include/pcl-1.7/pcl/visualization/area_picking_event.h:64:30: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
           if (nb_points_ == -1)
                              ^
Linking CXX shared library /home/lex/catkin_ws/devel/lib/libkinect2_bridge_nodelet.so
[100%] Built target kinect2_bridge_nodelet
Linking CXX executable /home/lex/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge
/usr/bin/ld: warning: libopencv_imgproc.so.3.0, needed by /home/lex/catkin_ws/devel/lib/libdepth_registration.so, may conflict with libopencv_imgproc.so.2.4
[100%] Built target kinect2_bridge
Linking CXX executable /home/lex/catkin_ws/devel/lib/registration_viewer/viewer
/usr/bin/ld: warning: libopencv_objdetect.so.2.4, needed by /usr/lib/x86_64-linux-gnu/libopencv_contrib.so.2.4.8, may conflict with libopencv_objdetect.so.3.0
[100%] Built target viewer
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

lakehanne, 2015-05-29:
Okay, so I installed OpenCV 2.4.9 on my Ubuntu trusty and everything now seems honky dory.
Here is the output of my catkin_make:

>[ 66%] Built target depth_registration
Scanning dependencies of target kinect2_bridge_nodelet
Scanning dependencies of target kinect2_bridge
[ 83%] [100%] Building CXX object iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o
Building CXX object iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/src/kinect2_bridge.cpp.o
In file included from /usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:52:0,
                 from /usr/include/pcl-1.7/pcl/visualization/cloud_viewer.h:39,
                 from /home/lex/catkin_ws/src/iai_kinect2/registration_viewer/src/viewer.cpp:32:
/usr/include/pcl-1.7/pcl/visualization/area_picking_event.h: In member function ‘bool pcl::visualization::AreaPickingEvent::getPointsIndices(std::vector<int>&) const’:
/usr/include/pcl-1.7/pcl/visualization/area_picking_event.h:64:30: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
           if (nb_points_ == -1)
                              ^
Linking CXX shared library /home/lex/catkin_ws/devel/lib/libkinect2_bridge_nodelet.so
[100%] Built target kinect2_bridge_nodelet
Linking CXX executable /home/lex/catkin_ws/devel/lib/kinect2_calibration/kinect2_calibration
[100%] Built target kinect2_calibration
Linking CXX executable /home/lex/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge
/usr/bin/ld: warning: libopencv_core.so.3.0, needed by /usr/local/lib/libfreenect2.so, may conflict with libopencv_core.so.2.4
[100%] Built target kinect2_bridge
Linking CXX executable /home/lex/catkin_ws/devel/lib/registration_viewer/viewer
[100%] Built target viewer

But I still need to square off the exception at 

> In file included from /usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:52:0,
                 from /usr/include/pcl-1.7/pcl/visualization/cloud_viewer.h:39,
                 from /home/lex/catkin_ws/src/iai_kinect2/registration_viewer/src/viewer.cpp:32:
/usr/include/pcl-1.7/pcl/visualization/area_picking_event.h: In member function ‘bool pcl::visualization::AreaPickingEvent::getPointsIndices(std::vector<int>&) const’:
/usr/include/pcl-1.7/pcl/visualization/area_picking_event.h:64:30: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
           if (nb_points_ == -1)
                              ^

Thanks for your help, Thiemo!
'Preciate you, man!

wiedemeyer, 2015-05-29:
OpenCV 3.0 changed some APIs, so you can not simple switch to it.
The other thing is just a warning. Just ignore it.

lakehanne, 2015-05-29:
Yeah, I figured. 
So I sourced `source ~/catkin_ws/devel/setup.bash`, in `~/.bashrc` , launched `roscore`, and executed ` rosrun kinect2_bridge kinect2_bridge` but I get the following:

>$ rosrun kinect2_bridge kinect2_bridge 
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/lex/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[Freenect2Impl] enumerating devices...
[Freenect2Impl] 12 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:3 with serial 501185143042
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 501185143042 (selected)
[Freenect2DeviceImpl] opening...
[UsbControl::claimInterfaces(ControlAndRgbInterfaceId)] failed! libusb error -6: LIBUSB_ERROR_BUSY
[Freenect2DeviceImpl] closing...
[Freenect2DeviceImpl] deallocating usb transfer pools...
[Freenect2DeviceImpl] closing usb device...
[Freenect2DeviceImpl] closed
[Freenect2DeviceImpl] failed to open Kinect v2 @4:3!
no device connected or failure opening the default one!
using sensor defaults for color intrinsic parameters.
using sensor defaults for ir intrinsic parameters.
using defaults for rotation and translation.
using defaults for depth shift.
Segmentation fault (core dumped)
lex@Lex:~$ rosrun kinect2_bridge kinect2_bridge 
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/lex/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[Freenect2Impl] enumerating devices...
[Freenect2Impl] 12 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:3 with serial 501185143042
[Freenect2Impl] found 1 devices
[... 129 lines cut ...]
[1081.3720703125, 0, 959.5;
  0, 1081.3720703125, 539.5;
  0, 0, 1]
distortion coefficients color:
[0, 0, 0, 0, 0]
camera matrix ir:
[364.5779113769531, 0, 259.4354858398438;
  0, 364.5779113769531, 209.3105010986328;
  0, 0, 1]
distortion coefficients ir:
[0.09317874908447266, -0.2718816995620728, 0, 0, 0.0968393012881279]
rotation:
[1, 0, 0;
  0, 1, 0;
  0, 0, 1]
translation:
[-0.052;
  0;
  0]
depth shift:
0

[DepthRegistration::New] Using CPU registration method!
[DepthRegistration::New] Using CPU registration method!
[kinect2_bridge] waiting for clients to connect

wiedemeyer, 2015-05-29:
Thats an USB issue. Are you sure, that you connected the Kinect2 to a USB 3.0 port? It will not work with USB 2.0. Under linux the Kinect2 will also only work with some USB 3.0 controllers. The ones from Intel and NEC/Renesas work fine, others might not.

lakehanne, 2015-05-29:
But Protonect works and it is on USB 3.0 Port... I saw a similar issue 83 on libfreenect2 issues page but it seems disjointed slightly from my problem

wiedemeyer, 2015-05-29:
Ok, then check to with libusb libfreenect2.so links. `ldd kinect2_bridge | grep libfreenect2` and the `ldd` that lib. It should link to the libusb in `libfreenect2/depends`.
If not, you might missed to add the modifications to the CMakeLists.txt mentioned in the README. If so, add them, make and make install libfreenect2 again. Then test it again.

lakehanne, 2015-05-29:
$ ldd kinect2_bridge | grep libfreenect2
ldd: ./kinect2_bridge: No such file or directory

wiedemeyer, 2015-05-29:
you have to give the path to the executable or go to the directory. `catkin_ws/devel/lib/kinect2_bridge`

lakehanne, 2015-05-29:
#:~/catkin_ws/devel/lib/kinect2_bridge$ ldd kinect2_bridge | grep libfreenect2
libfreenect2.so => /usr/local/lib/libfreenect2.so (0x00007fc0e41e7000)
Apparently it's linked.

wiedemeyer, 2015-05-29:
so now ldd that libfreenect2.so file. `ldd /usr/local/lib/libfreenect2.so | grep libusb`.
If it is not pointing to `libfreenect2/depends/libusb/...` then something is wrong.
Did you add the modifications to the CMakeLists.txt mentioned in the README?

lakehanne, 2015-05-29:
So here is the output after I did,
>ldd /usr/local/lib/libfreenect2.so | grep libusb

>#:~/catkin_ws/devel/lib/kinect2_bridge$ ldd /usr/local/lib/libfreenect2.so | grep libusb
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007fce24ee4000)
But I did add the modifications to CMakeLists.txt in the Readme. Here are the added ines copied from my libfreenect2/examples/protonect/CMakeLists.txr file: 

># LibUSB
GET_FILENAME_COMPONENT(LIBUSB_DIR "${MY_DIR}/../../depends/libusb/" REALPATH)

>INCLUDE_DIRECTORIES("${LIBUSB_DIR}/include/libusb-1.0/")
LINK_DIRECTORIES("${LIBUSB_DIR}/lib/")

wiedemeyer, 2015-05-29:
And you replaced the other lines, right? Then run `sudo make install` in the build directory again.
If that does not help. remove everything in the build folder and run `cmake ../../examples/protonect/ -DENABLE_CXX11=ON` and `make && sudo make install` again.

lakehanne, 2015-05-29:
>"And you replaced the other lines, right?"

Yes, I did.

>Then run `sudo make install` in the build directory again.

I noticed the cloned `libfreenect2` folder does not come with a build directory but in your READMe file you mentioned that 
>cd ../build
mkdir linux

Did you mean
>mkdir ../build

and 

>cd ../build
mkdir linux?

or can I just skip the three lines:
>cd ../build
mkdir linux
cd linux

and do 
>cmake ../../examples/protonect/ -DENABLE_CXX11=ON
make && sudo make install

Also, I think I might have found why my output was not transferring. The former opencv installation I used created another CMakeLists.txt file in protonect which I had not modified. I have modified this now.

wiedemeyer, 2015-05-29:
on previous versions there was a build folder. I updated the README today. You can see the changed instructions if you switch the github view to devel.

lakehanne, 2015-05-29:
Ok. I have rebuilt libfreenect2. Executed `lldd /usr/local/lib/libfreenect2.so | grep libusb` and 
>lex@Lex:~/Documents/libfreenect2/build$ ldd /usr/local/lib/libfreenect2.so | grep libusb
	libusb-1.0.so.0 => /home/lex/Documents/libfreenect2/depends/libusb/lib/libusb-1.0.so.0 (0x00007f154d9ce000)
Looks like it points to the depends file now. 
So I go back and do `roscore` within `catkin_ws`, opened a new shell, sourced setup.bash and `rosrun kinect2_bridge kinect2_bridge` and we get back to square one:

> [TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[Freenect2DeviceImpl] stopped

Mybe I need to rebuild iai_kinect2 after reinstalling libfreenect2?

lakehanne, 2015-05-30:
I was reading through [CMakeLists.txt]_https://github.com/code-iai/iai_kinect2/blob/master/kinect2_bridge/CMakeLists.txt _ https://github.com/code-iai/iai_kinect2/blob/master/kinect2_bridge/CMakeLists.txt and it seems you require at least version 2.8.3. I have version 2.4.9 on my computer. COuld this be the reason why I cannot transfer?

lakehanne, 2015-05-31:
I noticed the following when I recompile the libfreenect cmake

>CMake Warning (dev) in /home/lex/libfreenect2/depends/glfw_src/src/CMakeLists.txt:
  Policy CMP0022 is not set: INTERFACE_LINK_LIBRARIES defines the link
  interface.  Run "cmake --help-policy CMP0022" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  Target "glfw" has an INTERFACE_LINK_LIBRARIES property which differs from
  its LINK_INTERFACE_LIBRARIES properties.

  INTERFACE_LINK_LIBRARIES:

    /usr/lib/x86_64-linux-gnu/libX11.so;-lpthread;/usr/lib/x86_64-linux-gnu/libXrandr.so;/usr/lib/x86_64-linux-gnu/libXi.so;/usr/lib/x86_64-linux-gnu/libXxf86vm.so;/usr/lib/x86_64-linux-gnu/librt.so;/usr/lib/x86_64-linux-gnu/libm.so;/usr/lib/x86_64-linux-gnu/libGL.so

  LINK_INTERFACE_LIBRARIES:

    

This warning is for project developers.  Use -Wno-dev to suppress it.

Could it be what I am looking for?

wiedemeyer, 2015-06-03:
This warning is also normal. Did you managed to get it working, or are you still facing some problems?

lakehanne, 2015-06-04:
Protonect now outputs depth, ir and rgb but rosrun gives the following:

>root@lex:/home/lex/catkin_ws# rosrun kinect2_bridge kinect2_bridge 
X server found. dri2 connection failed! 
X server found. dri2 connection failed! 
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/lex/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

>X server found. dri2 connection failed! 
X server found. dri2 connection failed! 
X server found. dri2 connection failed! 
X server found. dri2 connection failed! 
X server found. dri2 connection failed! 
X server found. dri2 connection failed! 
[OpenCLDepthPacketProcessor::listDevice]  devices:
  0: Intel(R) HD Graphics Haswell GT2 Mobile (GPU)[Intel]
  1: Intel(R) HD Graphics Haswell GT2 Mobile (GPU)[Intel]
  2: Quadro K1100M (GPU)[NVIDIA Corporation]
[OpenCLDepthPacketProcessor::init]  selected device: Intel(R) HD Graphics Haswell GT2 Mobile (GPU)[Intel]
X server found. dri2 connection failed! 
X server found. dri2 connection failed! 
X server found. dri2 connection failed! 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 14 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:5 with serial 501185143042
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 501185143042 (selected)
[Freenect2DeviceImpl] opening...
[UsbControl::claimInterfaces(ControlAndRgbInterfaceId)] failed! libusb error -6: LIBUSB_ERROR_BUSY
[Freenect2DeviceImpl] closing...
[Freenect2DeviceImpl] deallocating usb transfer pools...
[Freenect2DeviceImpl] closing usb device...
[Freenect2DeviceImpl] closed
[Freenect2DeviceImpl] failed to open Kinect v2 @4:5!
no device connected or failure opening the default one!
using sensor defaults for color intrinsic parameters.
using sensor defaults for ir intrinsic parameters.
using defaults for rotation and translation.
using defaults for depth shift.
Segmentation fault (core dumped)

wiedemeyer, 2015-06-05:
So `Protonect gl` (with the opengl depth processor works) and `Protonect cl` does not work, right? Then try `rosrun kinect2_bridge kinect2_bridge _depth_method:=opengl reg_method:=cpu`.
Also there is a usb error `LIBUSB_ERROR_BUSY`. `Protonect` doesn't give this error? Then probably the wrong libusb is linked.

lakehanne, 2015-06-06:
I get this:

>$:~/catkin_ws$ rosrun kinect2_bridge kinect2_bridge _depth_method:=opengl reg_method:=cpu
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/lex/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[Freenect2Impl] enumerating devices...
[Freenect2Impl] 12 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:3 with serial 501185143042
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 501185143042 (selected)
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

starting kinect2

[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    2e213355202e2e2e2e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 09 13 11 47 4b 52 35 32 33 2e 31    31332e2e2e2e2e2e474b523532332e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  5d 26 00 00                                        5d262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  43 26 00 00                                        43262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[... 88 lines cut ...]
  0, 1081.3720703125, 539.5;
  0, 0, 1]
distortion coefficients color:
[0, 0, 0, 0, 0]
camera matrix ir:
[364.5779113769531, 0, 259.4354858398438;
  0, 364.5779113769531, 209.3105010986328;
  0, 0, 1]
distortion coefficients ir:
[0.09317874908447266, -0.2718816995620728, 0, 0, 0.0968393012881279]
rotation:
[1, 0, 0;
  0, 1, 0;
  0, 0, 1]
translation:
[-0.052; 0; 0]
depth shift:
0

[DepthRegistration::New] Using CPU registration method!
[DepthRegistration::New] Using CPU registration method!
[kinect2_bridge] waiting for clients to connect

/home/lex/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge/home/lex/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge: symbol lookup error: /home/lex/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge: symbol lookup error: /home/lex/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge: undefined symbol: _ZN12libfreenect222SyncMultiFrameListener15waitForNewFrameERSt3mapINS_5Frame4TypeEPS2_St4lessIS3_ESaISt4pairIKS3_S4_EEEi
: undefined symbol: _ZN12libfreenect222SyncMultiFrameListener15waitForNewFrameERSt3mapINS_5Frame4TypeEPS2_St4lessIS3_ESaISt4pairIKS3_S4_EEEilex@lex:~/catkin_ws$

lakehanne, 2015-06-06:
Is libfreenect2 supposed to be in the same folder as iai_kinect2?

wiedemeyer, 2015-06-08:
No. The message `[TransferPool::submit] failed to submit transfer` should not appear. And the `undefined symbol: _ZN12libfreenect222SyncMultiFrameListener15waitForNewFrameERSt3mapINS_5Frame4TypeEPS2_St4lessIS3_ESaISt4pairIKS3_S4_EEEi` says, that the config header from libfreenect2 differs from the config used while compiling libfreenect2.
Please checkout the recent libfreenect2 and iai_kinect2 repositories and follow the instructions in the [README](https://github.com/code-iai/iai_kinect2/blob/master/README.md) step by step.

lakehanne, 2015-06-08:
Thank you very kindly. I have reformatted Ubuntu all over again, compiled nvidia-331 from the binaries on Ubuntu server.
When I run ../Protonect, I get this:

>lex@lex:~/Documents/libfreenect2/examples/protonect/bin$ ./Protonect 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 12 usb devices connected
[Freenect2Impl] failed to open Kinect v2 @4:3!
[Freenect2Impl] found 0 devices
no device connected!
lex@lex:~/Documents/libfreenect2/examples/protonect/bin$ ./Protonect 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 11 usb devices connected
[Freenect2Impl] found 0 devices
no device connected!
lex@lex:~/Documents/libfreenect2/examples/protonect/bin$ ./Protonect 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 12 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:5 with serial 501185143042
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    2e213355202e2e2e2e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 09 13 11 47 4b 52 35 32 33 2e 31    31332e2e2e2e2e2e474b523532332e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

>[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  00 04 00 00                                        2e2e2e2e

>[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  02 04 00 00                                        2e2e2e2e

>[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 501185143042
device firmware: 4.3.3912.0.7
[DepthPacketStreamParser::onDataReceived] skipping depth packet
[DepthPacketStreamParser::onDataReceived] skipping depth packet
[OpenGLDepthPacketProcessor] avg. time: 6.6766ms -> ~149.777Hz
[OpenGLDepthPacketProcessor] avg. time: 5.70109ms -> ~175.405Hz
[OpenGLDepthPacketProcessor] avg. time: 5.71175ms -> ~175.078Hz
[OpenGLDepthPacketProcessor] avg. time: 5.69785ms -> ~175.505Hz
[OpenGLDepthPacketProcessor] avg. time: 5.7016ms -> ~175.389Hz
[OpenGLDepthPacketProcessor] avg. time: 5.73105ms -> ~174.488Hz
[OpenGLDepthPacketProcessor] avg. time: 5.73777ms -> ~174.284Hz
[OpenGLDepthPacketProcessor] avg. time: 5.72663ms -> ~174.623Hz
[OpenGLDepthPacketProcessor] avg. time: 5.82086ms -> ~171.796Hz
[OpenGLDepthPacketProcessor] avg. time: 6.66151ms -> ~150.116Hz

`glxinfo | grep "OpenGL version"` gives `OpenGL version string: 4.4.0 NVIDIA 331.113` and 

`glxinfo | grep "OpenGL"` gives

>OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: Quadro K1100M/PCIe/SSE2
OpenGL core profile version string: 4.3.0 NVIDIA 331.113
OpenGL core profile shading language version string: 4.30 NVIDIA via Cg compiler
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.4.0 NVIDIA 331.113
OpenGL shading language version string: 4.40 NVIDIA via Cg compiler
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL extensions:

When I do rosrun in catkin workspace, I have:
>lex@lex:~/catkin_ws$ rosrun kinect2_bridge kinect2_bridge
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/lex/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

>[Freenect2Impl] enumerating devices...
[Freenect2Impl] 12 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:7 with serial 501185143042
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 501185143042 (selected)
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

>starting kinect2

>[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    2e213355202e2e2e2e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 09 13 11 47 4b 52 35 32 33 2e 31    31332e2e2e2e2e2e474b523532332e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

>[CommandTransaction::receive] bulk transfer failed! libusb error -1: LIBUSB_ERROR_IO
*** Error in `/home/lex/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge': double free or corruption (top): 0x0000000003bd63e0 ***
Aborted (core dumped)

lakehanne, 2015-06-08:
I later installed the libusbX superspeed patch from Joshua Blake and I got Protonect outputting depth, IR and RGB. 
However, iai_kinect2 would still not fall in love with me with either `rosrun kinect2_bridge kinect2_bride depth_method:=<opengl | opencl>`
Here's my latest output:

>lex@lex:~/catkin_ws$ rosrun kinect2_bridge kinect2_bridge depth_method:=opengl
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/lex/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

>[Freenect2Impl] enumerating devices...
[Freenect2Impl] 12 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:3 with serial 501185143042
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 501185143042 (selected)
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

>starting kinect2

>[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    2e213355202e2e2e2e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 09 13 11 47 4b 52 35 32 33 2e 31    31332e2e2e2e2e2e474b523532332e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

>[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

>[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

>[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started

>device serial: 501185143042
device firmware: 4.3.3912.0.7
[Freenect2DeviceImpl] stopping...
[Freenect2DeviceImpl] disabling usb transfer submission...
[Freenect2DeviceImpl] canceling usb transfers...
[DepthPacketStreamParser::onDataReceived] not all subsequences received 0
>[DepthPacketStreamParser::onDataReceived] not all subsequences received 512
[Freenect2DeviceImpl] stopped

>default ir camera parameters: 
fx 364.578, fy 364.578, cx 259.435, cy 209.311
k1 0.0931787, k2 -0.271882, p1 0, p2 0, k3 0.0968393

default color camera parameters: 
fx 1081.37, fy 1081.37, cx 959.5, cy 539.5
using sensor defaults for color intrinsic parameters.
using sensor defaults for ir intrinsic parameters.
using defaults for rotation and translation.
using defaults for depth shift.

>camera parameters used:
camera matrix color:
[1081.3720703125, 0, 959.5;
  0, 1081.3720703125, 539.5;
  0, 0, 1]
distortion coefficients color:
[0, 0, 0, 0, 0]
camera matrix ir:
[364.5779113769531, 0, 259.4354858398438;
  0, 364.5779113769531, 209.3105010986328;
  0, 0, 1]
distortion coefficients ir:
[0.09317874908447266, -0.2718816995620728, 0, 0, 0.0968393012881279]
rotation:
[1, 0, 0;
  0, 1, 0;
  0, 0, 1]
translation:
[-0.052; 0; 0]
depth shift:
0

>[DepthRegistration::New] Using CPU registration method!
[DepthRegistration::New] Using CPU registration method!
[kinect2_bridge] waiting for clients to connect

wiedemeyer, 2015-06-09:
The output looks totally fine. It will wait until you connect to one of the topics.
Also why did you had to apply the patch from Joshua Blake? If you install the dependencies in `libfreenect2/depends` with `./install_ubuntu.sh`, it will download libusb with the patch and compile it.

lakehanne, 2015-06-09:
I did install from the depends folder. When I had the LIBUSB -IO ERROR, I figured something was not right with my libusb configuration. That's why I resolved to use the libusbX of Blake.

lakehanne, 2015-06-09:
What is the unit of the error that the calibration codes return?
I have the following from my sync calibration. 

>calibrating Color and Ir extrinsics...
error: 58.5421

Collected 100+ images. I'm assuming something out of the ordinary so I am redoing the sync experiment.

lakehanne, 2015-06-10:
I also notice that everytime I rosrun to calibrate the depth images, I get this toward the end:

>frame: ./0103_depth.png
frame: ./0103_sync_depth.png
frame: ./0104_depth.png
frame: ./0105_depth.png
Segmentation fault (core dumped)

There'e just no end to errors, are there?

wiedemeyer, 2015-06-10:
It is error in pixels. Good results should be under 0.3. The lower the better. In #66 are some tips for good calibrations.
Before calibrating `depth` you need to calibrate `ir`, `color` and `sync`.

lakehanne, 2015-06-10:
Yes, I did. Now, I got this:

>[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistration::New] Using OpenCL registration method!
X server found. dri2 connection failed! 
/dev/dri/card0 not authenticated
/dev/dri/card1 not authenticated

wiedemeyer, 2015-06-11:
Then something is wrong with you openCL installation. Run kinect2_bidge with the parameter `_reg_method:=cpu`

lakehanne, 2015-06-11:
Yeah, I figured. I tried adding the _local patch that was discussed on beignet's README.md page but I wasn't successful. It (kinect2_bridge) works well on cpu however. But I keep getting off in my extrinsic calibration.
Thanks for your help anyway!

wiedemeyer, 2015-06-12:
You can also try to install beignet from the ppa that is mentioned in the readme. Or you can try to run kinect2_bridge with sudo. Does `clinfo` provide any output?

lakehanne, 2015-06-15:
Thank you! I've since finished the calibration actually. But everytime I do `rosrun registration_viewer viewer image`, I get `[rospack] Error: package 'registration_viewer' not found` even though my setup.bash files are properly sourced. Libfreenect2 rund fine though.

lakehanne, 2015-06-15:
I have installed beignet from the ppa. It publishes in the terminal window but I see no images:

>[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistration::New] Using OpenCL registration method!
X server found. dri2 connection failed! 
/dev/dri/card0 not authenticated
/dev/dri/card1 not authenticated
Device open failed, aborting...
cl_get_gt_device(): error, unknown device: ffffffff
[DepthRegistrationOpenCL::init] ERROR: clGetDeviceIDs(-1)
[kinect2_bridge] client connected. starting device...
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    2e213355202e2e2e2e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 09 13 11 47 4b 52 35 32 33 2e 31    31332e2e2e2e2e2e474b523532332e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

>[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  5d 26 00 00                                        5d262e2e

>[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  5f 26 00 00                                        5f262e2e

>[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
[OpenGLDepthPacketProcessor] avg. time: 6.12049ms -> ~163.386Hz
[TurboJpegRgbPacketProcessor] avg. time: 16.4175ms -> ~60.9104Hz
[OpenGLDepthPacketProcessor] avg. time: 5.78558ms -> ~172.843Hz
[TurboJpegRgbPacketProcessor] avg. time: 16.3528ms -> ~61.1516Hz
[OpenGLDepthPacketProcessor] avg. time: 5.78245ms -> ~172.937Hz
[TurboJpegRgbPacketProcessor] avg. time: 16.4252ms -> ~60.8819Hz
[OpenGLDepthPacketProcessor] avg. time: 5.82945ms -> ~171.543Hz
[TurboJpegRgbPacketProcessor] avg. time: 16.4733ms -> ~60.7044Hz
[OpenGLDepthPacketProcessor] avg. time: 5.8193ms -> ~171.842Hz

lakehanne, 2015-06-15:
>lex@lex:~/catkin_ws$ rosrun registration_viewer viewer image
[rospack] Error: package 'registration_viewer' not found

wiedemeyer, 2015-06-16:
hi,
it was renamed last week or so. its now called kinect2_viewer.

lakehanne, 2015-06-16:
Okay. kinect2_viewer is apparently working but I still cannot get the images published.

wiedemeyer, 2015-06-17:
What do you mean with "cannot get the images published"? Is kinect2_viewer showing images?

lakehanne, 2015-06-17:
it's showing images now after a little tweak of opencl. Thanks!

Are there face recognition api's that come with ros that I can integrate to
the code?

Also I'm seeing a shadow offset of about 5-8mm on the kinect2 cloud
grabber. I think it has something to do with my intrinsic calibration. Is
there a way I can manually add an offset in the pose yaml file to correct
the shadows?

lakehanne, 2015-06-17:
I meant my extrinsic calibration.

wiedemeyer, 2015-06-17:
What tweak did you had to apply to openCL?

I not in to face recognition. You may want to search the ROS websites for that.

Yes, you can edit the translation in one of the calibration files in `kinect2_bridge/data`. But it is better to do a proper calibration.

lakehanne, 2015-06-17:
[This](https://01.org/zh/beignet/downloads/linux-kernel-patch-hsw-support?langredirect=1)

closed 2015-06-19 by lakehanne

lakehanne, 2015-07-14:
I made a face detector and eye tracker based on opencv haar classifiers with your kinect2_viewer code. It's [here](https://github.com/lakehanne/iai_kinect2/tree/master/kinect2_viewer). It's pretty rusty at the moment giving eye center coordinates for both eyes and and putting a bounding rectangular box around the face. Ultimately, my goal would be to put a mesh similar to [this](https://github.com/lakehanne/iai_kinect2/tree/master/kinect2_viewer) and get a more robust face tracker irrespective of race, age, face cosmetics and occlusion.