code-iai/iai_kinect2 #99: [Question] Is depth rectified before the registration?

JuantAldea, 2015-05-28:
Hello Thiemo,
I might have gotten it wrong but.. why the depth is not being rectified by the bridge before the depth registration takes place?

wiedemeyer, 2015-05-28:
because the rectification and registration is done in one step.

JuantAldea, 2015-05-28:
Could you please point me to the right DepthRegistrationCPU method where this (lens correction) is happening?

JuantAldea, 2015-05-28:
Nevermind, found it, thanks!

closed 2015-05-28 by JuantAldea

wiedemeyer, 2015-05-28:
It is happening [here](https://github.com/code-iai/iai_kinect2/blob/master/depth_registration/src/depth_registration_cpu.cpp#L129). The map gives for each coordinate in the rectified image the corresponding coordinate in the input image. This is then used to interpolate the values.