code-iai/iai_kinect2 #103: No depth stream

SamTew, 2015-05-30:
When i am running kinect2_bridge, there are rgb stream. I can access rgb image but no depth image. When i try to rostopic echo /kinect2/depth/image, nothing was shown. This situation happen when i compile freenect2 with opengl and opencl are disabled.
cmake ../../examples/protonect/ -DENABLE_CXX11=ON -DENABLE_OPENCL=OFF -DENABLE_OPENGL=OFF
I would like to ask whether OpenGL is necessary to get depth data stream?

wiedemeyer, 2015-06-01:
When you run `Protonect`, does it display depth an ir images? If not, then it is probably an USB issue. Does it say `submit transfer failed`?

SamTew, 2015-06-02:
When i run Protonect, the output is
sam@sam-X450LCP:~/libfreenect2/examples/protonect$ ./bin/Protonect 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 11 usb devices connected
[Freenect2Impl] found valid Kinect v2 @3:3 with serial 508442542542
[Freenect2Impl] found 1 devices
Xlib:  extension "GLX" missing on display ":0".
Error: OpenGL version 3.3 not supported.
       Your version is 0.0.
       Try updating your graphics driver.
Segmentation fault (core dumped)

when i run Protonect cpu, the output is
sam@sam-X450LCP:~/libfreenect2/examples/protonect$ ./bin/Protonect cpu
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 11 usb devices connected
[Freenect2Impl] found valid Kinect v2 @3:3 with serial 508442542542
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 08 0d 01 47 4d 41 39 38 33 2e 31    31332e2e2e2e2e2e474d413938332e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  7f 26 00 00                                        7f262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  7f 26 00 00                                        7f262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 508442542542
device firmware: 4.3.3912.0.7
[TurboJpegRgbPacketProcessor] avg. time: 22.4266ms -> ~44.5899Hz
[TurboJpegRgbPacketProcessor] avg. time: 23.5759ms -> ~42.4163Hz
^C[TurboJpegRgbPacketProcessor] avg. time: 23.1104ms -> ~43.2705Hz
[TurboJpegRgbPacketProcessor] avg. time: 22.0777ms -> ~45.2946Hz
[TurboJpegRgbPacketProcessor] avg. time: 23.5685ms -> ~42.4295Hz
[TurboJpegRgbPacketProcessor] avg. time: 23.3936ms -> ~42.7467Hz

wiedemeyer, 2015-06-03:
since this is related to libfreenect2 and you already have an [issue](https://github.com/OpenKinect/libfreenect2/issues/248) there, we can close this one.

closed 2015-06-03 by wiedemeyer