code-iai/iai_kinect2 #104: have rgb image, but no image in depth as well as ir

horxianfeng, 2015-05-30:
I am using ubutu 14.04. Ros Indigo.
intel i5. Nvidia 840.

So far, i install libreenect2 succesfully.
However, when i try to run rosrun kinect2_bridge kinect2_bridge, i face the error below( I had tried to update my kernel and etc, but i still cant solve the problem):

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started

device serial: 508442542542
device firmware: 4.3.3912.0.7
[Freenect2DeviceImpl] stopping...
[Freenect2DeviceImpl] disabling usb transfer submission...
[Freenect2DeviceImpl] canceling usb transfers...
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[DepthPacketStreamParser::onDataReceived] not all subsequences received 0
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[Freenect2DeviceImpl] stopped

default ir camera parameters: 
fx 365.861, fy 365.861, cx 254.502, cy 205.253
k1 0.0964056, k2 -0.275756, p1 0, p2 0, k3 0.0921489

default color camera parameters: 
fx 1081.37, fy 1081.37, cx 959.5, cy 539.5
using sensor defaults for color intrinsic parameters.
using sensor defaults for ir intrinsic parameters.
using defaults for rotation and translation.
using defaults for depth shift.

camera parameters used:
camera matrix color:
[1081.3720703125, 0, 959.5;
0, 1081.3720703125, 539.5;
0, 0, 1]
distortion coefficients color:
[0, 0, 0, 0, 0]
camera matrix ir:
[365.8608093261719, 0, 254.502197265625;
0, 365.8608093261719, 205.2525939941406;
0, 0, 1]
distortion coefficients ir:
[0.09640563279390335, -0.2757557928562164, 0, 0, 0.09214892983436584]
rotation:
[1, 0, 0;
0, 1, 0;
0, 0, 1]
translation:
[-0.052; 0; 0]
depth shift:
0

[DepthRegistration::New] Using CPU registration method!
[DepthRegistration::New] Using CPU registration method!
[kinect2_bridge] waiting for clients to connect

/home/xfhor/kinectv2/devel/lib/kinect2_bridge/kinect2_bridge/home/xfhor/kinectv2/devel/lib/kinect2_bridge/kinect2_bridge: symbol lookup error: /home/xfhor/kinectv2/devel/lib/kinect2_bridge/kinect2_bridge: undefined symbol: _ZN12libfreenect222SyncMultiFrameListener15waitForNewFrameERSt3mapINS_5Frame4TypeEPS2_St4lessIS3_ESaISt4pairIKS3_S4_EEEi
: symbol lookup error: /home/xfhor/kinectv2/devel/lib/kinect2_bridge/kinect2_bridge: undefined symbol: _ZN12libfreenect222SyncMultiFrameListener15waitForNewFrameERSt3mapINS_5Frame4TypeEPS2_St4lessIS3_ESaISt4pairIKS3_S4_EEEi

lakehanne, 2015-05-31:
Possible duplicate to issue {101](https://github.com/code-iai/iai_kinect2/issues/102). 
What's your kernel version? Also, are you using the libusb patch? In other words, did you make the changes to the [CMakelists.txt](https://github.com/OpenKinect/libfreenect2/blob/master/examples/protonect/CMakeLists.txt#L48-50) file before installing libfreenect2?

horxianfeng, 2015-05-31:
@lakehanne  thx for your advice.i have tried to change the CMakelists.txt. I then rebuild the workspace as well. I can run the command rosrun kinect2_bridge kinect2_bridge now. However , when i rostopic echo, i can successfully have the rgb data, but fail to access the depth data. 
Terminal shows: 
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[Freenect2DeviceImpl] started
[RgbPacketStreamParser::onDataReceived] packetsize or sequence doesn't match!
[TurboJpegRgbPacketProcessor] avg. time: 23.4626ms -> ~42.621Hz
[kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~7100.91Hz (0.140827ms) publishing rate: ~23.3051Hz
[TurboJpegRgbPacketProcessor] avg. time: 19.9854ms -> ~50.0364Hz
[kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~6934.01Hz (0.144217ms) publishing rate: ~29.9732Hz

At the same time, when i run  protonect example, ./bin/Protonect, it also shows rgb image perfectly but depth and ir image are black.

horxianfeng, 2015-05-31:
Now I delete all my libfreenect2 and workspace..
I redo it again. 
At 1st step of code-iai, i cant run the follow instruction:
sudo apt-get install -y build-essential libturbojpeg libtool autoconf libudev-dev cmake mesa-common-dev freeglut3-dev libxrandr-dev doxygen libxi-dev libopencv-dev

Then, i try to ignore this instruction. I skip to git clone https://github.com/OpenKinect/libfreenect2

Everything seems okay until i run the follow command:
rosdep install -r --from-paths .

The terminal appears:
 xfhor@xfhor-TP500LN:~/kinectv2_try3/src/iai_kinect2$ rosdep install -r --from-paths .
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
registration_viewer: Cannot locate rosdep definition for [kinect2_bridge]
kinect2_bridge: Cannot locate rosdep definition for [depth_registration]
kinect2_calibration: Cannot locate rosdep definition for [kinect2_bridge]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

lakehanne, 2015-05-31:
>"At 1st step of code-iai, i cant run the follow instruction: sudo apt-get
install -y build-essential libturbojpeg libtool autoconf libudev-dev cmake
mesa-common-dev freeglut3-dev libxrandr-dev doxygen libxi-dev libopencv-dev"

You can't do all that cause the programs are already installed.

Also, ```
rosdep install -r --from-paths . ERROR: the following packages/stacks could
not have their rosdep
```

should be ignored as those keys are part of iai_kinect. @wiedemeyer, this
is something you should probably point out in your ReadMe.md file.

So, in my estimation, I think you are fine. Now, do rosrun etc...
On May 31, 2015 12:09 PM, "horxianfeng" <notifications@github.com> wrote:

> Now I delete all my libfreenect2 and workspace..
> I redo it again.
> At 1st step of code-iai, i cant run the follow instruction:
> sudo apt-get install -y build-essential libturbojpeg libtool autoconf
> libudev-dev cmake mesa-common-dev freeglut3-dev libxrandr-dev doxygen
> libxi-dev libopencv-dev
>
> Then, i try to ignore this instruction. I skip to git clone
> https://github.com/OpenKinect/libfreenect2
>
> Everything seems okay until i run the follow command:
> rosdep install -r --from-paths .
>
> The terminal appears:
> xfhor@xfhor-TP500LN:~/kinectv2_try3/src/iai_kinect2$ rosdep install -r
> --from-paths .
> ERROR: the following packages/stacks could not have their rosdep keys
> resolved
> to system dependencies:
> registration_viewer: Cannot locate rosdep definition for [kinect2_bridge]
> kinect2_bridge: Cannot locate rosdep definition for [depth_registration]
> kinect2_calibration: Cannot locate rosdep definition for [kinect2_bridge]
> Continuing to install resolvable dependencies...
> #All required rosdeps installed successfully
>
> --
> Reply to this email directly or view it on GitHub
> <https://github.com/code-iai/iai_kinect2/issues/104#issuecomment-107224926>
> .
>

horxianfeng, 2015-05-31:
@lakehanne  thank you so much. According to your advice, i keep it on and the result is as below:

I can run the rosrun kinect2_bridge kniect2_bridge

When I rostopic echo /kinect2/rgb/image. I get lots of data.

However, when i rostopic echo /kinect2/depth/image, my data appears zero. The terminal appears like below : 

[kinect2_bridge] color processing: ~4916.86Hz (0.203382ms) publishing rate: ~29.6333Hz
[RgbPacketStreamParser::onDataReceived] skipping rgb packet!
[RgbPacketStreamParser::onDataReceived] skipping rgb packet!
[OpenGLDepthPacketProcessor] avg. time: 5.10208ms -> ~195.999Hz
[kinect2_bridge] depth processing: ~789.767Hz (1.2662ms) publishing rate: ~29.9217Hz
[kinect2_bridge] color processing: ~4788.39Hz (0.208838ms) publishing rate: ~29.2568Hz
[RgbPacketStreamParser::onDataReceived] skipping rgb packet!
[TurboJpegRgbPacketProcessor] avg. time: 25.5908ms -> ~39.0766Hz
[RgbPacketStreamParser::onDataReceived] skipping rgb packet!
[RgbPacketStreamParser::onDataReceived] skipping rgb packet!
[kinect2_bridge] depth processing: ~761.475Hz (1.31324ms) publishing rate: ~30.2356Hz
[kinect2_bridge] color processing: ~4871.43Hz (0.205278ms) publishing rate: ~29.2389Hz
[OpenGLDepthPacketProcessor] avg. time: 5.18674ms -> ~192.799Hz
[TurboJpegRgbPacketProcessor] avg. time: 25.8959ms -> ~38.6161Hz

Furthermore, when i run ./bin/Protonect
i also get the rgb image only

lakehanne, 2015-05-31:
What is the kernel version of your ubuntu installation? Can you paste the
output of `uname -a`. You are using the nvidia card. Can you confirm if you
followed through with the nvidia issues that Thiemo raised in the ReadMe.md
file?

horxianfeng, 2015-05-31:
The reuslt is : 
xfhor@xfhor-TP500LN:~$ uname -a
Linux xfhor-TP500LN 3.16.0-38-generic #52~14.04.1-Ubuntu SMP Fri May 8 09:43:57 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

I try to change
 OpenGLDepthPacketProcessor *depth_processor = new OpenGLDepthPacketProcessor(0, debug_); (on src/packet_pipeline.cpp:108)
to
 CpuDepthPacketProcessor *depth_processor = new CpuDepthPacketProcessor();
It works but the output is slow. Depth, rgb, ir image is lagging.

What is the nvidia issues that Thiemo raised in the ReadMe.md
file?  Are you mind to kindly post me the link?

horxianfeng, 2015-05-31:
I try :

sudo su
echo 0 > /sys/module/i915/parameters/enable_cmd_parser

but nothing happen..
I still cant get depth image

lakehanne, 2015-05-31:
Your kernel seems fine. 
```
What is the nvidia issues that Thiemo raised in the ReadMe.md
file? Are you mind to kindly post me the link?
```

Here you go: Install the latest version of the Nvidia drivers, for example nvidia-346 from ppa:xorg-edgers and opencl-headers.

wiedemeyer, 2015-06-01:
Hi,
let me try to answer the questions:
- `failed to submit transfer` this means there is something wrong with the USB connection. If `Protonect` is not showing depth and ir images, kinect2_bridge will not work.
- `symbol lookup error` means that the libfreenect.so that is linked is not compiled with C++11, but the config header states that it is.
- the rosdep errors are totally fine. Those packages listed are the ones within the iai_kinect2 meta package, rosdep does not know them and thats ok.
First make sure Protonect works fine, before trying to run kinect2_bridge.

horxianfeng, 2015-06-01:
@lakehanne  I have install nvidea 352 by following the steps of https://help.ubuntu.com/community/BinaryDriverHowto/Nvidia

what about the opencl.header? i totally dont g=have any ideas of it..

@wiedemeyer  . So, what should i do know? Hope for your help.

wiedemeyer, 2015-06-02:
`sudo apt-get install opencl-headers`, then rebuild libfreenect2 (remove everything in the build folder, and run cmake and make again), then try `./Protonect cl`.

horxianfeng, 2015-06-02:
@wiedemeyer 
My terminal shows :

xfhor@xfhor-TP500LN:~/libfreenect2/build$ make && sudo make install
make: *** No targets specified and no makefile found.  Stop.

horxianfeng, 2015-06-02:
Thx..I fix it.

wiedemeyer, 2015-06-03:
can we close this now?

lakehanne, 2015-06-03:
Still same old Problem. When I launch Protonect:

>[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 501185143042
device firmware: 4.3.3912.0.7
[DepthPacketStreamParser::onDataReceived] not all subsequences received 0
[DepthPacketStreamParser::onDataReceived] skipping depth packet
[DepthPacketStreamParser::onDataReceived] skipping depth packet
[DepthPacketStreamParser::onDataReceived] skipping depth packet
[DepthPacketStreamParser::onDataReceived] skipping depth packet
[DepthPacketStreamParser::onDataReceived] skipping depth packet
[DepthPacketStreamParser::onDataReceived] skipping depth packet
drm_intel_gem_bo_context_exec() failed: Invalid argument
[OpenCLDepthPacketProcessor::run] ERROR: clEnqueueNDRangeKernel (-5)
terminate called after throwing an instance of 'cl::Error'
  what():  clEnqueueNDRangeKernel
Aborted (core dumped)
lex@Lex:~/libfreenect2/examples/protonect/bin$ ./Protonect 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 12 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:3 with serial 501185143042
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    2e213355202e2e2e2e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 09 13 11 47 4b 52 35 32 33 2e 31    31332e2e2e2e2e2e474b523532332e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

>[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  23 26 00 00                                        23262e2e

>[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  23 26 00 00                                        23262e2e

>[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 501185143042
device firmware: 4.3.3912.0.7
[DepthPacketStreamParser::onDataReceived] not all subsequences received 0
init done 
opengl support available 
[OpenGLDepthPacketProcessor] avg. time: 3.58061ms -> ~279.282Hz
[TurboJpegRgbPacketProcessor] avg. time: 15.7286ms -> ~63.5785Hz
[RgbPacketStreamParser::onDataReceived] skipping rgb packet!
[RgbPacketStreamParser::onDataReceived] skipping rgb packet!
[DepthPacketStreamParser::onDataReceived] not all subsequences received 512
[OpenGLDepthPacketProcessor] avg. time: 3.56809ms -> ~280.262Hz
[TurboJpegRgbPacketProcessor] avg. time: 15.8664ms -> ~63.0264Hz
[OpenGLDepthPacketProcessor] avg. time: 4.04086ms -> ~247.472Hz
[TurboJpegRgbPacketProcessor] avg. time: 16.3197ms -> ~61.2757Hz
[OpenGLDepthPacketProcessor] avg. time: 4.2359ms -> ~236.077Hz
[TurboJpegRgbPacketProcessor] avg. time: 16.3209ms -> ~61.2712Hz
[OpenGLDepthPacketProcessor] avg. time: 4.06713ms -> ~245.874Hz
[TurboJpegRgbPacketProcessor] avg. time: 16.5205ms -> ~60.531Hz
[OpenGLDepthPacketProcessor] avg. time: 3.90755ms -> ~255.915Hz
[TurboJpegRgbPacketProcessor] avg. time: 16.4382ms -> ~60.834Hz
[OpenGLDepthPacketProcessor] avg. time: 4.11971ms -> ~242.735Hz
[TurboJpegRgbPacketProcessor] avg. time: 16.0371ms -> ~62.3556Hz
[OpenGLDepthPacketProcessor] avg. time: 3.96849ms -> ~251.985Hz
[... 173 lines cut ...]
  0, 0, 1]
distortion coefficients ir:
[0.09317874908447266, -0.2718816995620728, 0, 0, 0.0968393012881279]
rotation:
[1, 0, 0;
  0, 1, 0;
  0, 0, 1]
translation:
[-0.052;
  0;
  0]
depth shift:
0

>[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistrationOpenCL::listDevice]  devices:
  0: Intel(R) HD Graphics Haswell GT2 Mobile (GPU)[Intel]
[DepthRegistrationOpenCL::init]  selected device: Intel(R) HD Graphics Haswell GT2 Mobile (GPU)[Intel]
[DepthRegistrationOpenCL::listDevice]  devices:
  0: Intel(R) HD Graphics Haswell GT2 Mobile (GPU)[Intel]
[DepthRegistrationOpenCL::init]  selected device: Intel(R) HD Graphics Haswell GT2 Mobile (GPU)[Intel]
[kinect2_bridge] waiting for clients to connect

I think I'm just gonna have to reformat my Ubuntu and see if this would fix the problem.

costashatz, 2015-06-04:
I have the same problem.

@wiedemeyer I made the modifications and Protonect works like a charm. But kinect2_bridge still links to system libusb and not the patched one. Any ideas?

@lakehanne Did you solve the problem somehow?

Thanks!

wiedemeyer, 2015-06-04:
You have to add the modifications, then create or empty the build folder. Run `cmake` again, then `make` and `sudo make install`. Then rebuild `kinect2_bridge`, to make sure rm the `devel` and `build` folder before running `catkin_make`.

wiedemeyer, 2015-06-04:
@lakehanne please listen to one of the topics, so that `kinect2_bridge` starts the device. Either use `rostopic hz ...` or `kinect2_viewer`.

costashatz, 2015-06-04:
@wiedemeyer I solved the problem by rebuilding everything! Thanks!

When running kinect2_viewer or kinect2_calibration should a window open or are they just bash? Because no window opens.. Thanks again!

wiedemeyer, 2015-06-04:
A window should open as soon as both, depth and color frames are received. If not, then probably one of them is missing. check the log output of `kinect2_bridge`. Whats the timings for depth and color.

costashatz, 2015-06-04:
I get the following output running kinect2_bridge:

```
[Freenect2DeviceImpl] started
[RgbPacketStreamParser::onDataReceived] packetsize or sequence doesn't match!
[kinect2_bridge] depth processing: ~0Hz (infms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~56.1349Hz (17.8142ms) publishing rate: ~22.996Hz
[TurboJpegRgbPacketProcessor] avg. time: 21.0197ms -> ~47.5745Hz
```

So something's wrong. Protonect still works like a charm! Any ideas on what's wrong?

wiedemeyer, 2015-06-04:
probably it links against the wrong libusb. Add the modifications the the `CMakeLists.txt`, run `cmake`, `make` and `sudo make install`. Maybe recompile `kinect2_bridge` and it should work.

lakehanne, 2015-06-04:
I think the CMakeLists.txt file in libfreenect's github page was changed
these past few days.
Noticed this after a clean installation of Ubuntu.
I am using that now and the only problem during compilation seem to be

>-- Could NOT find OpenCL (missing:  OPENCL_INCLUDE_DIRS)
Linking with these libraries:
opencv_calib3d;opencv_contrib;opencv_core;opencv_features2d;opencv_flann;opencv_gpu;opencv_highgui;opencv_imgproc;opencv_legacy;opencv_ml;opencv_objdetect;opencv_ocl;opencv_photo;opencv_stitching;opencv_superres;opencv_ts;opencv_video;opencv_videostab;usb;/usr/lib/x86_64-linux-gnu/libturbojpeg.so;pthread;glfw3
-- Configuring done
-- Generating done
-- Build files have been written to:
/home/lex/libfreenect2/examples/protonect

I am trying to find the best way to install OpenCL as the rgb image that I
was getting in ./Protonect is not even outputting right now.
Thanks guys! Will keep you posted.

Sincerely,
Olalekan

E: patlekano@gmail.com

"Towering genius disdains a beaten path. It seeks regions hitherto
unexplored." - Abraham Lincoln

lakehanne, 2015-06-04:
Hi @wiedemeyer , I have been able to acceess the depth image using Protonect.
So I moved on and I tried to rosrun kinect2_bridge but I was getting:

>root@lex:/home/lex/catkin_ws# rosrun kinect2_bridge kinect2_bridge 
X server found. dri2 connection failed! 
X server found. dri2 connection failed! 
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/lex/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

>X server found. dri2 connection failed! 
X server found. dri2 connection failed! 
X server found. dri2 connection failed! 
X server found. dri2 connection failed! 
X server found. dri2 connection failed! 
X server found. dri2 connection failed! 
[OpenCLDepthPacketProcessor::listDevice]  devices:
  0: Intel(R) HD Graphics Haswell GT2 Mobile (GPU)[Intel]
  1: Intel(R) HD Graphics Haswell GT2 Mobile (GPU)[Intel]
  2: Quadro K1100M (GPU)[NVIDIA Corporation]
[OpenCLDepthPacketProcessor::init]  selected device: Intel(R) HD Graphics Haswell GT2 Mobile (GPU)[Intel]
X server found. dri2 connection failed! 
X server found. dri2 connection failed! 
X server found. dri2 connection failed! 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 14 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:5 with serial 501185143042
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 501185143042 (selected)
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

costashatz, 2015-06-05:
@wiedemeyer kinect2_bridge was linking to wrong libusb again. I tried it with a clean install but it is linking to the wrong libusb again. I fixed the issue changing the LD_LIBRARY_PATH but this is only a workaround. Could you provide the correct CMakeLists so I can check if I am doing something wrong there?

Thanks a lot!

wiedemeyer, 2015-06-08:
@costashatz libfreenect2 changed a bit the last days. Now no modifications the the CMakeLists.txt are needed anymore. Please update the repositories and try again.

costashatz, 2015-06-16:
I haven't retried it yet. It's a really busy period right now. The package works fine when setting (the first time of compilation) the LD_LIBRARY_PATH to point to the correct libusb. I will try it again when I have time.

wkentaro, 2015-09-28:
Same problem here. I have installed libfreenect2 and iai_kinect2 today, so they're latest.
* Ubuntu 14.04
* indigo

wkentaro, 2015-09-28:
the output is 
```
% rosrun kinect2_bridge kinect2_bridge
[ERROR] [1443451971.142472546]: [registerPublisher] Failed to contact master at [localhost:11311].  Retrying...
[ INFO] [1443451973.857964376]: Connected to master at [localhost:11311]
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/wkentaro/ros/indigo/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[Freenect2Impl] enumerating devices...
[Freenect2Impl] 9 usb devices connected
[Freenect2Impl] found valid Kinect v2 @2:4 with serial 502986542542
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 502986542542 (selected)
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

starting kinect2

[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 0a 0b 0f 47 4d 46 31 33 33 2e 31    31332e2e2e2e2e2e474d463133332e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[... 148 lines cut ...]
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[Freenect2DeviceImpl] started
[RgbPacketStreamParser::onDataReceived] packetsize or sequence doesn't match!
[RgbPacketStreamParser::onDataReceived] skipping rgb packet!
[kinect2_bridge] depth processing: ~0Hz (infms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~41.9909Hz (23.8147ms) publishing rate: ~22.9772Hz
[TurboJpegRgbPacketProcessor] avg. time: 23.7759ms -> ~42.0594Hz
[kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~42.2818Hz (23.6508ms) publishing rate: ~29.9892Hz
```
the images are published but points are not.

wiedemeyer, 2015-10-14:
Sorry for the late response, but it looks like some libusb issue. Does it also happens when using Protonect? If not, please make sure that the libusb from libfreenect2 is used. You could check with `ldd`. If it also happens with Protonect please look/ask at libfreenect2 issues.

wkentaro, 2015-10-14:
Sorry, the problem with my kinect2 is fixed at https://github.com/code-iai/iai_kinect2/issues/165#issuecomment-147059607 .
I forgot to announce here about it.

closed 2015-10-14 by wiedemeyer