code-iai/iai_kinect2 #105: "OpenCV Error: Insufficient memory..." error after the updating from the repository

orajput, 2015-06-01:
Hello all,

Today I updated the iai_kinect2 repository (git pull). I also updated the libfreenect2 as well. Before that I was able to use packages in iai_kinect2 without any problem, but this update caused a couple of issues:
- Usually I compile libfreenect2 using `ENABLE_OPENCL`=ON but `ENABLE_OPENGL`=OFF (since, my system has Intel's integrated GPU and using OpenGL shows black IR & depth images as Protonect outputs). But, compiling `iai_kinect2`'s `kinect2_bridge` failed when the `ENABLE_OPENGL` option was set to OFF with the following error at [line 354](https://github.com/code-iai/iai_kinect2/blob/master/kinect2_bridge/src/kinect2_bridge.cpp#L354):
```bash 
CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o: In function `Kinect2Bridge::initPipeline(std::string const&, int, bool, bool, double, double)':
/home/Rajput/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:354: undefined reference to `libfreenect2::OpenGLPacketPipeline::OpenGLPacketPipeline(void*, bool)'
```
This I managed to solve by enabling both OpenCL and OpenGL while compiling libfreenect2/Protonect.

- Secondly, I am getting **OpenCV's Insufficient memory** error as soon as I run the executables like `viewer` from `registration_viewer` package or `kinect2_calibration` from `kinect2_calibration`. The error is as follows:
```bash 
OpenCV Error: Insufficient memory (Failed to allocate 44845819908 bytes) in OutOfMemoryError, file /build/buildd/opencv-2.4.8+dfsg1/modules/core/src/alloc.cpp, line 52
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/buildd/opencv-2.4.8+dfsg1/modules/core/src/alloc.cpp:52: error: (-4) Failed to allocate 44845819908 bytes in function OutOfMemoryError

Aborted (core dumped)
```
I also tried to run it in debug mode, and it seems like some frames were successfully read before the same Insufficient Memory error occurs. Please see the following screenshot.
[image]

*(Note: I am successfully able to run `Protonect` even after the update.)*

**My system specification:**
Processor: Intel Xeon E3-1225 v3 @ 3.20GHz
RAM: 4GiB
Graphics card: Integrated Graphics Controller (Xeon E3-1200 v3 Processor)
OS: Xubuntu 14.04.2 LTS (with Linux kernel 3.13.0-53-generic)

**Libraries:**
OpenCV  2.4.8 (from APT)
PCL 1.7.2-1 (from APT)
ROS Indigo (from APT)
OpenCL (compiled from source provided by Beignet - branch used: Release_v1.0)

wiedemeyer, 2015-06-02:
Hi,
the first problem: libfreenect2 creates a header called `config.h`, where the configuration like OpenCL, OpenGL, etc is stored. Everytime you change the configuration you need to `make install` again. So if you disable OpenGL check that the installed `config.h` also changed. If not, kinect2_bridge thinks OpenGL support is enabled, which leads to the undefined symbol error.

Second problem: Never seen this before. Please look at the memory consumption while execution. For example with `htop`. My guess is, that you don't have enough free memory left.

wiedemeyer, 2015-06-02:
I see, OpenCV tries to allocate 42 GB of RAM, that could be to much. Please let it run in debug and check where it happens.

orajput, 2015-06-02:
I always do a `sudo make install` after making any changes to the `libfreenect2`. Since, the `iai_kinect2` is looking for the `libfreenect2` in the system paths.

I did run the `kinect2_bridge` in debug-mode as shown in the screenshot in the main post. The error typically occurs in the `receiveColor()` or `receiveIr()` functions at lines [824](https://github.com/code-iai/iai_kinect2/blob/master/kinect2_bridge/src/kinect2_bridge.cpp#L824) or [863](https://github.com/code-iai/iai_kinect2/blob/master/kinect2_bridge/src/kinect2_bridge.cpp#L863). 

I will use `htop` and get back if I find something interesting.

Also, I tried to recreate the environment on another laptop (with OS: Ubuntu 14.04.2, Proc.: i7-3630, RAM: 12 GiB, GPU: NVIDIA Quadro K1000M). Although I had to make some workarounds due to the nvidia drivers, but `kinect2_bridge` ran without any problems.

Thanks!

orajput, 2015-06-02:
After creating a fresh workspace, and using updated `libfreenect2` and `iai_kinect2` that issue is gone. It could be that `build`/`devel` folders kept some old configurations responsible for such errors (I can't exactly say 'how', though). 

Thanks!

wiedemeyer, 2015-06-02:
If something strange happens, you can simply remove the devel and build folder. You don't need to create a new workspace. `catkin_make clean` might also work.

wiedemeyer, 2015-06-02:
So, if the issue is solved, you can close it.

orajput, 2015-06-02:
Actually I did only delete the `build` and `devel` folders.

closed 2015-06-02 by orajput