code-iai/iai_kinect2 #106: Depth points are missing and frame ids are not matching

UnforgiveNX, 2015-06-02:
Hi there,
 
I've changed the hardware I use for Kinect v2. Now I'm on Nvidia GTX 850M (nvidia-346).
First, when i run 
```
roslaunch kinect2_bridge kinect2_bridge.launch
```
the output is 

```
...
[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started

device serial: 508765442542
device firmware: 4.3.3912.0.7
[Freenect2DeviceImpl] stopping...
[Freenect2DeviceImpl] disabling usb transfer submission...
[Freenect2DeviceImpl] canceling usb transfers...
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[DepthPacketStreamParser::onDataReceived] not all subsequences received 0
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[Freenect2DeviceImpl] stopped

default ir camera parameters: 
fx 365.64, fy 365.64, cx 259.634, cy 205.005
k1 0.0949731, k2 -0.271987, p1 0, p2 0, k3 0.092491

default color camera parameters: 
fx 1081.37, fy 1081.37, cx 959.5, cy 539.5
using sensor defaults for color intrinsic parameters.
using sensor defaults for ir intrinsic parameters.
using defaults for rotation and translation.
using defaults for depth shift.

camera parameters used:
camera matrix color:
[1081.3720703125, 0, 959.5;
  0, 1081.3720703125, 539.5;
  0, 0, 1]
distortion coefficients color:
[0, 0, 0, 0, 0]
camera matrix ir:
[365.6398010253906, 0, 259.6340026855469;
  0, 365.6398010253906, 205.0050964355469;
  0, 0, 1]
distortion coefficients ir:
[0.09497310221195221, -0.2719865143299103, 0, 0, 0.09249096363782883]
rotation:
[1, 0, 0;
  0, 1, 0;
  0, 0, 1]
translation:
[-0.052; 0; 0]
depth shift:
0

[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistrationOpenCL::listDevice]  devices:
  0: GeForce GTX 850M (GPU)[NVIDIA Corporation]
[DepthRegistrationOpenCL::init]  selected device: GeForce GTX 850M (GPU)[NVIDIA Corporation]
[DepthRegistrationOpenCL::listDevice]  devices:
  0: GeForce GTX 850M (GPU)[NVIDIA Corporation]
[DepthRegistrationOpenCL::init]  selected device: GeForce GTX 850M (GPU)[NVIDIA Corporation]
[kinect2_bridge] waiting for clients to connect
...
```

When I want to echo `/kinect2/qhd/points`,

the output goes on like this:

```
[kinect2_bridge] client connected. starting device...
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 01 12 0f 47 4d 41 39 38 33 2e 31    31332e2e2e2e2e2e474d413938332e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
[DepthPacketStreamParser::onDataReceived] not all subsequences received 0
[DepthPacketStreamParser::onDataReceived] not all subsequences received 960
[RgbPacketStreamParser::onDataReceived] packetsize or sequence doesn't match!
[ERROR] [1433250028.432726615]: Depth image frame id [kinect2_ir_optical_frame] doesn't match RGB image frame id [kinect2_rgb_optical_frame]
[kinect2_bridge] depth processing: ~236.271Hz (4.23244ms) publishing rate: ~26.6573Hz
[kinect2_bridge] color processing: ~131.737Hz (7.59088ms) publishing rate: ~23.3251Hz
[OpenCLDepthPacketProcessor] avg. time: 4.39707ms -> ~227.424Hz
[TurboJpegRgbPacketProcessor] avg. time: 16.7564ms -> ~59.6788Hz
[ERROR] [1433250033.463011468]: Depth image frame id [kinect2_ir_optical_frame] doesn't match RGB image frame id [kinect2_rgb_optical_frame]
[kinect2_bridge] depth processing: ~244.612Hz (4.08811ms) publishing rate: ~29.9979Hz
[kinect2_bridge] color processing: ~133.435Hz (7.4943ms) publishing rate: ~29.9979Hz
[OpenCLDepthPacketProcessor] avg. time: 4.23444ms -> ~236.159Hz
[TurboJpegRgbPacketProcessor] avg. time: 16.5009ms -> ~60.6027Hz
[kinect2_bridge] depth processing: ~249.359Hz (4.01028ms) publishing rate: ~29.9896Hz
[kinect2_bridge] color processing: ~130.635Hz (7.65494ms) publishing rate: ~29.9896Hz
[OpenCLDepthPacketProcessor] avg. time: 4.20861ms -> ~237.608Hz
[TurboJpegRgbPacketProcessor] avg. time: 16.4437ms -> ~60.8135Hz
[ERROR] [1433250038.463266946]: Depth image frame id [kinect2_ir_optical_frame] doesn't match RGB image frame id [kinect2_rgb_optical_frame]
[kinect2_bridge] depth processing: ~250.386Hz (3.99384ms) publishing rate: ~29.9956Hz
[kinect2_bridge] color processing: ~131.405Hz (7.61005ms) publishing rate: ~29.9956Hz
[OpenCLDepthPacketProcessor] avg. time: 4.30005ms -> ~232.555Hz
[TurboJpegRgbPacketProcessor] avg. time: 16.492ms -> ~60.6354Hz
[kinect2_bridge] depth processing: ~246.467Hz (4.05733ms) publishing rate: ~29.9966Hz
[kinect2_bridge] color processing: ~129.633Hz (7.71407ms) publishing rate: ~29.9966Hz
```

I get all images (depth, rgb, ir etc.) in rviz but I can not get any message in points topic on terminal or rviz. I'm first curious about mass "[TransferPool::submit] transfer submission disabled!" message which I do not have before. Another point is this error: "[ERROR] [1433250038.463266946]: Depth image frame id [kinect2_ir_optical_frame] doesn't match RGB image frame id [kinect2_rgb_optical_frame]".

Any idea about the problem? I want to get points in PointCloud2 type.
By the way Protonect example works perfect with all configurations.

wiedemeyer, 2015-06-02:
Hi,
I found this problem today too. I already fixed it and will commit the fix in the next hour.

wiedemeyer, 2015-06-02:
should be fixed now.

UnforgiveNX, 2015-06-02:
Hi Thiemo,

Yes it is fixed with your last commit (eac65adedf7a63555e90430cf152a74231b3183f).

Thank you!

closed 2015-06-02 by UnforgiveNX