code-iai/iai_kinect2 #108: [kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz

guanchar, 2015-06-03:
Hello, I've recently updated to the latest code for both libfreenect2 and iai_kinect2, but depth images are no longer working.

Protonect works well, I can see rgb, ir, depth, and registered images, many packets are skipped, but depth processing is happening: 
```
[OpenGLDepthPacketProcessor] avg. time: 43.7642ms -> ~22.8497Hz
[TurboJpegRgbPacketProcessor] avg. time: 24.4506ms -> ~40.8987Hz
[DepthPacketStreamParser::onDataReceived] skipping depth packet
[RgbPacketStreamParser::onDataReceived] skipping rgb packet!
```

However, when I try to run iai_kinect2, depth images are not published (rgb images are could be seen through rviz): 
```
[Freenect2DeviceImpl] started
[RgbPacketStreamParser::onDataReceived] packetsize or sequence doesn't match!
[kinect2_bridge] depth processing: ~0Hz (infms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~16081.7Hz (0.0621823ms) publishing rate: ~22.9742Hz
[TurboJpegRgbPacketProcessor] avg. time: 20.6938ms -> ~48.3237Hz
```

Any help would be greatly appreciated!

yoneken, 2015-06-03:
Hi,

I had exactly the same issue.
Did you check a linked libusb of your kinect2_bridge? 
In my case, Protonect is linked to a special patched libusb, but the iai_kinect2 was linked to a system default one.

You can check the linked library with ldd command.
ldd /path/to/Protonect | grep libusb
ldd /path/to/kinect2_bridge | grep libusb

guanchar, 2015-06-03:
That was it! I had made the modifications to another version of libfreenect2 on my system, not the one I ended up installing. For reference: follow the following section of the readme to change which libusb iai_kinect2 links to!

Replace line 48 to 50 in libfreenect2/examples/protonect/CMakeLists.txt with: 

```
# LibUSB
GET_FILENAME_COMPONENT(LIBUSB_DIR "${MY_DIR}/../../depends/libusb/" REALPATH)

INCLUDE_DIRECTORIES("${LIBUSB_DIR}/include/libusb-1.0/")
LINK_DIRECTORIES("${LIBUSB_DIR}/lib/")
SET(CMAKE_INSTALL_RPATH_USE_LINK_PATH TRUE)
```

Thank you!

closed 2015-06-03 by guanchar