code-iai/iai_kinect2 #109: kinect2_calibration: unknow depth

MarioIannotta, 2015-06-04:
Hi, I'm trying to calibrate the kinect2 following the guide but at the ninth point, when I do

```$ rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 calibrate depth```

I get 

```Unknown argument: depth

/opt/ros/indigo/lib/kinect2_calibration/kinect2_calibration [options]
  mode: 'record' or 'calibrate'
  source: 'color', 'ir', 'sync'
  board:
    'circle<WIDTH>x<HEIGHT>x<SIZE>'  for symmentric cirle grid
    'acircle<WIDTH>x<HEIGHT>x<SIZE>' for asymmentric cirle grid
    'chess<WIDTH>x<HEIGHT>x<SIZE>'   for chessboard pattern
  topics: '-color <TOPIC>' and/or '-ir <TOPIC>'
  output path: '<PATH>'
```

wiedemeyer, 2015-06-04:
Hi,
there is not even one line in the code which outputs `unknown argument`. So you are not using the master branch of this repository. Also how did you installed it in `/opt/ros/indigo/lib/`? Did you used some fork?

MarioIannotta, 2015-06-04:
I have installed this fork https://github.com/ethz-asl/kinect2-ros following the instructions in the readme file

wiedemeyer, 2015-06-04:
That one seems to be outdated. You could install this repository like described in the README, or ask at the other repository for help. I cannot help with issues related to code in other repositories.

MarioIannotta, 2015-06-04:
Ok, thank you!

closed 2015-06-05 by wiedemeyer