code-iai/iai_kinect2 #110: Transform between Camera space and "Pointcloud space"

svope, 2015-06-09:
Hi guys,

what is the transformation between Camera space ( x and y from RGB and z from depth map) and Pointcloud( topic /kinect2/qhd/points )?
From what I see in viewer.cpp the transformation from Camera space to PoinCloud is:

1) computing "lookup table" in void createLookup(size_t width, size_t height)
2) in function void createCloud(...) these equations:
                                               ....code...
                                               itP->z = depthValue;
                                               itP->x = *itX * depthValue;
                                               itP->y = y * depthValue;
                                               ....etc..

I tried to "undo"(reverse) these equations but with no success (I got some weird number intervals exceeding Camera resolution 960x540). Did I solved these "reversed" eq. badly or I'm just doing something totally wrong.

Thank you!

wiedemeyer, 2015-06-10:
Hi,
color, depth and point cloud in the same topic group, for example `qhd`, are in the same space. `x` and `y` image coordinates are the same in all. The color pixel at position `xy` belongs to the depth pixel at `xy` and the point cloud pixel at `xy` is the combination of both.
The formula to compute the y world coordinate is `y_w = (y - cy) / fy * z`, the inverse of it would be `y = (y_w / z_w) * fy + cy`. But it is much simpler to use the `xy` coordinates of the pixel. And if you have just the index of the point cloud pixel just do `x = i % width` and `y = i / width`.

svope, 2015-06-10:
Thank you for your explanation.

I've got one more specific question. What I'm doing is subscribing PointCloud (I'm using older API so "/kinect2/depth_lowres/points"), finding there some "objects" using PCL library. When I find some object I get a 3D point ( basically cloud_pointer->points[index].x, cloud_pointer->points[index].y and cloud_pointer->points[index].z ). I think I can't use that index as you mentioned (x = index % width and y = index / width) because when I'm looking for some object in PointCloud, the algorithm subtract some (already used) points from PointCloud, so there could be some index mismatch and even if there won't be any index mismatch I would like to make a translation from that 3D Point (cloud_pointer->points[index].x, cloud_pointer->points[index].y, cloud_pointer->points[index].z ) to Camera RGB image plane ( X,Y ).
Is that where should I use that formulas?
Y = (y_w / z_w) * fy + cy
X = (x_w / z_w) * fx + cx
where x_w is cloud_pointer->points[index].x, y_w is cloud_pointer->points[index].y ..., fy,fx,cy,cx is from calib_data.yaml (calibration data) or CameraInfo topic.

Is my understanding right? I'm asking because when I use these mentioned formulas the X and Y are often "out of scope" of Camera RGB image plane.

Thank you

wiedemeyer, 2015-06-11:
Yes, it should be like that, if I am not mistaken.

svope, 2015-06-11:
If anyone interested I solved it using image_geometry ROS package:

pseudo code:
image_geometry::PinholeCameraModel cam_model_;
cam_model_.fromCameraInfo(const sensor_msgs::CameraInfoConstPtr cam_infos);
cv::Point3d point_cv(3D Point.x,.y,.z);
cv::Point2d point_image;
point_image = cam_model_.project3dToPixel(point_cv);

Thank you for your help!

wiedemeyer, 2015-06-12:
The formula in `project3dToPixel()` is the same one as above. [See here.](https://github.com/ros-perception/vision_opencv/blob/indigo/image_geometry/src/pinhole_camera_model.cpp#L262-263)
But anyway, good that it's solved.

closed 2015-06-12 by wiedemeyer