code-iai/iai_kinect2 #111: Citation

jeppewalther, 2015-06-11:
How would you like iai_kinect2 to be cited in a thesis and or paper.

wiedemeyer, 2015-06-11:
Yes, I would greatly appreciate a citation.

jeppewalther, 2015-06-11:
Sorry it was a bad formulation. In what format would you like to be cited. Is there a reference paper and what names should i cite.

wiedemeyer, 2015-06-12:
Just created a bibtex entry:

```tex
@misc{iai_kinect2,
  author = {Wiedemeyer, Thiemo},
  title = {{IAI Kinect2}},
  organization = {Institute for Artificial Intelligence},
  address = {University Bremen},
  year = {2014 -- 2015},
  howpublished = {\url{https://github.com/code-iai/iai\_kinect2}},
  note = {Accessed June 12, 2015}
}
```

It may depend on the bibliography style used, but it should result in something like:
```
T. Wiedemeyer, “IAI Kinect2,” https://github.com/code-iai/iai_kinect2,
Institute for Artificial Intelligence, University Bremen, 2014 – 2015,
accessed June 12, 2015.
```

jeppewalther, 2015-06-16:
Thats great, thank you.

closed 2015-06-17 by wiedemeyer