code-iai/iai_kinect2 #112: Installation failure

anshulvj, 2015-06-12:
I am installing iai_kinect2 on ubuntu 14.04. I have successfully installed OpenCL, ROS Hydro, OpenCV, PCL, Eigen3, libfreenect2 in that order. When I run the iai_kinect2 setup the following error occurs and I'm not sure what to do; didn't glean too much info from google, and I re-checked the installation (everything except libturbojpeg0-dev, as I could not find it, but that command is commented out in the libfreenect2 setup). If something is obviously wrong please let me know. I just spent 4 days installing Opencv, opencl, Ros and pcl so I hope nothing is wrong with that :(. Thanks in advance

The entire error I am getting is:

[100%] Building CXX object iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o
In file included from /usr/include/pcl-1.7/pcl/visualization/cloud_viewer.h:39:0,
                 from /home/anshulvj/catkin_ws/src/iai_kinect2/kinect2_viewer/src/viewer.cpp:32:
/usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:283:88: error: macro "PCL_DEPRECATED" requires 2 arguments, but only 1 given
         "addCoordinateSystem (scale, id, viewport) with id a unique string identifier.")
                                                                                        ^
/usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:304:97: error: macro "PCL_DEPRECATED" requires 2 arguments, but only 1 given
         "addCoordinateSystem (scale, x, y, z, id, viewport) with id a unique string identifier.")
                                                                                                 ^
/usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:327:91: error: macro "PCL_DEPRECATED" requires 2 arguments, but only 1 given
         "addCoordinateSystem (scale, t, id, viewport) with id a unique string identifier.")
                                                                                           ^
/usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:374:81: error: macro "PCL_DEPRECATED" requires 2 arguments, but only 1 given
         "addCoordinateSystem (id, viewport) with id a unique string identifier.")
                                                                                 ^
/home/anshulvj/catkin_ws/src/iai_kinect2/kinect2_calibration/src/kinect2_calibration.cpp: In member function ‘void CameraCalibration::calibrateExtrinsics()’:
/home/anshulvj/catkin_ws/src/iai_kinect2/kinect2_calibration/src/kinect2_calibration.cpp:682:56: error: cannot convert ‘const cv::TermCriteria’ to ‘int’ for argument ‘13’ to ‘double cv::stereoCalibrate(cv::InputArrayOfArrays, cv::InputArrayOfArrays, cv::InputArrayOfArrays, cv::InputOutputArray, cv::InputOutputArray, cv::InputOutputArray, cv::InputOutputArray, cv::Size, cv::OutputArray, cv::OutputArray, cv::OutputArray, cv::OutputArray, int, cv::TermCriteria)’
                                 cv::CALIB_FIX_INTRINSIC);
                                                        ^
/home/anshulvj/catkin_ws/src/iai_kinect2/kinect2_calibration/src/kinect2_calibration.cpp: In member function ‘void DepthCalibration::getPlane(size_t, cv::Mat&, double&) const’:
/home/anshulvj/catkin_ws/src/iai_kinect2/kinect2_calibration/src/kinect2_calibration.cpp:969:138: error: ‘ITERATIVE’ is not a member of ‘cv’
     cv::solvePnPRansac(board, points[index], cameraMatrix, distortion, rvec, translation, false, 300, 0.05, board.size(), cv::noArray(), cv::ITERATIVE);
                                                                                                                                          ^
make[2]: *** [iai_kinect2/kinect2_calibration/CMakeFiles/kinect2_calibration.dir/src/kinect2_calibration.cpp.o] Error 1
make[1]: *** [iai_kinect2/kinect2_calibration/CMakeFiles/kinect2_calibration.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:281:9: error: ‘PCL_DEPRECATED’ does not name a type
         PCL_DEPRECATED (
         ^
/usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:302:9: error: ‘PCL_DEPRECATED’ does not name a type
         PCL_DEPRECATED (
         ^
/usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:325:9: error: ‘PCL_DEPRECATED’ does not name a type
         PCL_DEPRECATED (
         ^
/usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:372:9: error: ‘PCL_DEPRECATED’ does not name a type
         PCL_DEPRECATED (
         ^
Linking CXX executable /home/anshulvj/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge
Linking CXX shared library /home/anshulvj/catkin_ws/devel/lib/libkinect2_bridge_nodelet.so
/usr/bin/ld: warning: libopencv_imgproc.so.3.0, needed by /home/anshulvj/catkin_ws/devel/lib/libkinect2_registration.so, may conflict with libopencv_imgproc.so.2.4
/usr/local/lib/libfreenect2.a(turbo_jpeg_rgb_packet_processor.cpp.o): In function `libfreenect2::TurboJpegRgbPacketProcessor::process(libfreenect2::RgbPacket const&)':
/home/anshulvj/libfreenect2/examples/protonect/src/turbo_jpeg_rgb_packet_processor.cpp:118: undefined reference to `tjDecompress2'
/home/anshulvj/libfreenect2/examples/protonect/src/turbo_jpeg_rgb_packet_processor.cpp:129: undefined reference to `tjGetErrorStr'
/usr/local/lib/libfreenect2.a(turbo_jpeg_rgb_packet_processor.cpp.o): In function `~TurboJpegRgbPacketProcessorImpl':
/home/anshulvj/libfreenect2/examples/protonect/src/turbo_jpeg_rgb_packet_processor.cpp:67: undefined reference to `tjDestroy'
/home/anshulvj/libfreenect2/examples/protonect/src/turbo_jpeg_rgb_packet_processor.cpp:69: undefined reference to `tjGetErrorStr'
/usr/local/lib/libfreenect2.a(turbo_jpeg_rgb_packet_processor.cpp.o): In function `libfreenect2::TurboJpegRgbPacketProcessor::TurboJpegRgbPacketProcessor()':
/home/anshulvj/libfreenect2/examples/protonect/src/turbo_jpeg_rgb_packet_processor.cpp:50: undefined reference to `tjInitDecompress'
/home/anshulvj/libfreenect2/examples/protonect/src/turbo_jpeg_rgb_packet_processor.cpp:53: undefined reference to `tjGetErrorStr'
/usr/local/lib/libfreenect2.a(libfreenect2.cpp.o): In function `libfreenect2::Freenect2DeviceImpl::close()':
/home/anshulvj/libfreenect2/examples/protonect/src/libfreenect2.cpp:616: undefined reference to `libusb_close'
/usr/local/lib/libfreenect2.a(libfreenect2.cpp.o): In function `libfreenect2::operator<<(std::ostream&, libfreenect2::PrintBusAndDevice const&)':
/home/anshulvj/libfreenect2/examples/protonect/src/libfreenect2.cpp:109: undefined reference to `libusb_get_device_address'
/home/anshulvj/libfreenect2/examples/protonect/src/libfreenect2.cpp:109: undefined reference to `libusb_get_bus_number'
/usr/local/lib/libfreenect2.a(libfreenect2.cpp.o): In function `libfreenect2::Freenect2DeviceImpl::isSameUsbDevice(libusb_device*)':
/home/anshulvj/libfreenect2/examples/protonect/src/libfreenect2.cpp:352: undefined reference to `libusb_get_bus_number'
/home/anshulvj/libfreenect2/examples/protonect/src/libfreenect2.cpp:353: undefined reference to `libusb_get_device_address'
/home/anshulvj/libfreenect2/examples/protonect/src/libfreenect2.cpp:355: undefined reference to `libusb_get_bus_number'
/home/anshulvj/libfreenect2/examples/protonect/src/libfreenect2.cpp:356: undefined reference to `libusb_get_device_address'
/usr/local/lib/libfreenect2.a(libfreenect2.cpp.o): In function `libfreenect2::Freenect2Impl::clearDeviceEnumeration()':
/home/anshulvj/libfreenect2/examples/protonect/src/libfreenect2.cpp:217: undefined reference to `libusb_unref_device'
/usr/local/lib/libfreenect2.a(libfreenect2.cpp.o): In function `libfreenect2::Freenect2::openDevice(int, libfreenect2::PacketPipeline const*, bool)':
/home/anshulvj/libfreenect2/examples/protonect/src/libfreenect2.cpp:694: undefined reference to `libusb_open'
/home/anshulvj/libfreenect2/examples/protonect/src/libfreenect2.cpp:704: undefined reference to `libusb_reset_device'
/home/anshulvj/libfreenect2/examples/protonect/src/libfreenect2.cpp:716: undefined reference to `libusb_close'
[... 89 lines cut ...]
/home/anshulvj/libfreenect2/examples/protonect/src/command_transaction.cpp:138: undefined reference to `libusb_error_name'
/usr/local/lib/libfreenect2.a(command_transaction.cpp.o): In function `libfreenect2::protocol::CommandTransaction::receive(libfreenect2::protocol::CommandTransaction::Result&)':
/home/anshulvj/libfreenect2/examples/protonect/src/command_transaction.cpp:156: undefined reference to `libusb_bulk_transfer'
/home/anshulvj/libfreenect2/examples/protonect/src/command_transaction.cpp:160: undefined reference to `libusb_error_name'
[... cut ...]
collect2: error: ld returned 1 exit status
make[2]: *** [/home/anshulvj/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge] Error 1
make[1]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/all] Error 2
[100%] Built target kinect2_bridge_nodelet
make[2]: *** [iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/src/viewer.cpp.o] Error 1
make[1]: *** [iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
anshulvj@anshulspc:~/catkin_ws$

wiedemeyer, 2015-06-15:
Hi,
just install OpenCV and PCL from the official Ubuntu repositories. It will not work with OpenCV 3.0.0.
The rosdep error is normal. It is explained in the FAQ section of the [README](..).

anshulvj, 2015-06-15:
Thanks Thiemo,

I only replaced Opencv3.0 with opencv 2.4.9 and it compiled successfully. Before that I also recompiled PCL from sources instead of using binaries but I don't know whether that made a difference.

Also, when I run kinect2_bridge I get the following error:

[ERROR] [1434379251.163641822]: [registerPublisher] Failed to contact master at [localhost:11311].  Retrying...

I could not see anyone else getting this error in issues.

anshulvj, 2015-06-15:
Ok. This was a classical noob error: I ran roscore in another terminal and then it worked. However, when I run kinect_bridge it gives out all the output with params etc, then waits saying "[kinect2_bridge] waiting for clients to connect". It's waiting for a while now. Any thoughts on how I should proceed?

wiedemeyer, 2015-06-15:
It will wait for clients to connect. So as soon as you start listening to a topic, it will start processing data.

anshulvj, 2015-06-15:
Meaning what should I do on the side while it's waiting for clients to connect, or starting receiver? Should I be opening another window and running another command or something?

anshulvj, 2015-06-15:
Ok. I started roscore in one terminal, kinect2_bridge in another terminal, and kinect2_viewer in third terminal. This time it showed a window were it was apparently trying to display something. It was hard to see as it kept on refreshing (mostly all I could see was something magenta colored on the top half of the image) and the fps kept bouncing around

wiedemeyer, 2015-06-16:
Please make sure to use the latest version of libfreenect2 and iai_kinect2, because there were changes that made the older versions incompatible to the newer ones.

anshulvj, 2015-06-16:
I think I'm using the latest version since I re-cloned both of them in the last couple of days.

wiedemeyer, 2015-06-17:
just make sure, because there were some updates in the last couple of days.

anshulvj, 2015-06-17:
Thanks, I'll do that.

wiedemeyer, 2015-06-22:
Did it work?

anshulvj, 2015-06-24:
Sorry for the late reply, I was on vacation. Actually I did not try installing it later coz I figured out the equations for getting x,y,z locations from depth image, in the registration.cpp. So now I'm just using libfreenect2. Thanks for your suggestions though.

I had another question though. I haven't explored it yet, but maybe you might have an answer: is there any way to suspend/pause the kinect streaming process? I did some stress testing today and my embedded machine-being the low power machine it is- gets wedged if it runs protonect for too long. I wanted to know if there was anything simple I could do to pause the process and resume it quickly, to shed some load?

wiedemeyer, 2015-06-26:
Right now, if no topics are subscribed, the kinect2_bridge shuts down the device and the processing. So simply unsubscribe if you don't need data. Also you could limit the fps to a lower value.
I saw your issue https://github.com/OpenKinect/libfreenect2/issues/307 there. If there is a pause in libfreenect2, I could replace the shutdown by a pause, so that it can start processing more quickly.

anshulvj, 2015-06-26:
I have created a pause() function as suggested by @floe, in https://github.com/OpenKinect/libfreenect2/issues/307, and it seems to be working. I.e. I pause for a few seconds, after a fixed number of frames have been accessed, then reset the frame counter and continue again (by calling SetStreamEnabledCommand..).

closed 2015-10-14 by wiedemeyer