code-iai/iai_kinect2 #113: Point cloud data in rviz?

sj312931, 2015-06-16:
Hello,

I am trying to obtain point cloud data from Kinect2 in ROS. I am able to obtain rgb,ir,depth images when I run ./Protenect cpu. But only get rgb image when I run ./Protonect {cl/gl}. 

Also, the CPU based depth processing works fine for me. I see the cloud viewer and image viewer windows successfully when I do : rosrun kinect2_bridge kinect2_bridge _reg_method:=cpu _depth_method:=cpu. But it doesnt work for me if I try to use OpenCL or OpenGL for depth processing.   

I wish to obtain the point cloud data from kinect2 in Rviz but I dont see any topics under the pointcloud2 display. When I run -  roslaunch kinect2_bridge kinect2_bridge.launch _reg_method:=cpu _depth_method:=cpu I get this: 

[OpenCLDepthPacketProcessor::init] ERROR: clGetDeviceIDs(-1)
[FATAL] [1434472306.201188965]: Service call failed!
[FATAL] [1434472306.201206386]: Service call failed!
[FATAL] [1434472306.201318588]: Service call failed!
[FATAL] [1434472306.201526654]: Service call failed!
[kinect2-1] process has died [pid 9203, exit code -11, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=kinect2 __log:=/home/anarang/.ros/log/a115cd66-1442-11e5-abb9-34e6d73a9623/kinect2-1.log].
log file: /home/anarang/.ros/log/a115cd66-1442-11e5-abb9-34e6d73a9623/kinect2-1*.log
[kinect2_points_xyzrgb_qhd-4] process has died [pid 9253, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/qhd/camera_info rgb/image_rect_color:=kinect2/qhd/image_color_rect depth_registered/image_rect:=kinect2/qhd/image_depth_rect depth_registered/points:=kinect2/qhd/points __name:=kinect2_points_xyzrgb_qhd __log:=/home/anarang/.ros/log/a115cd66-1442-11e5-abb9-34e6d73a9623/kinect2_points_xyzrgb_qhd-4.log].
log file: /home/anarang/.ros/log/a115cd66-1442-11e5-abb9-34e6d73a9623/kinect2_points_xyzrgb_qhd-4*.log
[kinect2_points_xyzrgb_qhd-4] restarting process
process[kinect2_points_xyzrgb_qhd-4]: started with pid [9337]
[kinect2_bridge-2] process has died [pid 9223, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/anarang/.ros/log/a115cd66-1442-11e5-abb9-34e6d73a9623/kinect2_bridge-2.log].
log file: /home/anarang/.ros/log/a115cd66-1442-11e5-abb9-34e6d73a9623/kinect2_bridge-2*.log
[kinect2_points_xyzrgb_sd-3] process has died [pid 9239, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/sd/camera_info rgb/image_rect_color:=kinect2/sd/image_color_rect depth_registered/image_rect:=kinect2/sd/image_depth_rect depth_registered/points:=kinect2/sd/points __name:=kinect2_points_xyzrgb_sd __log:=/home/anarang/.ros/log/a115cd66-1442-11e5-abb9-34e6d73a9623/kinect2_points_xyzrgb_sd-3.log].
log file: /home/anarang/.ros/log/a115cd66-1442-11e5-abb9-34e6d73a9623/kinect2_points_xyzrgb_sd-3*.log
[kinect2_points_xyzrgb_hd-5] process has died [pid 9300, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/hd/camera_info rgb/image_rect_color:=kinect2/hd/image_color_rect depth_registered/image_rect:=kinect2/hd/image_depth_rect depth_registered/points:=kinect2/hd/points __name:=kinect2_points_xyzrgb_hd __log:=/home/anarang/.ros/log/a115cd66-1442-11e5-abb9-34e6d73a9623/kinect2_points_xyzrgb_hd-5.log].
log file: /home/anarang/.ros/log/a115cd66-1442-11e5-abb9-34e6d73a9623/kinect2_points_xyzrgb_hd-5*.log
[kinect2_bridge-2] restarting process
process[kinect2_bridge-2]: started with pid [9354]
[kinect2_points_xyzrgb_sd-3] restarting process
process[kinect2_points_xyzrgb_sd-3]: started with pid [9355]
[kinect2_points_xyzrgb_hd-5] restarting process
process[kinect2_points_xyzrgb_hd-5]: started with pid [9356]
[ INFO] [1434472306.562822672]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1434472306.570050142]: waitForService: Service [/kinect2/load_nodelet] could not connect to host [anarang-Latitude-E7440:41752], waiting...

Can somebody please help me out? 

Thanks

wiedemeyer, 2015-06-17:
Hi,
for the launch file you don't need the `_` in front of the parameter name, that is just needed for the node. so instead of `_param:=...` use `param:=...`. You can also edit the default parameters in the launch file itself, it is a simple xml text file.

sj312931, 2015-06-17:
Thanks for the help. I have the kinect2 working now. I just realized that I didn't have the correct version of Beignet. Now, I have Beignet 1.0 and it works.

closed 2015-06-17 by sj312931

lucbettaieb, 2015-11-02:
I'm getting a very similar error

rosrun output:
```
$ rosrun kinect2_bridge kinect2_bridge depth_method:=opencl
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/luc/indigo/src/cwru/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: GeForce GTX 645 (GPU)[NVIDIA Corporation]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: GeForce GTX 645 (GPU)[NVIDIA Corporation]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Warning] [OpenCLDepthPacketProcessorImpl] Your ocl-icd has deadlock bugs. Update to 2.2.4+ is recommended.
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:7 with serial 003668550247
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 003668550247 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_IO Input/Output Error
[Error] [IrCameraTables] 217088 pixels in x/ztable have incorrect undistortion.
*** Error in `/home/luc/indigo/devel/lib/kinect2_bridge/kinect2_bridge': double free or corruption (top): 0x000000000262ad10 ***
Aborted (core dumped)
```

roslaunch output
```
$ roslaunch kinect2_bridge kinect2_bridge.launch 
... logging to /home/luc/.ros/log/803bacf2-8187-11e5-b082-989096a2efa9/roslaunch-atlas4-6150.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://atlas4:37087/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/luc/indigo/...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
[... 76 lines cut ...]
[FATAL] [1446494105.136915360]: Service call failed!
[FATAL] [1446494105.136915388]: Service call failed!
[FATAL] [1446494105.138616535]: Service call failed!
[kinect2-1] process has died [pid 6159, exit code -11, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=kinect2 __log:=/home/luc/.ros/log/803bacf2-8187-11e5-b082-989096a2efa9/kinect2-1.log].
log file: /home/luc/.ros/log/803bacf2-8187-11e5-b082-989096a2efa9/kinect2-1*.log
[kinect2_bridge-2] process has died [pid 6174, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/luc/.ros/log/803bacf2-8187-11e5-b082-989096a2efa9/kinect2_bridge-2.log].
log file: /home/luc/.ros/log/803bacf2-8187-11e5-b082-989096a2efa9/kinect2_bridge-2*.log
[kinect2_points_xyzrgb_qhd-4] process has died [pid 6189, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/qhd/camera_info rgb/image_rect_color:=kinect2/qhd/image_color_rect depth_registered/image_rect:=kinect2/qhd/image_depth_rect depth_registered/points:=kinect2/qhd/points __name:=kinect2_points_xyzrgb_qhd __log:=/home/luc/.ros/log/803bacf2-8187-11e5-b082-989096a2efa9/kinect2_points_xyzrgb_qhd-4.log].
log file: /home/luc/.ros/log/803bacf2-8187-11e5-b082-989096a2efa9/kinect2_points_xyzrgb_qhd-4*.log
[kinect2_points_xyzrgb_hd-5] process has died [pid 6194, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/hd/camera_info rgb/image_rect_color:=kinect2/hd/image_color_rect depth_registered/image_rect:=kinect2/hd/image_depth_rect depth_registered/points:=kinect2/hd/points __name:=kinect2_points_xyzrgb_hd __log:=/home/luc/.ros/log/803bacf2-8187-11e5-b082-989096a2efa9/kinect2_points_xyzrgb_hd-5.log].
log file: /home/luc/.ros/log/803bacf2-8187-11e5-b082-989096a2efa9/kinect2_points_xyzrgb_hd-5*.log
[kinect2_bridge-2] restarting process
process[kinect2_bridge-2]: started with pid [6214]
[kinect2_points_xyzrgb_qhd-4] restarting process
process[kinect2_points_xyzrgb_qhd-4]: started with pid [6215]
[kinect2_points_xyzrgb_hd-5] restarting process
process[kinect2_points_xyzrgb_hd-5]: started with pid [6216]
[ INFO] [1446494105.357231459]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1446494105.369090102]: waitForService: Service [/kinect2/load_nodelet] could not connect to host [atlas4:59177], waiting...
[... cut ...] a_info rgb/image_rect_color:=kinect2/sd/image_color_rect depth_registered/image_rect:=kinect2/sd/image_depth_rect depth_registered/points:=kinect2/sd/points __name:=kinect2_points_xyzrgb_sd __log:=/home/luc/.ros/log/803bacf2-8187-11e5-b082-989096a2efa9/kinect2_points_xyzrgb_sd-3.log].
log file: /home/luc/.ros/log/803bacf2-8187-11e5-b082-989096a2efa9/kinect2_points_xyzrgb_sd-3*.log
[kinect2_points_xyzrgb_sd-3] restarting process
process[kinect2_points_xyzrgb_sd-3]: started with pid [6229]
```
```

vishu2287, 2015-11-03:
hello @lucbettaieb 
Are you able to run ./Protonect without errors?
I see its usb error which you might have not shutdown ./Protonect properly once used. or you must me running it parallel. 
Connect your sensor properly and restart pc. Now run kinect2_bridge properly.

eog-2017, 2016-12-26:
ss