code-iai/iai_kinect2 #114: kinect2_registration has no executable nodelets

affan91, 2015-06-22:
Hi,

My issue is related to #54. I noticed that there was a kinect2_registration node made just for registration. However I cannot access any of the nodelets under kinect2_registration.In other words there is no nodelet I can type after entering rosrun kinect2_registration.  I do see files in kinect2_registration/src but they are not executable. All other nodelets(kinect2_bridge,kinect2_calibration,kinect2_viewer) seem to work fine.I cloned the most up to date repository and noticed that the kinect2_registration was updated a few days ago. Is the kinect2_registration usable?

My other question is that if there's a way to record the rgb and depth images, process it offline and publish the registered pointcloud through /kinect2/qhd/points. I know how to record the depth and rgb topic using rosbag, however I do not know how to process it offline and publish it though /kinect2/qhd/points. Perhaps this is related to kinect2_registration? Any help in this matter will be greatly appreciated

wiedemeyer, 2015-06-23:
kinect2_registration is just a library that other programs can use, there is just a `.so` file, no nodes or executables.
You can simply record the camera_info, depth and color image to a bag file. And the replay the bag file and launch a `point_cloud_xyzrgb` nodelet from the [depth_image_proc package](http://wiki.ros.org/depth_image_proc) to create the point cloud offline. Or the simplest solution would be to just record the point clouds directly to a bag file and use those later.

affan91, 2015-06-23:
I see. Yes, that clears up a lot of things. I'll have a look into depth_image_proc package then. As for now, the issue can be closed. Thank you.

closed 2015-06-23 by affan91