code-iai/iai_kinect2 #115: kinect2_viewer: undefined reference OpenCV

iamwolf, 2015-06-29:
Hey! 
Thanks for writing this great library!

I compiled OpenCV **from source** and pkg-config says it's available:
```
-I/usr/local/include/opencv -I/usr/local/include  /usr/local/lib/libopencv_calib3d.so /usr/local/lib/libopencv_contrib.so /usr/local/lib/libopencv_core.so /usr/local/lib/libopencv_features2d.so /usr/local/lib/libopencv_flann.so /usr/local/lib/libopencv_gpu.so /usr/local/lib/libopencv_highgui.so /usr/local/lib/libopencv_imgproc.so /usr/local/lib/libopencv_legacy.so /usr/local/lib/libopencv_ml.so /usr/local/lib/libopencv_nonfree.so /usr/local/lib/libopencv_objdetect.so /usr/local/lib/libopencv_ocl.so /usr/local/lib/libopencv_photo.so /usr/local/lib/libopencv_stitching.so /usr/local/lib/libopencv_superres.so /usr/local/lib/libopencv_ts.a /usr/local/lib/libopencv_video.so /usr/local/lib/libopencv_videostab.so /usr/local/lib/libopencv_viz.so /usr/lib/x86_64-linux-gnu/libXext.so /usr/lib/x86_64-linux-gnu/libX11.so /usr/lib/x86_64-linux-gnu/libICE.so /usr/lib/x86_64-linux-gnu/libSM.so /usr/lib/x86_64-linux-gnu/libGL.so /usr/lib/x86_64-linux-gnu/libGLU.so -lcufft -lnpps -lnppi -lnppc -lcudart -ltbb -lrt -lpthread -lm -ldl  
```

However, when compiling iai_kinect2:
```
/catkin_ws/src/iai_kinect2/kinect2_viewer/src/viewer.cpp:380: undefined reference to `cv::_OutputArray::~_OutputArray()'
```
(Full output https://gist.github.com/iamwolf/b8c3d5fdc2d337a0b982)

When deleting the kinect2_viewer folder and compiling, all the other packages compile and the bridge can be started (OpenCL etc is working as shown by protonect, but I do not see neither depth nor tf - however, this seems to be another issue and will open another ticket for this).

I am on the following system:
- Ubuntu 14.04 LTS, ``Linux MindyIndigo 3.13.0-32-generic #57-Ubuntu SMP Tue Jul 15 03:51:08 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux`` (I have seen you recommend -35, therefore, is -32 too old?)
- CUDA5.5, nvidia-346 and opencl-headers as recommended

Thanks a lot in advance for your time and assistance :)

wiedemeyer, 2015-06-30:
I never seen this error before. Both packages bridge and viewer are using the same CMake code to find OpenCV and to link to it. I don't know why it should behave differently. Try to remove the `devel` and `build` folder and do a clean new build. You could also check if the values in `${OpenCV_INCLUDE_DIR}` and `${OpenCV_LIBRARIES}` are correct.
Also searching for ``undefined reference to `cv::_InputArray::~_InputArray()'`` gives a lot of results, maybe you will find a solution there.

iamwolf, 2015-06-30:
I changed workstations (running same set of software as the other but with newer kernel), updated all the dependencies there and it compiled fine (OpenCL and OpenGL are surprisingly not working on that other workstation - but I can fix that with the advice posted on other issues). I now have depth information etc. - however the images are flipped vertically ?!

Closing - thanks a lot for your advice.

closed 2015-06-30 by iamwolf

wiedemeyer, 2015-06-30:
The images from libfreenect2 are flipped (like looking into a mirror), in kinect2_bridge the images get flipped back to normal.

iamwolf, 2015-06-30:
Amazing - it's a high load on the GPU and network, but the results are amazing :+1: Thanks so much for this driver/bridge!

wiedemeyer, 2015-06-30:
If you use compressed topics the network load should get lower, while the CPU load gets higher. Always depends on where the bottleneck is.