code-iai/iai_kinect2 #116: no depth, IR data & tf information

iamwolf, 2015-06-29:
I followed the issues in libfreenect2 and here regarding not receiving any depth information (nothing published on the topics), as well as no tf transforms.

## tf transforms
I am trying to set my fixed frame in rviz to ``kinect2_rgb_optical_frame`` and it can't be found. Using the tf utils to see all frames, no frames are published at all.

## no depth data
I am starting the kinect2_bridge with the launch file and have tried all methods - cpu, opencl, opengl. The output of the full run is here: https://gist.github.com/iamwolf/3c85fa6cddf8e99db151
The output for Protonect is similar (no mentioning of depth whatsoever)

Note that only the [TurboJpeg...] is being published and no reference to depth being made (in contrast to other logs that I've read for people with no-depth-issues, e.g. https://github.com/OpenKinect/libfreenect2/issues/248).

When running rostopic echo on any of the depth or IR topics, there is no data being published.

Here's the ``dmesg`` output upon plugging the kinect 2 in:
```
[ 4727.967811] usb 3-2: new high-speed USB device number 6 using xhci_hcd
[ 4727.986465] usb 3-2: New USB device found, idVendor=045e, idProduct=02d9
[ 4727.986470] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 4727.986473] usb 3-2: Product: NuiSensor Adaptor      
[ 4727.986475] usb 3-2: Manufacturer: Microsoft Corporation  
[ 4727.986880] hub 3-2:1.0: USB hub found
[ 4727.987116] hub 3-2:1.0: 1 port detected
[ 4728.429524] usb 4-2: new SuperSpeed USB device number 6 using xhci_hcd
```
The port it's being plugged into is marked as a USB3 port.

Which other information could be useful in tracing this error?
I am not seeing many of the error messages that others have posted for the same problem (I see neither the TransmitPool error, nor the registration errors - the data simply appears not to be there at all).

Thanks a lot for your assistance :)

System Information (same as for #115):
- Ubuntu 14.04 LTS, Linux MindyIndigo 3.13.0-32-generic #57-Ubuntu SMP Tue Jul 15 03:51:08 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux (I have seen you recommend -35, therefore, is -32 too old?)
- CUDA5.5, nvidia-346 and opencl-headers as recommended

wiedemeyer, 2015-06-30:
To get TF published you need to enable it, either change the [default value](https://github.com/code-iai/iai_kinect2/blob/master/kinect2_bridge/launch/kinect2_bridge.launch#L5) in the launch file to `true`, or give it as an parameter `publish_tf:=true` (for the launchfile) or `_publish_tf:=true` for executing the node directly.
The other problem is probably related to USB. It is not receiving anything from the sensor. It could be, that your USB controller is not supported. What controller are you using? If I remember correctly you need at least 3.13.0-35 for the Intel USB 3.0 controllers to work. You could also try to switch to a newer kernel. 3.16 (`linux-headers-generic-lts-utopic`) and 3.19 (`linux-headers-generic-lts-vivid`) are working fine for me.

iamwolf, 2015-06-30:
Thanks for the comments - I followed them. As with the other issue, this problem could be resolved using another/second workstation. I assume you're spot on and it might be an issue with either the Kernel being too old or the specific USB3.0 controllers we use have in the other workstation. 

``lspci`` reveals (shortened):
```
00:14.0 USB controller: Intel Corporation 7 Series/C210 Series Chipset Family USB xHCI Host Controller (rev 04)
00:1a.0 USB controller: Intel Corporation 7 Series/C210 Series Chipset Family USB Enhanced Host Controller #2 (rev 04)
00:1d.0 USB controller: Intel Corporation 7 Series/C210 Series Chipset Family USB Enhanced Host Controller #1 (rev 04)
```

I'll close this since it's likely to do with our specific (old) kernel or the hardware in that machine and could not be reproduced on another workstation.

Thanks for the quick support and help :)

closed 2015-06-30 by iamwolf