code-iai/iai_kinect2 #117: SD Stream not aligned

matanyahorowitz, 2015-07-01:
Hello, really enjoying the Kinect2_Bridge. It's been very helpful for us.

Using the latest build, when using the topics

/kinect2/sd/image_depth_rect
/kinect2/sd/image_color_rect

The color frame isn't correctly aligned with the depth and doesn't seem to use our calibration parameters. The hd and qhd streams are perfectly aligned.

Is anyone else having the same issue?

wiedemeyer, 2015-07-02:
Hi,
well, the qhd and hd streams are using a different registration method. For those streams a depth image to color frame registration is used. Which only works if the device had been calibrated.
For the sd stream a color image to depth frame registration is used. This registration uses some parameters provided by the sensor itself, so it is always working, even if the device has not been calibrated yet.

As soon as you calibrate your device, kinect2_birdge will overwrite the default camera intrinsics. This is needed for the qhd and hd registration, but maybe the changed intrinsics make the sd registration worse. I don't know if that is the case or not, but could you run `rosrun kinect2_bridge kinect2_bridge _calib_path:=/some/directory` and change the calibration data path to some other directory and check if the sd registration looks better?

dbworth, 2015-07-13:
I have noticed the same problem:

When using my own calibration parameters in the _calib_path, the HD & QHD streams look good but the SD depth image is badly calibration.

To use the SD depth image I need to remove my own calibration parameters.

I'm happy use only one or the other, but if the HD and SD images are used together then the extrinsics would need to be calibrated into one kinect2 reference frame?

wiedemeyer, 2015-07-13:
Ok, I will change it, so that sd keeps using the parameters from the device instead of the calibrated ones.

wiedemeyer, 2015-08-06:
It should be fixed with commit 90ddda0ca85802e416127bc404cadca37ed8326b.

closed 2015-08-13 by wiedemeyer