code-iai/iai_kinect2 #119: Fail of initialize() is not obversely

MichaelKorn, 2015-07-08:
If initialize() fails the program still seems to run and an issue is not obversely.
e.g. with reg_method=opengl we get false in
https://github.com/code-iai/iai_kinect2/blob/master/kinect2_bridge/src/kinect2_bridge.cpp#L281 .
Because of this no threads for topic publishing are created. But images are grabbed and the program seems to work. The message about the unknown registration method gets lost because of the following terminal outputs (We get a lot of lines due to grabbing states.).

wiedemeyer, 2015-07-09:
Thanks for the hint. I just fixed it with commit 42787f9bdd7ac04668f8ff85f3209059fca085d2.
It should stop correctly on an initialization error now.

lakehanne, 2015-07-09:
What is the correct namespace to use for the kinect2 camera? I saw
`Kinect2` in kinect2_bridge definitions but it doesn't appear to work with
this [cob_people_detection](
https://github.com/ipa-rmb/cob_people_perception/blob/indigo_dev/cob_people_detection/ros/launch/people_detection.launch#L5)
.

wiedemeyer, 2015-07-09:
@lakehanne Your issue has nothing in common with the issue of this thread. Next time please open a new issue, if it does not match any other issue topics.

And a short answer to your question: I think you just have to change those parameter: https://github.com/ipa-rmb/cob_people_perception/blob/indigo_dev/cob_people_detection/ros/launch/people_detection.launch#L5-7

lakehanne, 2015-07-09:
Sorry for the inappropriate question.

wiedemeyer, 2015-07-10:
I don't know that package, but I would guess you have to use the topics with the same resolution, so either `sd`, `qhd` or `hd`, like:
```
<arg name="camera_namespace" default="kinect2_bridge"/>    
<arg name="colorimage_in_topic" default="/$(arg camera_namespace)/sd/image_color_rect"/> 
<arg name="pointcloud_rgb_in_topic" default="/$(arg camera_namespace)/sd/points"/>
```
But you may want to ask the maintainers of that package, they should know better.

wiedemeyer, 2015-07-10:
@MichaelKorn did the last commit fixed the issue for you?

MichaelKorn, 2015-07-10:
@wiedemeyer yes, it works. THX

closed 2015-07-10 by MichaelKorn

lakehanne, 2015-07-11:
Thanks Thiemo.