code-iai/iai_kinect2 #120: Pointcloud topic not found

w4zir, 2015-07-13:
I have installed everything according to the installation guide and then performed calibration successfully. Everything is fine except the message continuous stream of "skip depth packet" while running kinect2_bridge. I can also view pointcloud frame using provided viewer but I don't see any pointcloud topic being published. So I have two questions:
1) Is it ok if the bridge is continuously publishing  "skip depth packet" and sometimes  "skip rgb packet" ?
2) I want the data in ROS::pointcloud format so what should I do ?

wiedemeyer, 2015-07-13:
"skipping ... packet" means that your CPU is too busy. You should try to use the OpenCL or OpenGL depth processor from libfreenect2.
To get point clouds published you have to launch kinect2_bridge with the provided launch file.

w4zir, 2015-07-13:
Thanks found the pointcloud topic. Here is output from the launch. Should I select AMD A10 instead ?

[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistrationOpenCL::listDevice]  devices:
  0: Hainan (GPU)[Advanced Micro Devices, Inc.]
  1: AMD A10-5745M APU with Radeon(tm) HD Graphics (CPU)[AuthenticAMD]
[DepthRegistrationOpenCL::init]  selected device: Hainan (GPU)[Advanced Micro Devices, Inc.]
[DepthRegistrationOpenCL::listDevice]  devices:
  0: Hainan (GPU)[Advanced Micro Devices, Inc.]
  1: AMD A10-5745M APU with Radeon(tm) HD Graphics (CPU)[AuthenticAMD]
[DepthRegistrationOpenCL::init]  selected device: Hainan (GPU)[Advanced Micro Devices, Inc.]

wiedemeyer, 2015-07-13:
The Hainan is probably the dedicated GPU right, so that one should be faster, but you could try.
While executing kinect2_bridge, please take a look at the cpu usage with for example `htop`.
Also running kinect2_bridge and the viewer on the same system is demanding, to check the performance please only run kinect2_bridge and then listen to topics with `rostopic hz /topic/name`. For example:
```
rosrun kinect2_bridge kinect2_bridge
rostopic hz /kinect2/hd/image_color_rect
rostopic hz /kinect2/hd/image_depth_rect
```
If you don't need 30 fps, you can also limit the fps with `fps_limit:=5`, which should reduce the system load a bit.

w4zir, 2015-07-22:
I tried fps_limit:=5 but its still skipping packets. Output is fine as it does capture some packets but after running for a few seconds it fails with error.
failed to submit transfer: LIBUSB_ERROR_NO_DEVICE
and
depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz
color processing: ~-nanHz (-nanms) publishing rate: ~0Hz

wiedemeyer, 2015-07-29:
This is probably an issue related to the usb controller. I don't know how to solve this issue. You might want to ask at the libfreenect2 issues. You could try to use a newer kernel and/or a newer libusb version, but I am not sure if it fixes it. If you have the change to change use a different USB controller, it might help too, normally NEC/Renesas and Intel usb 3.0 controllers are working well.

kerolos, 2015-10-10:
How can i change [DepthRegistration::New] Using CPU registration method!   to   [DepthRegistration::New]Using OpenCL registration method! ?
 because i am always having ( "skip depth packet" and "skip rgb packet")!!

wiedemeyer, 2015-10-14:
`rosrun kinect2_bridge kinect2_bridge _depth_method:=opencl _reg_method:=opencl`, next time please look into the documentation first.

closed 2015-10-14 by wiedemeyer