code-iai/iai_kinect2 #121: Warped pointcloud not sure if calibration related

robinJKU, 2015-07-13:
Hello,

I have a strange issue regarding my kinect2, when facing down about 45° at a table I get a warped pointcloud, it looks pretty much the same with and without calibration.
I calibrated the kinect using a checker board with nearly 200 images.

Setup:
[image]
Pointcloud:
[image]

All other walls seem fine and the table is crucial for our demo, any information would be appreciated.

wiedemeyer, 2015-07-13:
I would guess that's exactly one of the down sides of the new Kinect and the ToF. Corners are rounded and at certain angles the measurements are incorrect. And depth measurements are also dependent on the material of the surface of objects. If you move the sensor further down, so the angle between sensor and plane gets smaller, you will probably end up with no depth values at all.
I think this is nothing that I could solve in kinect2_bridge, this (if possible) should to be addressed in the libfreenect2 driver. Maybe you can also post this on the libfreenect2 issues.
Is it possible to connect your Kinect to a Windows machine with the official SDK? I would really like to see, if it shows the same problems.

robinJKU, 2015-07-13:
Thank you for the fast reply, I will try to get a windows PC with a sufficient graphics cards connected to our kinect (the SDK needs DirectX 11 and Windows 8 if I recall correctly).
The problem is definitely the angle between the kinect and the surface, I will try some other configuration and open an issue in libfreenect2.

robinJKU, 2015-07-14:
I added 2 Screenshots from the Kinect SDK, looks like it is a ToF problem with angled surfaces.

jeppewalther, 2015-07-14:
Its is caused by multi path effects. When light from the sensor is reflected and in turn returned from a secondary surface it causes distortions. In the image below we see two setups one with a white table surface (left) and one with a black table (right)

[image]

lakehanne, 2015-07-14:
Jepp, your issue seems more a matter of background lighting. In the presence of background light, the memory elements in the sensor receive an additional part of the signal. This would disturb the distance measurement.

jeppewalther, 2015-07-15:
I am pretty sure it is not background light from another source than the sensor. If i place a small black plate just in front of the cube, issues with curved surfaces will disappear. It is also dependent on the angle between surface and sensor. It is correct that additional parts of the IR light is distorting the signal. However the origin of that light is from the sensors IR lamps.

robinJKU, 2015-07-28:
By increasing the angle to the table I got rid of most of the distortions, seems like the kinect works best with perpendicular surfaces, so I will try switching the view angle between table and object recognition for better results.

closed 2015-10-26 by wiedemeyer