code-iai/iai_kinect2 #122: Why I can not get point data image  ????

Guhnoo, 2015-07-14:
Hello, I´m ROS beginner and I try to do object detecting and SLAM.  

So, I fristly install code-iai  package ; https://github.com/code-iai/iai_kinect2 
My english is very poor so , I just directly follow install section.
and I think install process is well, because when I try to write    '$ ./Protonect cpu

The five  window (registered, ir, depth, undistorted,rgb) are showed to me.

But,  When I try to write
$ roscore
$ rosrun kinect2_bridge kinect2_bridge
$ rosrun kinect2_viewer kinect2_viewer sd cloud

I am showed only black window.  To check topic  $ rostopic list

## terminal show this 
/kinect2/hd/camera_info
/kinect2/hd/image_color
/kinect2/hd/image_color/compressed
/kinect2/hd/image_color_rect
/kinect2/hd/image_color_rect/compressed
/kinect2/hd/image_depth_rect
/kinect2/hd/image_depth_rect/compressed
/kinect2/hd/image_mono
/kinect2/hd/image_mono/compressed
/kinect2/hd/image_mono_rect
/kinect2/hd/image_mono_rect/compressed
/kinect2/qhd/camera_info
/kinect2/qhd/image_color
/kinect2/qhd/image_color/compressed
/kinect2/qhd/image_color_rect
/kinect2/qhd/image_color_rect/compressed
/kinect2/qhd/image_depth_rect
/kinect2/qhd/image_depth_rect/compressed
/kinect2/qhd/image_mono
/kinect2/qhd/image_mono/compressed
/kinect2/qhd/image_mono_rect
/kinect2/qhd/image_mono_rect/compressed
/kinect2/sd/camera_info
/kinect2/sd/image_color_rect
/kinect2/sd/image_color_rect/compressed
/kinect2/sd/image_depth
/kinect2/sd/image_depth/compressed
/kinect2/sd/image_depth_rect
/kinect2/sd/image_depth_rect/compressed
/kinect2/sd/image_ir
/kinect2/sd/image_ir/compressed
/kinect2/sd/image_ir_rect
/kinect2/sd/image_ir_rect/compressed
/rosout
/rosout_agg
## I think there are any topic about point data . 

So, I installed PCL in http://www.pointclouds.org/downloads/   ; prebuilt binaries  part.
and repeat upper  but  the result is same. 

How can I solve this problem ??? and.  How can I show kinect image date in Rviz ???  

and also do
$ roscore 
$ rosrun kinect2_bridge kinect2_bridge 
$ rosrun rviz rviz 

and I can not get topic  in rviz

wiedemeyer, 2015-07-15:
you don't need the pcl prebuild binaries. PCL is in the default ubuntu repositories and should be installed if you installed `ros-indigo-desktop-full`.
If you want to have point clouds published, launch kinect2_bridge with roslaunch.
If you need tf, then activate it with "publish_tf:=true"

Black windows mean that libfreenect2 is not working well. Please look at the FAQ: https://github.com/code-iai/iai_kinect2#kinect2_bridge-is-not-working--crashing-what-is-wrong

lakehanne, 2015-07-15:
Also if you want to see your topics in a visualization window like `rviz` or `rqt`, in a separate terminal, run `rqt &`, and then open the visualization tab under the plugins menu.

esillen, 2015-07-16:
For the black screen it might be that you do not have a gpu card. If you want to run on the cpu, you can run it like this:
rosrun kinect2_bridge kinect2_bridge _depth_method:=cpu _reg_method:=cpu

For the point cloud, like wiedemeyer says, run the roslaunch file to get the visualization going.
Then you should be able to see the PointCloud2 in rviz.

Guhnoo, 2015-07-18:
Thanks You for everyone!  Actually, I want to execute 2D SLAM with 'kinect xbox one' , 
and now  I understand that I only need 'rgb, depth' information except PCD . 
I checked rgb and depth topic in rviz on the cpu method.  
lol, It's very hard to be beginner. But, I really thank for your advice.

Guhnoo, 2015-07-18:
[image]

Ah,,, Sorry everyone. I have another problem . 
## I checked : 
>> roslaunch kinect2_bridge kinect2_birdge.launch depth_method:=cpu _reg_method:=cpu _publish_tf:=true

>> rosrun kinect2_viewer kinect2_viewer ir both
and I got cloud & image well. 

But, On the rviz (I add pointcloud2 ), I have that error. 
[image]

and the terminal I write ' rosrun rviz rviz'  show me  this

[ INFO] [1437247625.889882740]: rviz version 1.11.7
[ INFO] [1437247625.889953104]: compiled against OGRE version 1.8.1 (Byatis)
[ INFO] [1437247626.006538471]: Stereo is NOT SUPPORTED
[ INFO] [1437247626.006623392]: OpenGl version: 3 (GLSL 1.3).
[ WARN] [1437247670.268670458]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.rviz.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1437247674.863598867]: Messages of type 1 arrived closer (0.299784823) than the lower bound you provided (0.500000000) (will print only once)
[ WARN] [1437247689.017528923]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.rviz.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1437247705.048812486]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.rviz.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1437247749.017652412]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.rviz.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1437247767.185585333]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.rviz.message_notifier] rosconsole logger to DEBUG for more information.
[ INFO] [1437247789.403675047]: Setting pose: -0.086 -0.006 0.987 [frame=map]
[ INFO] [1437247793.213822654]: Setting goal: Frame:map, Position(-0.339, -0.202, 0.000), Orientation(0.000, 0.000, 0.405, 0.914) = Angle: 0.833

[ INFO] [1437247797.429213459]: Setting goal: Frame:map, Position(-0.370, 0.231, 0.000), Orientation(0.000, 0.000, -0.866, 0.501) = Angle: -2.093

[ WARN] [1437247809.018153102]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.rviz.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1437247827.212526313]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.rviz.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1437247869.027672847]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.rviz.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1437247887.239505739]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.rviz.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1437247929.037838067]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.rviz.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1437247947.266504822]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.rviz.message_notifier] rosconsole logger to DEBUG for more information.

Ha,,,,,  T . T

wiedemeyer, 2015-07-29:
using the `cpu` methods is consuming too much processing power. This leads to high delays between depth and color streams. Please try to use OpenCL.

closed 2015-10-14 by wiedemeyer

jesst17, 2016-06-12:
This topic probably dead end, but i need help so sorry,
Guhnoo can you   execute 2D SLAM with 'kinect xbox one'  because i want to do the same thing  but  only things that i can get are,
3dsensor.launch => no device connected, & in rviz i can't get depth map received ( 0 depth map received), but i can see the nodes get stream via rosrun image_view image_view image:=/kinect2/.....
so i am little bit confused

so if you can help me i be really thankful