code-iai/iai_kinect2 #123: Error running kinect2_bridge

renxinhe, 2015-07-15:
After first running `roscore`, I tried`rosrun kinect2_bridge kinect2_bridge`. I got the following error message:
>parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/jim/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
>   worker_threads: 4

>[OpenCLDepthPacketProcessor::init] ERROR: clGetPlatformIDs(-1001)
terminate called after throwing an instance of 'cl::Error'
  what():  clGetPlatformIDs
>Aborted (core dumped)

Is it an issue with my OpenCL setup? I have Ubuntu 14.04, and I don't think I've tried to manually install OpenCL. I assumed whatever default the system has.

wiedemeyer, 2015-07-15:
If you have an nvidia, it could help to run it one time with `sudo`. try `sudo ./protonect cl` and see if it works. After running it one time as root, you can start it without later.

lakehanne, 2015-07-15:
This is an issue I have ran into on my Ubuntu several times and it is mentioned in the beignet documentation, [I think](http://www.freedesktop.org/wiki/Software/Beignet/). 

In my opinion, I think it has to do with the i915 command parser parameter. 

The problem I think is with the self-test of beignet on kernels < 4.x.x. Either upgrade your kernel or  disable the register whitelist with 

```
root# echo 0 > /sys/module/i915/parameters/enable_cmd_parser
```

But make sure you have opencl installed on your system before all this. You can install from this `sudo apt-add-repository ppa:pmjdebruijn/beignet-testing` and then do `apt install beignet-dev`.

mu-777, 2015-07-16:
Hi, renxinhe

I had the same problem and nvidia-modprobe package solved it in my case.
Try
```
$ sudo apt-get install nvidia-modprobe
```

renxinhe, 2015-07-16:
I tried lakehanne's solution. I ran the beignet commands and installed `beignet-dev`. I switched to root with `sudo su` and ran `rosrun kinect2_bridge kinect2_bridge`. This is what I got.

>parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/jim/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
>   worker_threads: 4

>X server found. dri2 connection failed! 
>Trying to open directly...Device open failed

I tried to disable register whitelist, but I don't have an i915 folder under `/sys/module`.

I also tried mu-777's solution by installing the NVidia driver. (I have a NVidia GForce 970). It doesn't seem to fix it.

lakehanne, 2015-07-16:
dri2 connection failed is a normal error when using beignet on nvidia gpu.
I am surprised you have no i915 folder.

What kernel version is your Linux? Can you post the output of `uname -a`.

Also `Device open failed` means you either did not connect your kinect or
your kinect is plugged into a non-usb 3.0 port. What is the output of
`lsusb`.

BTW, you had protonect running with `Protonect cl` as suggested by
@wiedemeyer, yes?

renxinhe, 2015-07-16:
Here's the kernel: `Linux jim-Ubuntu 3.16.0-43-generic #58~14.04.1-Ubuntu SMP Mon Jun 22 10:21:20 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux`

I have USB 3.0 ports on my motherboard as well as case front panel. I tried to connect my Kinect to various USB 3.0 ports, and the output seem to be the same error. I don't think I have other 3.0 devices plugged in. All I have on my USB ports are some peripherals like keyboard, mouse, and webcam.

I've been having issues with Protonect as well. I just got `./bin/Protonect cpu` to work today. The default and `cl` parameter calls still don't work. Here's the thread where I'm trying to get those resolved: https://github.com/OpenKinect/libfreenect2/issues/321

lakehanne, 2015-07-16:
Do you have OpenGL installed? What's the output of `glxinfo | grep "OpenGL version"`

renxinhe, 2015-07-17:
Here you go: `OpenGL version string: 3.0 Mesa 10.3.2`

renxinhe, 2015-07-18:
After updating my Nvidia driver and OpenGL, Protonect seems to run fine. So I tried running kinect2_bridge. This is the message I got: http://pastebin.com/nMVbrj86.

I think I can identify the errors; both have to do with my libfreenect2 setup:
- [ ] sudo
I have to run Protonect with `sudo`, else I'll get the "no device found" error. In order to run kinect2_bridge, I had to `sudo su` into root.

- [ ] usblib LD_LIBRARY_PATH
My usblib from the libfreenect2 package isn't correctly linked for some reason. I've run `depends/install_ubuntu.sh` before, but it didn't change my LD_LIBRARY_PATH.
I had to use `sudo LD_LIBRARY_PATH=/home/jim/UC_Berkeley/Summer_2015/OpenKinect/libfreenect2/depends/libusb/lib ./Protonect` in order to get rid of the error:
>[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO

Both issues are being discussed in this thread: https://github.com/OpenKinect/libfreenect2/issues/321#issuecomment-122178221

lakehanne, 2015-07-18:
Here is the offending line from your pastebin:

>[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO

You still have a libusb error meaning your libusb is probably not linking to Protonect. And you do not have to run Protonect with `sudo`. I think you installed `./install_ubuntu.sh` from your root and that's why you are having to do this root hacks. 

If it is possible for you to delete everything in your libfreenect2 directory, re-clone and build without root permissions, that might be the solution you've been waiting for.

renxinhe, 2015-07-18:
Here're the outputs from various stages of a clean git clone installation: https://github.com/OpenKinect/libfreenect2/issues/321#issuecomment-122603235

For the first time, I can't find `cl.hpp` during `make` for some reason.

renxinhe, 2015-07-19:
Running Protonect with `./bin/Protonect cl`  poses similar error:
>[Freenect2Impl] enumerating devices...
[Freenect2Impl] 14 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:4 with serial 501314743042
[Freenect2Impl] found 1 devices
X server found. dri2 connection failed! 
Trying to open directly...Device open failed

wiedemeyer, 2015-07-29:
the `no device found` looks like you forgot to set the access rights in `/etc/udev/rules.d` correctly. Please look into the README for instructions.

OMARI1988, 2015-10-05:
Hi,

I have the same problem, I get the following error when I try to run kinect2_bridge 

[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistration::New] Using OpenCL registration method!
X server found. dri2 connection failed! 
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
cl_get_gt_device(): error, unknown device: 0
[DepthRegistrationOpenCL::init] ERROR: clGetDeviceIDs(-1)
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
Initialization failed!
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing

any solutions please ?

thanks,

closed 2015-10-14 by wiedemeyer

UniqueFool, 2015-12-04:
FWIW, I am also seeing those "dri2 connection failed" errors, also on a system with nvidia and intel ICDs installed - and it does not seem related to code-iai/kinect2 at all, i.e. is even happening when just running clinfo or similar tests