code-iai/iai_kinect2 #124: Kinect_bridge not recognizing calibration data

Alex-Keyes, 2015-07-21:
I have run the calibration successfully and put all of the .yaml files in a directory named with the same serial number as the Kinect2 I have been using.  I noticed however, there was no improvement when I was attempting to use the calibration files vs. an uncalibrated Kinect2.  It seems like the kinect2_bridge is not recognizing the calibration files.  There are distinct errors in both the ir and color images and its clear they are not calibrated.  

This a screengrab of the first lines generated by kinect2_bridge:
[image]

Note the abberations around the clipboard in the images.  
[image]
[image]

Is this an issue with where I have put the calibration files?

lakehanne, 2015-07-21:
See issue #117

wiedemeyer, 2015-07-29:
Hi,
it seems like there is a calibration used, since the color and depth image are undistorted. Could you check in the output of the console if it says something about the calibration data loaded? There should be some output.
Did you calibrated color and ir together (with the `sync` option of the calibration tool)?

Alex-Keyes, 2015-08-03:
Yeah I loaded both calibration data I made as well as loading the sample calibration data.  I definitely used the sync tool in calibrations methods. Neither seemed to offer any improvement on the kinect's ability to sense accurately. 

I just checked the console output for some kind of change and there has not been a deviation from the original screencapture of the console I posted originally.

wiedemeyer, 2015-08-05:
There should be output on the terminal, that stats which camera parameters are used. This would appear if no calibration data is found:
```
default ir camera parameters: 
fx 364.859, fy 364.859, cx 260.118, cy 208.079
k1 0.0952483, k2 -0.265004, p1 0, p2 0, k3 0.0870643

default color camera parameters: 
fx 1081.37, fy 1081.37, cx 959.5, cy 539.5
using sensor defaults for color intrinsic parameters.
using sensor defaults for ir intrinsic parameters.
using defaults for rotation and translation.
using defaults for depth shift.
```
How is the output from you terminal? (Just copy and paste it here)

closed 2015-10-14 by wiedemeyer

yukarisato, 2015-10-23:
Hi,
> Alex-Keyes 
> It seems like the kinect2_bridge is not recognizing the calibration files. 
>There are distinct errors in both the ir and color images and its clear they are not calibrated. 

Me too. 

I just checked a screengrab of the lines generated by $ rosrun kinect2_bridge kinect2_bridge .
~~~~~~~~~~
default ir camera parameters: 
fx 367.347, fy 367.347, cx 253.963, cy 206.633
k1 0.0848951, k2 -0.27307, p1 0, p2 0, k3 0.102904

default color camera parameters: 
fx 1081.37, fy 1081.37, cx 959.5, cy 539.5
using sensor defaults for color intrinsic parameters.
using sensor defaults for ir intrinsic parameters.
using defaults for rotation and translation.
using defaults for depth shift.
~~~~~~~~~~
It is different from yours.
Please tell me what I should do...

wiedemeyer, 2015-10-23:
Did you followed step 10 - 12 from the [README](https://github.com/code-iai/iai_kinect2/tree/master/kinect2_calibration#calibrating-the-kinect-one)?

yukarisato, 2015-10-23:
Thank you!
I'll check it again and try it.

TrinhTUHH, 2017-08-11:
Hi, I did follow step 10 -12 but, it still loads the defaults.