code-iai/iai_kinect2 #125: Wrong depth values

lakehanne, 2015-07-22:
Hi @wiedemeyer , I am picking up coordinates in the color frame from the viewer.cpp code and having to look up the depth equivalent in the depth map as follows:
```
            float depthIntensity = depth.at<float>(y, x);
```

but I notice the corresponding depth value I am picking up is a lot different from what the ground truth measurement should be. Picking up from the `dispDepth` doesn't help either. But when I pick the depth points from the depth map of libfreenect2's Protonect, it does yield sensible values.
I notice Protonect's depth map is scaled by a factor of `4500.0f` while your `dispdepth` is scaled by `12000.0f`. Could it be the reason for the discrepancy?

Is there a way to access the undistorted map the way it is in libfreenect2 within kinect2_bridge? I would appreciate your reply.

lakehanne, 2015-07-29:
Seems my problem is mostly from the way I am casting the depth values in ` float depthIntensity = depth.at<float>(y, x);`. After changing to  

```
unsigned short depthright  = depthDisp.at<unsigned short>(eye_y, eye_x)
```
 I seem to be obtaining the values of the depth pixel intensities. I am close but not there yet.

wiedemeyer, 2015-07-29:
just use the depth image that is received by the callback method. It contains the depth values as uint16_t. Each pixels value is the depth in mm.

lakehanne, 2015-07-29:
Thanks. This fixes my brobdinagian, unwanton quandary :)

lakehanne, 2015-07-30:
It seems the retrieved depth values from the ros code and protonect are off by a significant margin. In protonect, I find the depth map on a stationary object to be accurate to within 5mm after accounting for the static offset of approx. 37mm from ground truth. In ROS, strangely, I found the values to flunctuate by an even wider margin post calibration. This was after adding the static offset from my calibrated values. These charts would explain my thoughts better.

[image]

[image]

wiedemeyer, 2015-08-05:
The offset is just an offset that is added to the depth values, it does not change the depth values in any other way, so it cannot be the source of fluctuation. Maybe the offset that is estimated is wrong, you could just change the value in the `calib_depth.yaml`. The whole depth offset is a preliminary part.

lakehanne, 2015-08-31:
I was talking about the offset as a result of the incorrect tables of kinect2 libfreenect driver as discussed [here](https://github.com/OpenKinect/libfreenect2/issues/223).

wiedemeyer, 2015-10-26:
The hard coded tables is solved.

closed 2015-10-26 by wiedemeyer