code-iai/iai_kinect2 #126: Calibration problem: I can't get depth image

Guhnoo, 2015-07-23:
Hello,  I have some problem in calibration. 

## I can run ' calibration record ir' like this
$ optirun --no-failsafe rosrun kinect2_calibration kinect2_calibration chess5x7x0.034 record ir

I can't get nothing about depth except ir window, and I was given black depth img.

[image]

Is it right? or I have problem??  
when I test  Protonect  (try: $ optirun --no-failsafe ./Protonect)  , I was given four image well.

wiedemeyer, 2015-07-29:
Hi,
those depth images have a 16 bit depth value. some programs might have problems to display them. Also the depth values are very low, so they might appear as black in viewers. You could simple `rostopic echo -n 1 /kinect2/sd/image_depth` and see if it contains just zeros, if not, everything is fine.

Guhnoo, 2015-08-02:
Thank you for your answer.   when I run 
## $ roslaunch kinect2_bridge kinect2_bridge.launch publish_tf:=true 
and after
##  rostopic echo -n 1 /kinect2/sd/image_depth

I got this . 
[image]

and,.... Do I have problem?, doesn't it ?

wiedemeyer, 2015-08-05:
For using OpenGL depth processor you need at least OpenGL version 3.3. Please check that. You can also try using OpenCL depth processor. Try out `./Protonect cl` from libfreenect2. And check the FAQ from the README.

closed 2015-10-14 by wiedemeyer