code-iai/iai_kinect2 #127: controlling multiple kinect2 cameras  and camera intrinsic related issues- design approach not a bug

cnanda-zbr, 2015-07-28:
Hello ,
The problem is not a bug but a design decision that we would like to make. Also wanted to know whether having the following functionality in the kinect2_bridge would be helpful. 
I am trying to develop a multiple camera acquisition system with kinect2 cameras and deciding on the best approach for that. 
The camera system should accept a command then acquire images/ pointcloud from the sensors based on that. 
Currently I have the kinect2_viewer application  blocking on a command , after receiving a command it starts acquiring  data from the kinect2_bridge publisher node and then sends them over to a server that is listening for data. 
Will it make sense to add a blocking function call to the kinect2_bridge ? So it serves as a modification that can accept "acquire " command and the number of pictures as arguments? I am doing that in the kinect2_viewer code now. 
Where is the camera serial published ?  Is it buried somewhere in the camerainfo ? 
 
Question about camera intrinsic parameters. 
1. How do the parameters get calibrated ? Does the driver provide you calibrated values ? Are they unique for each camera ?
2. When you use the nodelet to do conversions to pointcloud2 format , does it take into account the intrinsic parameters ? 

Best 
Chinmay

lakehanne, 2015-07-28:
>_1. How do the parameters get calibrated ? Does the driver provide you calibrated values ? Are they unique for each camera?_

OpenCV calibration. On the readme page of kinect2_calibration, you can find the instructions on how to recover the calibration parameters. They will be in yaml files in your calibration directory.

2. When you use the nodelet to do conversions to pointcloud2 format , does it take into account the intrinsic parameters ?

The Pointcloud viewer in `kinect2_viewer` uses color -> image registration, which is not as good as the depth -> color registration in the imageviewer function in kinect2_viewer. This may explain your feeling of the pointcloud + RGB images not being as synchronized as the imageviewer.

See issue #124 .

wiedemeyer, 2015-07-29:
Hi,
kinect2_bridge is just a bridge between libfreenect2 and ROS. this additional functionality should be better placed in an additional node. This node might then also work with different sensors too.
There is no serial number in the camera_info.
To your questions:
1. The calibration for each sensor is differently. You need to calibrate it using the provided calibration tool. For further information please look here: https://github.com/code-iai/iai_kinect2/tree/master/kinect2_calibration
2. They do use the intrinsics from the camera_info.

closed 2015-08-07 by cnanda-zbr