code-iai/iai_kinect2 #128: CPU suddenly becomes overloaded in running kinect2_bridge

mu-777, 2015-07-29:
Hello.
Now I have a little bit strange problem that CPU suddenly become overloaded.

**In Brief**
I recorded my problem. This video should help you to understand the problem. 
Watch it: 
[image]

**Details**
I have no ideas to describe the problem without saying CPU suddenly become overloaded. And I have not had such problem in running other any programs.
It took about 10 minutes until the CPU overloaded in the time I record the video, however the time depends. Sometimes, it takes 30minutes and sometimes it takes 5 minutes.
Always some cores overloaded and the others are 0~10% usages so I think nodelet multi-threading in kinect2_bridge has some problems or bugs but this is just my impression. Now I have no ideas to fit it.  

**What I tried**
- Run ``$ Protonect`` through the night
- Run ``$ Protonect cl``, ``$ Protonect gl`` and ``$ Protonect cpu``

-> Works well

- Run kinect2_bridge with ``<arg name="depth_method" default="opengl"/>``, ``default="opencl"`` and ``default="cpu"``
- Run kinect2_bridge with ``<arg name="reg_method" default="opencl"/>`` and ``default="cpu"``
- Run kinect2_bridge with ``<arg name="fps_limit" default="30"/>``
- Run kinect2_bridge with ``<arg name="queue_size" default="1"/>``
- Run kinect2_bridge with ``<arg name="worker_threads" default="1"/>``
- Run kinect2_bridge with ``<arg name="worker_threads" default="12"/>``

-> The problem happens

**My system**
- OS: ubuntu 15.04 64bit
(but I have the same problem in ubuntu 14.04)
- CPU: Intel Core i7-5820K
- Memory: 16GiB
- Graphic: GeForce GTX 980
- ROS distribution: indigo(installed from source)

Could you help me?
Thanks,

wiedemeyer, 2015-07-29:
Hi,
seems like it is hanging somewhere. please compile it in debug and run it inside a debugger. As soon as the cpu usage goes to 100%, stop the execution and you should see where it hangs.
Using a `queue_size` of `1` sometimes leads to problems. Please try with `5`.

mu-777, 2015-07-30:
Hi, thank you for a quick responce :)

I tried gdb referring to [here](http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB) .
I added ``launch-prefix="xterm -e gdb --args"`` into five ``<node>`` tags in a default ``kinect2_bridge.launch``, which are ( you know more than me) nodelet_manager, kinect2_bridge_nodelet, and sd, qhd and hd point cloud nodes.
And I launched ``$ roslaunch kinect2_bridge kinect2_bridge.launch`` then five xterm windows appeared like this.
[image]

After several minutes, some cpu core usages went to 100% so I stopped the upper left xterm window with Ctrl+C.
[image]

To be honest, I'm not familar with programing in UNIX and don't know so much about debugging.
So I couldn't find anything to help to solve the problem. I'm not sure this was a correct way actually.

Could you tell me the good way to debug and find the causes of the problem?

wiedemeyer, 2015-07-30:
First of all, please test with a `queue_size` set to 5. Just to make sure it's not related to that setting.
Then I would suggest to run only kinect2_bridge without the nodelet things `rosrun kinect2_bridge kinect2_bridge`.
If the issue is still present, compile it with `catkin_make -DCMAKE_BUILD_TYPE=Debug` (but delete devel and build folders first) and run kinect2_bridge in gdb with `gdb catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge` (or use for example qt creator, which has a graphical interface to gdb).
As soon as the issue starts, press `CTRL+C`, do a backtrace of the different threads. Maybe you can also see which thread is using so much CPU and what that thread is doing. To see the threads and their usage you can also use `htop`

mu-777, 2015-08-13:
Sorry, now I cannot touch my kinect.
I will try what you said next week.

mu-777, 2015-08-18:
OK, I tried some things.

I had the problem when I did ``rosrun kinect2_bridge kinect2_bridge``, so I tried ``catkin_make -DCMAKE_BUILD_TYPE=Debug`` and ``gdb catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge``

The image shows the screen when the problem happens.
[image]

From the image, you can know the process 5451 was something wrong, so I used the command ``ps u --pid 5451 -L`` to check the details about the process before stopping the kinect2_bridge.
And I stopped the kinect2_bridge with ``Ctrl+C`` and check a log using ``thread apply all backtrace``.
The results of two commands are too long to paste here so I upload [here](https://gist.github.com/mu-777/c83f59cb8b8d3e1c2dd3)

If you need other infomation and something I should do, please let me know. :)
Thanks

wiedemeyer, 2015-08-20:
I really don't know, what the problem could be. You could try to start it with less threads and see what happens. Try to use the default of 4.
You are also using ROS Jade, right? I didn't tried out Jade yet. But you could try to run it on Indigo, to see if the error shows up too.
Other than this I have no idea.

mu-777, 2015-08-24:
Yeah, I used jade in that case but I had the same issue when I tried under indigo.

Anyway, thank you for having helped me.
No others comment here, which means this should not be a common issue and my computer might have some problems.
If you will have some ideas or find out something, tell me please.

thanks

closed 2015-10-14 by wiedemeyer