code-iai/iai_kinect2 #129: Segmentation fault and

Mosa-, 2015-08-06:
Hi,

I get a segmentation fault if I start  the bridge with a high fps. I also had a build problem, which I describe here https://github.com/OpenKinect/libfreenect2/issues/355 .
I have to add in the CMakefile:
```
    SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -msse -msse2 -msse3")
  ELSEIF(COMPILER_SUPPORTS_CXX0X)
    SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++0x -msse -msse2 -msse3")
```

wiedemeyer, 2015-08-08:
What do you mean by `high fps`? The highest frame rate possible with the Kinect 2 is 30 fps. It always helpful if you could provide more information on the issue, like which parameters did you used, what is the console output? You could also try to compile it in debug, start it with `gdb` and provide a stack trace.

CMakeLists: Normally the compiler should know your CPU architecture. Try to export the correct `-march=...` flag to the default compiler options, [see here](https://gcc.gnu.org/onlinedocs/gcc-4.8.2/gcc/i386-and-x86-64-Options.html). But this is a system configuration issue and is not related to this package in any way.

Mosa-, 2015-08-10:
To start the bridge I use this command:
```
rosrun kinect2_bridge kinect2_bridge _fps_limit:=30 _depth_method:=opencl _reg_method:=cpu _max_depth:=0.2 _min_depth:=0.1 _queue_size:=100 _bilateral_filter:=false _edge_aware_filter:=false _worker_threads:=8
```

If I start two ROS-Nodes I get at the kinect2_bridge a segmentation fault:
```
[RgbPacketStreamParser::onDataReceived] skipping rgb packet!
[DepthPacketStreamParser::onDataReceived] skipping depth packet
Speicherzugriffsfehler (Speicherabzug geschrieben) <- seg fault
```

Should I change the CMakeList of kinect2_bridge?

wiedemeyer, 2015-08-11:
What do you mean by `If I start two ROS-Nodes`? What are the nodes that you are starting, what is the output of the kinect2_bridge on the termial?

Please compile it in debug `catkin_make -DCMAKE_BUILD_TYPE="Debug"`, start kinect2_bridge using gdb (`gdb catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge`), then type `run`, and after it crashed `bt`.

But even if it works, the parameters are not ideal. I think the minimum distance for the kinect2 is arround 0.4 to 0.5 meters, so setting the max distance to 0.2 meters might result in no depth data.

Mosa-, 2015-08-12:
I done it and I got a segmentation fault with this bt:
```
[RgbPacketStreamParser::onDataReceived] skipping rgb packet!

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xa29d8b40 (LWP 8579)]
0xb7c2fdd0 in ?? () from /usr/lib/i386-linux-gnu/libopencv_core.so.2.4
(gdb) bt
#0  0xb7c2fdd0 in ?? () from /usr/lib/i386-linux-gnu/libopencv_core.so.2.4
#1  0xb7c31da0 in cv::flip(cv::_InputArray const&, cv::_OutputArray const&, int) () from /usr/lib/i386-linux-gnu/libopencv_core.so.2.4
#2  0x08091c2b in Kinect2Bridge::processColor (this=0xbfffdf14, color=..., 
    images=..., status=...)
    at /home/felix/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1047
#3  0x08090d73 in Kinect2Bridge::receiveColor (this=0xbfffdf14)
    at /home/felix/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:922
#4  0x080904bc in Kinect2Bridge::threadDispatcher (this=0xbfffdf14, id=2)
    at /home/felix/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:847
#5  0x080a5944 in std::_Mem_fn<void (Kinect2Bridge::*)(unsigned int)>::operator()<unsigned int, void>(Kinect2Bridge*, unsigned int&&) const (this=0x8527acc, 
    __object=0xbfffdf14) at /usr/include/c++/4.8/functional:601
#6  0x080a57ff in std::_Bind_simple<std::_Mem_fn<void (Kinect2Bridge::*)(unsigned int)> (Kinect2Bridge*, unsigned int)>::_M_invoke<0u, 1u>(std::_Index_tuple<0u, 1u>) (this=0x8527ac4) at /usr/include/c++/4.8/functional:1732
#7  0x080a56b5 in std::_Bind_simple<std::_Mem_fn<void (Kinect2Bridge::*)(unsigned int)> (Kinect2Bridge*, unsigned int)>::operator()() (this=0x8527ac4)
    at /usr/include/c++/4.8/functional:1720
#8  0x080a5620 in std::thread::_Impl<std::_Bind_simple<std::_Mem_fn<void (Kinect---Type <return> to continue, or q <return> to quit---
2Bridge::*)(unsigned int)> (Kinect2Bridge*, unsigned int)> >::_M_run() (
    this=0x8527ab8) at /usr/include/c++/4.8/thread:115
#9  0xb7611d1e in ?? () from /usr/lib/i386-linux-gnu/libstdc++.so.6
#10 0xb788cf70 in start_thread (arg=0xa29d8b40) at pthread_create.c:312
#11 0xb7448bee in clone () at ../sysdeps/unix/sysv/linux/i386/clone.S:129
```

I mean with two ROS-Nodes, that I created two application (ROS-Nodes) and both subscribed the topic '/kinect2/qhd/image_mono'.

wiedemeyer, 2015-08-13:
There must be something wrong. This line `cv::flip(color, tmp, 1);` only flips the the color frame received from libfreenect2. The only thing that I could imagine is, that you are using an old version of libfreenect2. Because some time ago, the color images from libfreenect2 had 3 bytes per pixel, but this was changed to 4 bytes per pixel. Please check that you are using a recent version of libfreenect2.

Mosa-, 2015-08-13:
I think I use the current version. I did the installation instruction (for installing libfreenect2) again and if I run  make && sudo make install and I get the error which is describes here https://github.com/OpenKinect/libfreenect2/issues/355 .  I added in /example/protonect/CMakeLists.txt 
```
    SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -msse -msse2 -msse3")
  ELSEIF(COMPILER_SUPPORTS_CXX0X)
    SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++0x -msse -msse2 -msse3")
``` And the make and make install works, but I have the same problem (seg fault).

wiedemeyer, 2015-08-14:
I looked at the code to figure out what could have happened. There existed a possibility that the frame buffer could be deallocated before the processing is finished. I commited a fix to the devel branch. Could you check out that branch and test again?

Mosa-, 2015-08-14:
I download the branch and it works. So far I had not no segmentation fault.
Thanks mister.

closed 2015-08-17 by wiedemeyer