code-iai/iai_kinect2 #131: catkin_make error - No rule to make target `/usr/lib/x86_64-linux-gnu/libopencv_ts.so.2.4.8'

ghotson, 2015-08-16:
Protonect is working fine, having trouble getting iai_kinect2 to build with catkin_make. 

I have opencv 2.4.10.1 installed. I might have had 2.4.8 installed at one point, so this might be related to that.

Here's the errors from running catkin_make:

make[2]: *** No rule to make target `/usr/lib/x86_64-linux-gnu/libopencv_ts.so.2.4.8', needed by `/home/ghotson/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge'.  Stop.
[ 73%] make[2]: *** make[1]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
No rule to make target `/usr/lib/x86_64-linux-gnu/libopencv_ts.so.2.4.8', needed by `/home/ghotson/catkin_ws/devel/lib/kinect2_calibration/kinect2_calibration'.  Stop.
make[1]: *** [iai_kinect2/kinect2_calibration/CMakeFiles/kinect2_calibration.dir/all] Error 2
make[2]: *** No rule to make target `/usr/lib/x86_64-linux-gnu/libopencv_ts.so.2.4.8', needed by `/home/ghotson/catkin_ws/devel/lib/libkinect2_bridge_nodelet.so'.  Stop.
make[1]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/all] Error 2
make[2]: *** No rule to make target `/usr/lib/x86_64-linux-gnu/libopencv_ts.so.2.4.8', needed by `/home/ghotson/catkin_ws/devel/lib/kinect2_viewer/kinect2_viewer'.  Stop.
make[1]: *** [iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/all] Error 2
make: *** [all] Error 2

Thanks!

lakehanne, 2015-08-16:
It seems it can't find opencv 2.4.8 files.  What's the output of `pkg-config opencv --modversion`. If it says anything otherwise opencv 2.4.8, then it's looking for the wrong opencv files. 
Maybe you should export the opencv version you have installed to your PATH and try again.

ghotson, 2015-08-16:
pkg-config opencv --modversion says 2.4.10.1. 
I don't have 2.4.8 installed, is it required that I use that version? Or is there a way to change the version it looks for?

wiedemeyer, 2015-08-17:
OpenCV is found by `find_package(OpenCV REQUIRED)` and the resulting variables `${OpenCV_INCLUDE_DIR}` and `${OpenCV_LIBRARIES}` are used. There is no version number named. Please check the values of those variables. Maybe you have to change some environment variables, so that CMake finds your custom installed OpenCV correctly. It works totally fine with the default OpenCV package that comes with Ubuntu.

ghotson, 2015-08-17:
Couldn't figure out how to change whatever environmental variables had to be changed, so I uninstalled opencv and reinstalled 2.4.8 and was able to get it to compile.

Thanks!

closed 2015-08-17 by ghotson