code-iai/iai_kinect2 #132: Protonect works, iai_kinect2 repeatedly crashes and restarts.

jeff-o, 2015-08-19:
I am having an issue where everything works great when I try the various ./Protonect scripts, but the kinect2_bridge package repeatedly crashes.  

First, the hardware.  I am running 14.04LTS an ROS Indigo.  The GPU is a GeForce GTX750Ti, running driver version 340.76.  The driver was installed by selecting it from the list of "Additional Drivers" in the System Settings.

I installed the libfreenect2 package and iai_kinect2 package as per instructions.

I can run ./Protonect gl or cl or cpu and they all launch a viewer window as expected, with good frame rates.

The trouble comes when attempting to use the kinect2_bridge launch file.  Everything seems to start fine, but it would appear that the sd, qhd and hd processes repeatedly crash and restart.  What's amazing is that I can still subscribe to the topics and view the data!  Here's a snippet:

[kinect2_points_xyzrgb_sd-3] restarting process
process[kinect2_points_xyzrgb_sd-3]: started with pid [13807]
[kinect2_points_xyzrgb_qhd-4] restarting process
process[kinect2_points_xyzrgb_qhd-4]: started with pid [13808]
[kinect2_points_xyzrgb_hd-5] restarting process
process[kinect2_points_xyzrgb_hd-5]: started with pid [13809]
[ERROR] [1440016481.655724042]: Failed to load nodelet [/kinect2_points_xyzrgb_qhd] of type [depth_image_proc/point_cloud_xyzrgb]: According to the loaded plugin descriptions the class depth_image_proc/point_cloud_xyzrgb with base class type nodelet::Nodelet does not exist. Declared types are  kinect2_bridge/kinect2_bridge_nodelet nodelet_tutorial_math/Plus
[FATAL] [1440016481.655885286]: Service call failed!
[ERROR] [1440016481.676099923]: Failed to load nodelet [/kinect2_points_xyzrgb_sd] of type [depth_image_proc/point_cloud_xyzrgb]: According to the loaded plugin descriptions the class depth_image_proc/point_cloud_xyzrgb with base class type nodelet::Nodelet does not exist. Declared types are  kinect2_bridge/kinect2_bridge_nodelet nodelet_tutorial_math/Plus
[FATAL] [1440016481.676335319]: Service call failed!
[ERROR] [1440016481.696378445]: Failed to load nodelet [/kinect2_points_xyzrgb_hd] of type [depth_image_proc/point_cloud_xyzrgb]: According to the loaded plugin descriptions the class depth_image_proc/point_cloud_xyzrgb with base class type nodelet::Nodelet does not exist. Declared types are  kinect2_bridge/kinect2_bridge_nodelet nodelet_tutorial_math/Plus
[FATAL] [1440016481.696533999]: Service call failed!

So certainly it's usable for now, but it would be great if it were stable in the background.

jeff-o, 2015-08-20:
It looks like the issue is solved by installing image tools, likely image-proc.  Will do more testing and confirm.

jeff-o, 2015-08-20:
Yup, that seems to clear things up.  Closing.

closed 2015-08-20 by jeff-o

pammirato, 2016-01-27:
same thing happened to me. installed image_pipleline and it worked.
https://github.com/ros-perception/image_pipeline

to install just:
git clone https://github.com/ros-perception/image_pipeline
rosdep install -r --from-paths .
catkin_make

Maybe this should be added as a dependency?

wiedemeyer, 2016-01-27:
It is a dependency for the point cloud nodelets from the launch file. kinect2_bridge itself, does not need it, but I can add it to the package.xml

wiedemeyer, 2016-01-27:
I just added it to the dependencies.
BTW: to install it, you could just use `sudo apt-get install ros-indigo-image-pipeline`.