code-iai/iai_kinect2 #133: Subscribing to the /kinect2/sd/points topic causes crashes after recent ff3758f commit.

barryridge, 2015-08-21:
Hi,

This may be related to Issues [#104](https://github.com/code-iai/iai_kinect2/issues/104), [#106](https://github.com/code-iai/iai_kinect2/issues/106) and [#108](https://github.com/code-iai/iai_kinect2/issues/108).

I've been running iai_kinect2 on the following system:

Processor: Intel Core i7-4770 CPU @ 3.40GHz × 8
Graphics: GeForce GTX 960/PCIe/SSE2
OS: Ubuntu 14.04.3 LTS, 64-bit
Kernel: 3.16.0-031600-generic
ROS: Indigo
Nvidia driver version: 352.30

Things had been working fine, but I did an update recently and started experiencing problems with errors of the form:
```
[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
[RgbPacketStreamParser::onDataReceived] packetsize or sequence doesn't match!
[DepthPacketStreamParser::onDataReceived] not all subsequences received 512
*** Error in `/opt/ros/indigo/lib/nodelet/nodelet': corrupted double-linked list: 0x00007f5f5810a760 ***
[kinect2-1] process has died [pid 19126, exit code -6, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=kinect2 __log:=/home/kinekt2/.ros/log/549d6006-47e2-11e5-b1e4-801f020068e9/kinect2-1.log].
log file: /home/kinekt2/.ros/log/549d6006-47e2-11e5-b1e4-801f020068e9/kinect2-1*.log
```
I was able to reproduce this by simply running `roslaunch kinect2_bridge kinect2_bridge.launch` and `rostopic echo /kinect2/sd/points` in separate terminals.  Interestingly, however, echoing from either the `/kinect2/qhd/points` or `/kinect2/hd/points` topics appears to run stably, although each of them still report something like:
```
[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
[DepthPacketStreamParser::onDataReceived] not all subsequences received 127
[DepthPacketStreamParser::onDataReceived] not all subsequences received 896
[RgbPacketStreamParser::onDataReceived] packetsize or sequence doesn't match!
[kinect2_bridge] depth processing: ~88.1681Hz (11.342ms) publishing rate: ~26.6284Hz
[kinect2_bridge] color processing: ~49.8251Hz (20.0702ms) publishing rate: ~22.967Hz
[OpenCLDepthPacketProcessor] avg. time: 0ms -> ~infHz
[TurboJpegRgbPacketProcessor] avg. time: 0ms -> ~infHz
[kinect2_bridge] depth processing: ~83.5983Hz (11.962ms) publishing rate: ~29.9488Hz
[kinect2_bridge] color processing: ~45.3793Hz (22.0365ms) publishing rate: ~30.2816Hz
[OpenCLDepthPacketProcessor] avg. time: 0ms -> ~infHz
[TurboJpegRgbPacketProcessor] avg. time: 0ms -> ~infHz
```

Having read through the previous issue reports that I mentioned above, I checked to see if both libfreenect2 and kinect2_bridge were using the same libusbx shared library using `ldd /path/to/Protonect | grep libusb` and `ldd /path/to/kinect2_bridge | grep libusb` respectively.  That seemed to check out ok with each of them reporting:
```
libusb-1.0.so.0 => /path/to/libfreenect2/depends/libusb/lib/libusb-1.0.so.0 (0x00007f72c2b92000)
```

After much head-scratching, I realised that the problem had emerged after updating iai_kinect2 at some point recently, so I reverted back to [90ddda0](https://github.com/code-iai/iai_kinect2/commit/90ddda0ca85802e416127bc404cadca37ed8326b) and now things run stably for all three of the `/kinect2/sd/points`, `/kinect2/qhd/points` and`/kinect2/hd/points`, although I still get reports of mismatching packets/sequences:
```
[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
[DepthPacketStreamParser::onDataReceived] not all subsequences received 1016
[RgbPacketStreamParser::onDataReceived] packetsize or sequence doesn't match!
[OpenCLDepthPacketProcessor] avg. time: 0ms -> ~infHz
[TurboJpegRgbPacketProcessor] avg. time: 0ms -> ~infHz
[kinect2_bridge] depth processing: ~93.769Hz (10.6645ms) publishing rate: ~26.6646Hz
[kinect2_bridge] color processing: ~13023.4Hz (0.0767846ms) publishing rate: ~22.9982Hz
[OpenCLDepthPacketProcessor] avg. time: 0ms -> ~infHz
[TurboJpegRgbPacketProcessor] avg. time: 0ms -> ~infHz
[kinect2_bridge] depth processing: ~97.9288Hz (10.2115ms) publishing rate: ~29.9963Hz
[kinect2_bridge] color processing: ~13765.4Hz (0.0726461ms) publishing rate: ~29.9963Hz
```

I'm ok with using [90ddda0](https://github.com/code-iai/iai_kinect2/commit/90ddda0ca85802e416127bc404cadca37ed8326b) for now, but I thought I had better submit this issue report anyway.  Maybe it will be of help to you guys.  Thank you very much for all of the great work you've been doing with this driver!

Cheers,
Barry

wiedemeyer, 2015-08-21:
Hi Barry,
the output `[OpenCLDepthPacketProcessor]...`, `[TurboJpegRgbPacketProcessor]...`, `[DepthPacketStreamParser::onDataReceived]...` and `[RgbPacketStreamParser::onDataReceived]...` are generated by libfreenect2. The messages containing `(sub)sequence` appear if something was not received correctly, maybe due to a high CPU load. That the package processors are reporting a frame rate of 0 while kinect2_bridge is still receiving images from libfreenect2 is weird.
The last commit was pretty small and I don't see why it should introduce new issues. Could you try to build libfreenect2 and kinect2_bridge from scratch. For libfreenect2 remove everything in the build folder and run `cmake` and `make install` again, for the catkin workspace remove the devel and build folder and run `catkin_make`.

barryridge, 2015-08-21:
Hi Thiemo,

Ok, I've just tried building libfreenect2 and kinect2_bridge from scratch and have ended up with the same results as reported above.

This may or may not be related, but while doing this, it seemed as though I had broken the catkin_workspace by deleting the build and devel folders.  `catkin_make` was throwing this error: `ImportError: "from catkin_pkg.package import parse_package" failed: No module named catkin_pkg.package`.  Upon further investigation, it turned out that this had something to do with the fact that I had recently installed anaconda on top of previously installed python and catkin versions.  The system python installation is version 2.7.6, while the anaconda version is 2.7.10.  Meanwhile, I managed to resolve the issue by running `pip install catkin_pkg`.  Now `dpkg -l python-catkin-pkg` shows version 0.2.10-1 and `pip freeze | grep catkin` shows `catkin==0.6.14` and `catkin-pkg==0.2.10`.  In any case, after the pip install of catkin_pkg, I was able to successfully run catkin_make in the catkin workspace and build kinect2_bridge again.  Here is the output of that:

```
kinekt2@kinekt2-desktop:~/catkin_ws$ catkin_make
Base path: /home/kinekt2/catkin_ws
Source space: /home/kinekt2/catkin_ws/src
Build space: /home/kinekt2/catkin_ws/build
Devel space: /home/kinekt2/catkin_ws/devel
Install space: /home/kinekt2/catkin_ws/install
####
#### Running command: "cmake /home/kinekt2/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/kinekt2/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/kinekt2/catkin_ws/install -G Unix Makefiles" in "/home/kinekt2/catkin_ws/build"
####
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Using CATKIN_DEVEL_PREFIX: /home/kinekt2/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/kinekt2/Desktop/workspace/devel;/home/kinekt2/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/kinekt2/Desktop/workspace/devel;/home/kinekt2/catkin_ws/devel;/opt/ros/indigo
-- Found PythonInterp: /home/kinekt2/anaconda/bin/python (found version "2.7.10") 
-- Using PYTHON_EXECUTABLE: /home/kinekt2/anaconda/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/kinekt2/catkin_ws/build/test_results
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.14
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 5 packages in topological order:
-- ~~  - iai_kinect2 (metapackage)
-- ~~  - kinect2_registration
-- ~~  - kinect2_bridge
-- ~~  - kinect2_calibration
-- ~~  - kinect2_viewer
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'iai_kinect2'
-- ==> add_subdirectory(iai_kinect2/iai_kinect2)
-- +++ processing catkin package: 'kinect2_registration'
-- ==> add_subdirectory(iai_kinect2/kinect2_registration)
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Try OpenMP C flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Try OpenMP CXX flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
[... 154 lines cut ...]

After reverting back to [90ddda0](https://github.com/code-iai/iai_kinect2/commit/90ddda0ca85802e416127bc404cadca37ed8326b), `rostopic echo /kinect2/sd/points` gives me:
```
[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
[DepthPacketStreamParser::onDataReceived] not all subsequences received 1020
[RgbPacketStreamParser::onDataReceived] packetsize or sequence doesn't match!
[kinect2_bridge] depth processing: ~110.946Hz (9.01336ms) publishing rate: ~26.6585Hz
[kinect2_bridge] color processing: ~11546.7Hz (0.0866047ms) publishing rate: ~22.993Hz
[OpenCLDepthPacketProcessor] avg. time: 0ms -> ~infHz
[TurboJpegRgbPacketProcessor] avg. time: 0ms -> ~infHz
[kinect2_bridge] depth processing: ~95.4423Hz (10.4775ms) publishing rate: ~29.993Hz
[kinect2_bridge] color processing: ~13342.4Hz (0.0749492ms) publishing rate: ~29.6598Hz
[OpenCLDepthPacketProcessor] avg. time: 0ms -> ~infHz
[TurboJpegRgbPacketProcessor] avg. time: 0ms -> ~infHz
[kinect2_bridge] depth processing: ~95.3067Hz (10.4924ms) publishing rate: ~29.9938Hz
[kinect2_bridge] color processing: ~12993.5Hz (0.0769615ms) publishing rate: ~29.9938Hz
```
running stably.  Pretty much the same with the hd and qhd topics.

If you spot any problems with my setup, let me know.  I agree that the 90ddda0 commit seems small and is unlikely to be the cause.  Strange problem!

Cheers,
Barry

wiedemeyer, 2015-08-21:
Really a strange problem. I just ran it with the current master and got the following output:
```
[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
[DepthPacketStreamParser::onDataReceived] not all subsequences received 1020
[RgbPacketStreamParser::onDataReceived] packetsize or sequence doesn't match!
[kinect2_bridge] depth processing: ~60.9728Hz (16.4007ms) publishing rate: ~26.9131Hz
[kinect2_bridge] color processing: ~17333.9Hz (0.0576905ms) publishing rate: ~23.2582Hz
[OpenCLDepthPacketProcessor] avg. time: 2.07261ms -> ~482.483Hz
[TurboJpegRgbPacketProcessor] avg. time: 12.9213ms -> ~77.3917Hz
[kinect2_bridge] depth processing: ~53.7293Hz (18.6118ms) publishing rate: ~30.2368Hz
[kinect2_bridge] color processing: ~18765.1Hz (0.0532905ms) publishing rate: ~30.2368Hz
[OpenCLDepthPacketProcessor] avg. time: 2.10542ms -> ~474.965Hz
[TurboJpegRgbPacketProcessor] avg. time: 13.0531ms -> ~76.61Hz
```

```
rostopic hz /kinect2/sd/points 
subscribed to [/kinect2/sd/points]
average rate: 28.999
	min: 0.030s max: 0.050s std dev: 0.00445s window: 18
average rate: 29.689
	min: 0.029s max: 0.050s std dev: 0.00293s window: 48
average rate: 29.805
```

If you run `./protonect cl`, will it then also print `avg. time: 0ms -> ~infHz`? Maybe it's related to the to libfreenect2. Currently I am using https://github.com/OpenKinect/libfreenect2/commit/9a70d5ea4f6a2eff4e8a47dd6c0af56eeba9926b, I will try out the newest master on monday. Or you could try to use that commit to see if something changes.

barryridge, 2015-08-21:
`./Protonect cl` gives me:

```
[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 009867541947
device firmware: 4.1.3911.0.7
[DepthPacketStreamParser::onDataReceived] not all subsequences received 0
[DepthPacketStreamParser::onDataReceived] not all subsequences received 512
[OpenCLDepthPacketProcessor] avg. time: 2.82562ms -> ~353.905Hz
[TurboJpegRgbPacketProcessor] avg. time: 15.1351ms -> ~66.0716Hz
[OpenCLDepthPacketProcessor] avg. time: 2.70877ms -> ~369.171Hz
[TurboJpegRgbPacketProcessor] avg. time: 14.9061ms -> ~67.0868Hz
[OpenCLDepthPacketProcessor] avg. time: 2.72327ms -> ~367.206Hz
[TurboJpegRgbPacketProcessor] avg. time: 14.8411ms -> ~67.3805Hz
[OpenCLDepthPacketProcessor] avg. time: 2.63597ms -> ~379.367Hz
[TurboJpegRgbPacketProcessor] avg. time: 14.7858ms -> ~67.6326Hz
```

As an aside, the Protonect viewer is showing all of the image outputs inverted along the horizontal axis, but I don't know if that's in any way related.  I don't think it used to do that though afair.

I will try to check [9a70d5e](https://github.com/OpenKinect/libfreenect2/commit/9a70d5ea4f6a2eff4e8a47dd6c0af56eeba9926b) at some point and let you know.

barryridge, 2015-08-21:
Btw, I get this for `rostopic hz /kinect2/sd/points`:

```

subscribed to [/kinect2/sd/points]
average rate: 29.716
	min: 0.030s max: 0.042s std dev: 0.00265s window: 17
average rate: 29.874
	min: 0.030s max: 0.042s std dev: 0.00181s window: 47
average rate: 29.923
```

wiedemeyer, 2015-08-24:
I just tried out the newest master of libfreenect2 and it is working without issues. Just to make sure, is kinect2_bridge linking against the correct libfreenect2.so?

barryridge, 2015-08-24:
`ldd ./devel/lib/kinect2_bridge/kinect2_bridge | grep libfreenect` returns:
```
libfreenect2.so => /usr/local/lib/libfreenect2.so (0x00007f48e4bc7000)
libusb-1.0.so.0 => /home/kinekt2/libfreenect2/depends/libusb/lib/libusb-1.0.so.0 (0x00007f48dd416000)
```
which I think looks ok because the default install folder specified by cmake in the protonect directory is `/usr/local`.  In any case, I checked the dates on both of those, and they're both recent builds I made at the end of last week.

wiedemeyer, 2015-08-24:
You could compile libfreenect2 and kinect2_bridge in debug, start it within a debugger and try to find out what causes the crash.

barryridge, 2015-08-24:
Ok, I'll try to find some time to try that out a bit later.  Thanks for the help!

wiedemeyer, 2015-08-26:
I was able to reproduce the issue. Sadly I couldn't figure out why it is not working and what the source of this error was.
I changed some stuff and hopefully it solves the issue for you as well. Could you checkout the devel branch and test if it is still crashing for you?

barryridge, 2015-08-26:
Hi Thiemo,

I haven't had a chance to get into the debugging yet, but I've just checked out the devel branch and tested it and there's some good news and bad news there.  The good news is that I can stream the three topics `/kinect2/sd/points`, `/kinect2/qhd/points` and `echo /kinect2/hd/points` successfully.
`echo /kinect2/sd/points` produced the following for example:
```
[kinect2_bridge] client connected. starting device...
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 01 04 04 47 4b 54 34 30 38 2e 31    31332e2e2e2e2e2e474b543430382e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
[RgbPacketStreamParser::onDataReceived] packetsize or sequence doesn't match!
[kinect2_bridge] depth processing: ~1.53849e-169Hz (6.49987e+171ms) publishing rate: ~26.9111Hz
[kinect2_bridge] color processing: ~1.09682e-222Hz (9.11724e+224ms) publishing rate: ~22.592Hz
[OpenCLDepthPacketProcessor] avg. time: 0ms -> ~infHz
[TurboJpegRgbPacketProcessor] avg. time: 0ms -> ~infHz
[kinect2_bridge] depth processing: ~86.2146Hz (11.599ms) publishing rate: ~29.9001Hz
[kinect2_bridge] color processing: ~175.143Hz (5.70963ms) publishing rate: ~29.5679Hz
[OpenCLDepthPacketProcessor] avg. time: 0ms -> ~infHz
[TurboJpegRgbPacketProcessor] avg. time: 0ms -> ~infHz
```
The bad news is that, as you can see from the above, the issue of measuring the frame rate seemingly remains.  Still though- it looks like you have solved my original issue!

Let me know if you need any more help with the debugging or anything.

Cheers,
Barry

hersh, 2015-08-26:
Devel works for me too, thanks @wiedemeyer .  I also see the 0ms report, but that doesn't affect my code, so I'm OK with it.

wiedemeyer, 2015-08-27:
The `0ms` is still strange. Each packet processor has a `startTiming` and `stopTiming` method. You could add some `std::cout` to see why it always shows `0ms`. My guess is that `timing_acc` is zero and that `timer.stop()` is giving back zero. Could you check which timer implementation is used in your case? For me it is using the CXX11 version. You can check by looking into the `config.h` and see if `LIBFREENECT2_WITH_CXX11_SUPPORT` is defined. If not, then the `GLFW` implementation is used.

barryridge, 2015-08-27:
In `libfreenect/examples/protonect/include/libfreenect2/config.h` I've got the following:
```
#ifndef LIBFREENECT2_CONFIG_H
#define LIBFREENECT2_CONFIG_H

#ifdef _MSC_VER
#define LIBFREENECT2_PACK( __Declaration__ ) __pragma( pack(push, 1) ) __Declaration__ __pragma( pack(pop) )
#else
#define LIBFREENECT2_PACK( __Declaration__ ) __Declaration__ __attribute__((__packed__))
#endif

#ifdef _WIN32
#define LIBFREENECT2_API __declspec(dllexport)
#else
#define LIBFREENECT2_API
#endif

#define LIBFREENECT2_WITH_OPENGL_SUPPORT

#define LIBFREENECT2_WITH_OPENCL_SUPPORT

/* #undef LIBFREENECT2_THREADING_STDLIB */

#define LIBFREENECT2_THREADING_TINYTHREAD

/* #undef LIBFREENECT2_WITH_CXX11_SUPPORT */

#endif // LIBFREENECT2_CONFIG_H
```

wiedemeyer, 2015-08-27:
Then try to compile and install libfreenect2 with CXX11 support. `cmake ../examples/protonect/ -DENABLE_CXX11=ON`
If it is working, then there must be something wrong with `GLFW` when used from the `libfreenect2.so`.

barryridge, 2015-08-27:
That appears to have worked!

```
[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
[DepthPacketStreamParser::onDataReceived] not all subsequences received 768
[RgbPacketStreamParser::onDataReceived] packetsize or sequence doesn't match!
[kinect2_bridge] depth processing: ~86.6812Hz (11.5365ms) publishing rate: ~26.9861Hz
[kinect2_bridge] color processing: ~152.068Hz (6.57602ms) publishing rate: ~23.3213Hz
[OpenCLDepthPacketProcessor] avg. time: 3.00899ms -> ~332.337Hz
[TurboJpegRgbPacketProcessor] avg. time: 15.5695ms -> ~64.2279Hz
[kinect2_bridge] depth processing: ~83.047Hz (12.0414ms) publishing rate: ~29.9986Hz
[kinect2_bridge] color processing: ~171.724Hz (5.82329ms) publishing rate: ~29.6653Hz
```

wiedemeyer, 2015-08-27:
Maybe you could open an Issue on libfreenect2. Maybe they have an idea why the `GLFW` is not working.

closed 2015-08-27 by wiedemeyer

lakehanne, 2015-09-21:
I also sufferred this problem recently and compiling and installing libfreenect2 with CXX11 support resolved the issue.
Thanks @wiedemeyer!