code-iai/iai_kinect2 #134: Crash after running just a few seconds

hersh, 2015-08-25:
Hi, and thanks for all your work on this.  I couldn't use the Kinect2 without it. :)

I find that the latest commit ff3758f29719428c645cc4f8a7934d75abfbb7c0 seems to cause a seg fault after just a few seconds of running ```roslaunch kinect2_bridge kinect2_bridge.launch```

It was all working fine yesterday, but today I made a fresh clone of the repo and I saw this crash.  Comparing versions, that commit is the only difference.

I can send info about my system etc if that's helpful, not sure what all you would need.

wiedemeyer, 2015-08-26:
Hi,
sounds like the issue #133. Do you subscribe to topics and then it crashes, or does it crash even without? What commands are you running? Is it also crashing if you just `rosrun kinect2_bridge kinect2_bridge`? If this also crashed, build it in debug, start it in gdb and let it run until it crashes, to see where the error might be.

hersh, 2015-08-26:
Yup, probably a duplicate.  I just ran master as ```roslaunch kinect2_bridge kinect2_bridge.launch```, let it run (without crashing) for a minute or so, then ran rviz to view the clouds, and it crashed after a couple seconds.

I'll see if I can get a stack trace next...

hersh, 2015-08-26:
Closed as duplicate of #133.

closed 2015-08-26 by hersh