code-iai/iai_kinect2 #135: libopencv_hal.a doesn't exist

ankitbhatia, 2015-08-26:
I'm trying to install iai_kinect2 package on ROS inside catkin_ws.

When I make catkin_make, I encounter the following error:

CMake Error at /usr/local/share/OpenCV/OpenCVModules.cmake:117 (message):
  The imported target "opencv_hal" references the file

     "/usr/local/lib/libopencv_hal.a"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/local/share/OpenCV/OpenCVModules.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  /usr/local/share/OpenCV/OpenCVConfig.cmake:86 (include)
  vision_opencv/image_geometry/CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/satellite/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/satellite/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

How can I fix this error?

Thanks,

wiedemeyer, 2015-08-26:
I would guess something with you OpenCV installation is wrong. Please install the version that comes with Ubuntu.

ankitbhatia, 2015-08-26:
I fixed the issue. Thanks!

closed 2015-08-26 by wiedemeyer