code-iai/iai_kinect2 #136: Jade Question

lakehanne, 2015-08-27:
Hey,

I know you did not build this package to work with jade but I was fiddling a little bit with it to see the result of compiling with jade. It turns out that it could work if I know how to lock onto the compiled `Protonect shared object libbrary`. This is what I got when I tried compiling with jade:

```
/usr/local/lib/libfreenect2.so: undefined reference to `libusb_free_ss_endpoint_companion_descriptor'
/usr/local/lib/libfreenect2.so: undefined reference to `libusb_get_ss_endpoint_companion_descriptor'
collect2: error: ld returned 1 exit status
iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/build.make:185: recipe for target '/home/lex/catkin_jade_ws/devel/lib/kinect2_bridge/kinect2_bridge' failed
make[2]: *** [/home/lex/catkin_jade_ws/devel/lib/kinect2_bridge/kinect2_bridge] Error 1
```

My question: do you have an advice on how I can make kinect2_bridge rightly link to `libfreenect2.so` without all the dependency errors I have above?

Looking forward to your reply.

lakehanne, 2015-08-27:
Pardon me, the error was because I installed `freenect` from `libfreenect` earluy on and this had installed its version of libusb in my `/usr/local/lib` directory. After purging that directory of all `libusb*` files, I reinstalled the latest `libfreenect2`, recompiled iai-kinect 2 with jade and BAM! It works!

You could update the README with support for jade, methinks!

wiedemeyer, 2015-08-28:
Good to know. Thanks!

closed 2015-08-29 by lakehanne