code-iai/iai_kinect2 #137: OpenCL Errors on Ros Jade/Ubuntu 15.04.

lakehanne, 2015-08-30:
Sorry to bring my problems again. 

I got kinect2 working on Thursday on Ubuntu 15.04 after a fresh install but found that after I compiled opencv with cuda on, I started experiencing issues with catkin_make.

Now, I have compiled Protonect and ran it in OpenCL, confirmed it worked fine on Ubuntu 15.04 running ROS Jade. 

But somehow, I could not get iai_kinect2 to run since yesterday. I have compiled OpenCL from beignet, tried uninstalling and reinstalling opencl-headers, ocl-icd-opencl, and also uninstalled all of the above to install the opencl official sdks from [intel](https://software.intel.com/en-us/articles/opencl-drivers#ubuntu64) but nothing is helping. I have exported the LD_LIBRARY_PATH and again, Protonect compiles fine with `Protonect cl`. Here's the output of `catkin_make`:
```
/usr/local/lib/libfreenect2.so: undefined reference to `clReleaseMemObject@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clRetainEvent@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clReleaseKernel@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clReleaseCommandQueue@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clReleaseEvent@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clEnqueueWriteBuffer@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clCreateBuffer@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clBuildProgram@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clSetKernelArg@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clRetainKernel@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clCreateContext@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clReleaseProgram@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clGetDeviceIDs@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clGetPlatformInfo@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clRetainCommandQueue@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clCreateProgramWithSource@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clCreateKernel@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clGetDeviceInfo@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clReleaseContext@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clRetainContext@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clCreateCommandQueue@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clEnqueueNDRangeKernel@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clWaitForEvents@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clEnqueueReadBuffer@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clRetainProgram@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clGetProgramBuildInfo@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clGetPlatformIDs@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clRetainMemObject@OPENCL_1.0'
collect2: error: ld returned 1 exit status
iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/build.make:189: recipe for target '/home/lex/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge' failed
make[2]: *** [/home/lex/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge] Error 1
CMakeFiles/Makefile2:1574: recipe for target 'iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/all' failed
make[1]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/all] Error 2
```

FWIW, here's my `clinfo`: 

```
X server found. dri2 connection failed! 
Number of platforms                               3
  Platform Name                                   NVIDIA CUDA
  Platform Vendor                                 NVIDIA Corporation
  Platform Version                                OpenCL 1.1 CUDA 7.0.35
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_copy_opts 
  Platform Extensions function suffix             NV

  Platform Name                                   Intel Gen OCL Driver
  Platform Vendor                                 Intel
  Platform Version                                OpenCL 1.2 beignet 1.0.3
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_icd
  Platform Extensions function suffix             Intel
X server found. dri2 connection failed! 
X server found. dri2 connection failed! 

  Platform Name                                   Intel(R) OpenCL
  Platform Vendor                                 Intel(R) Corporation
  Platform Version                                OpenCL 1.2 LINUX
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_icd cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_depth_images cl_khr_3d_image_writes cl_intel_exec_by_local_thread cl_khr_spir cl_khr_fp64 
  Platform Extensions function suffix             INTEL
[... 239 lines cut ...]
  Execution capabilities                          
    Run OpenCL kernels                            Yes
    Run native kernels                            Yes
    SPIR versions                                 1.2
  Prefer user sync for interop                    No
  printf() buffer size                            1048576 (1024KiB)
  Built-in kernels                                
  Device Available                                Yes
  Compiler Available                              Yes
  Linker Available                                Yes
  Device Extensions                               cl_khr_icd cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_depth_images cl_khr_3d_image_writes cl_intel_exec_by_local_thread cl_khr_spir cl_khr_fp64 

NULL platform behavior
  clGetPlatformInfo(NULL, CL_PLATFORM_NAME, ...)  No platform
  clGetDeviceIDs(NULL, CL_DEVICE_TYPE_ALL, ...)   No platform
  clCreateContext(NULL, ...) [default]            No platform
  clCreateContext(NULL, ...) [other]              Success [NV]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  No platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  No platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ACCELERATOR)  No platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CUSTOM)  No platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  No platform
```

Would appreciate any help~!

lakehanne, 2015-08-30:
Ah, I think I found the error:

Cmake was warning of libOpenCL.so library conflicts at `/usr/local/cuda/lib64/` and 
`/usr/lib/x86_64-linux-gnu/libOpenCL.so`. After probing each directory's file with 

>file libOpenCL.so

I found that the only one that seemed consequential was the one in `x86_64-linux-gnu` as it had the right symbolic link to /etc/alternatives/opencl-libOpenCL.so. 

So I moved the one in `cuda` directory to a temporary directory, ran `sudo ldconfig` and now my code compiles! But in executing the viewer, I ran into segmentation faults. A quick gdb debugger session reveals the following:

```
~/catkin_ws/devel/lib/kinect2_viewer$ gdb ./kinect2_viewer core
GNU gdb (Ubuntu 7.9-1ubuntu1) 7.9
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./kinect2_viewer...(no debugging symbols found)...done.
[New LWP 6262]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `./kinect2_viewer'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007fd8f73aefcc in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
(gdb) run
Starting program: /home/lex/catkin_ws/devel/lib/kinect2_viewer/kinect2_viewer 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007fffd4c95fcc in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
(gdb) bt
#0  0x00007fffd4c95fcc in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#1  0x00007ffff7de95ba in call_init (l=<optimized out>, argc=argc@entry=1, 
    argv=argv@entry=0x7fffffffd9b8, env=env@entry=0x7fffffffd9c8)
    at dl-init.c:72
#2  0x00007ffff7de96cb in call_init (env=<optimized out>, 
    argv=<optimized out>, argc=<optimized out>, l=<optimized out>)
    at dl-init.c:30
#3  _dl_init (main_map=0x7ffff7ffe188, argc=1, argv=0x7fffffffd9b8, 
    env=0x7fffffffd9c8) at dl-init.c:120
#4  0x00007ffff7dd9d0a in _dl_start_user () from /lib64/ld-linux-x86-64.so.2
#5  0x0000000000000001 in ?? ()
#6  0x00007fffffffdd58 in ?? ()
#7  0x0000000000000000 in ?? ()
```

lakehanne, 2015-08-31:
Any luck @wiedemeyer ?

wiedemeyer, 2015-10-14:
Hi, sorry for the late answer. Please compile it in debug and check again.

closed 2015-10-26 by wiedemeyer