code-iai/iai_kinect2 #140: Sensor diagnostics/status

bennilson, 2015-09-08:
I'm using the Kinect as part of a larger robotic application. I'm looking for a way to detect if the Kinect is functioning properly or if there is an issue with it, so that I can send a message to the user to check the Kinect. In this case, user will not beable to see the terminal window to check if the kinect is functioning or not. From what I’ve seen, there is no topic output about the status of the sensor that I could capture. Would it be possible get such functionality or would it need something from the libfreenect2?

wiedemeyer, 2015-09-09:
Hi,
there is no status. If something does not work, the node will not start. So you could check if the topics are advertised and are sending data.

bennilson, 2015-09-09:
My thought was more after the node started, but something happened to the connection to casue it to stop. But, the idea of polling the topics for data and a watchdog timer would work.

closed 2015-09-18 by wiedemeyer