code-iai/iai_kinect2 #141: cl::Error while running kinect_bridge2 on indigo, Ubuntu 14.04

wkentaro, 2015-09-09:
```
% rosrun kinect2_bridge kinect2_bridge
parameter:
        base_name: kinect2
           sensor:
        fps_limit: -1
       calib_path: /home/baxter/ros/indigo/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

terminate called after throwing an instance of 'cl::Error'
  what():  clGetPlatformIDs
[1]    4089 abort      rosrun kinect2_bridge kinect2_bridge
```

I tried `$ sudo ./Protonect cl`, `$ ./Protonect` and it works.

wiedemeyer, 2015-09-09:
try to install `nvidia-modprobe`

wkentaro, 2015-09-09:
That works!
Thank you for your quick response. That helps me a lot.

Should I write some information about this on README?

wkentaro, 2015-09-09:
[FIXED]
I just tried
```
$ sudo aptitude install nvidia-modprobe
```

closed 2015-09-09 by wkentaro