code-iai/iai_kinect2 #142: Why does PointCloud2 display in Rviz wrongly?

irazr, 2015-09-11:
Hi all,
i want to display pointcloud2 in rviz, i followed these steps:
(1) roslaunch kinect2_bridge kinect2_bridge.launch reg_method:=cpu depth_method:=cpu publish_tf:=true
(2) start rviz and add pointcloud2, choose /kinect2/sd/points topic, then rviz displayed like:
[image]
So what's wrong with this pointcloud2 image, or did i miss something? I suspect it's due to the tf, any other idea?
Thanks in advance.

wiedemeyer, 2015-09-11:
Hi,
looks like the pointcloud is displayed correctly. What is the issue? If you what to look at it from the front you might want to rotate the view in RVIZ. In RVIZ `z` is pointing up, thats why the pointcloud is displayed that way.

irazr, 2015-09-14:
Hi Wiedemeyer,
thanks for you reply, the pointcloud2 image in rviz seemed not so clear as in kinect2_viewer. 
Today I run the above commands, after a few seconds kinect2_bridge crashed and the error is:

*** Error in `/opt/ros/indigo/lib/nodelet/nodelet': corrupted double-linked list: 0x00007fcb080008b0 ***
[kinect2-2] process has died [pid 13171, exit code -6, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=kinect2 __log:=/home/robot/.ros/log/6a3ffa08-5aad-11e5-ad6d-f80f41a4a4bd/kinect2-2.log].
log file: /home/robot/.ros/log/6a3ffa08-5aad-11e5-ad6d-f80f41a4a4bd/kinect2-2*.log
[kinect2_points_xyzrgb_qhd-5] process has finished cleanly
log file: /home/robot/.ros/log/6a3ffa08-5aad-11e5-ad6d-f80f41a4a4bd/kinect2_points_xyzrgb_qhd-5*.log
[kinect2_points_xyzrgb_qhd-5] restarting process
process[kinect2_points_xyzrgb_qhd-5]: started with pid [13550]
[kinect2_points_xyzrgb_hd-6] process has finished cleanly
log file: /home/robot/.ros/log/6a3ffa08-5aad-11e5-ad6d-f80f41a4a4bd/kinect2_points_xyzrgb_hd-6*.log
[kinect2_points_xyzrgb_hd-6] restarting process
process[kinect2_points_xyzrgb_hd-6]: started with pid [13571]
[ INFO] [1442213692.126087456]: Bond broken, exiting
[kinect2_bridge-3] process has finished cleanly
log file: /home/robot/.ros/log/6a3ffa08-5aad-11e5-ad6d-f80f41a4a4bd/kinect2_bridge-3*.log
[kinect2_bridge-3] restarting process
process[kinect2_bridge-3]: started with pid [13627]
[ INFO] [1442213692.439563073]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1442213692.443229488]: waitForService: Service [/kinect2/load_nodelet] could not connect to host [robot-Lenovo-Product:51356], waiting...
[kinect2_points_xyzrgb_sd-4] process has finished cleanly
log file: /home/robot/.ros/log/6a3ffa08-5aad-11e5-ad6d-f80f41a4a4bd/kinect2_points_xyzrgb_sd-4*.log
[kinect2_points_xyzrgb_sd-4] restarting process
process[kinect2_points_xyzrgb_sd-4]: started with pid [13672]

wiedemeyer, 2015-09-14:
Hi,
for me it is running fine. But you should try to use OpenCL or OpenGL, it is much faster. Anyway, did you changed something, or why did this error occur? It seems to run fine a few days ago.

irazr, 2015-09-14:
Hi,
last week I tested ./Protonect cl, ./Protonect gl and ./Protonect cpu, cl and gl only showed rgb image and cpu worked, so I use it. Today I tried to install Beignet to use opencl, but it failed.

irazr, 2015-09-15:
I reinstall ubuntu and everything works.

closed 2015-09-15 by irazr