code-iai/iai_kinect2 #143: Frame Rate Issue with Kinect 2

shakith, 2015-09-11:
I am trying to understand whether maximum frame rate of 30 fps can be achieved.

I ran image kinect2_viweer with the kinect2_bridge after installing iai_linect2 and libfreenect2.

Even though the GUI window shows around 30 fps, console prints the publishing rate as 22 -25 fps for RGB (with 29 fps for depth). Is this the maximum that can be reached? I am using it in good daylight conditions. My notebook has a usb3 controller from intel (no other usb devices connected). I am using the special usblib with libfreenect2 on my ubuntu 14.04 setup.

Also here (http://www.ros.org/news/2014/09/microsoft-kinect-v2-driver-released.html ) it mentions that the driver was optimized to full framerate with parallelization of the image pipeline? Is there a special patch to libfreenect2 driver?

As far I understand the optimizations to iai kinect compared to libfreenect2 are:
- different registration function compared to libfreenect2 with an opencl implementation.
- time triggered capture to maintain frame rates
- multi-threaded implementation
Is there more optimizations?

Thanks for the explanations.

wiedemeyer, 2015-09-11:
Hi,
achieving 30 fps depends on the hardware and the options you are using. If you are using CPU depth processor you are not able to get 30 fps. While OpenGL or OpenCL depth processors are capable of providing 30 fps on normal hardware, like the Intel GPUs.
The other part is the CPU usage. For the color image decoding you need much CPU power, but that should be no problem for decent laptops. But if the CPU is stressed by other things as well, the frame rate might drop. You could watch the CPU load with `htop` or a system monitor. To check what your hardware is capable of, please don't start any viewer. Just run `kinect2_bridge` and then open two terminals and start `rostopic hz <depth topic>` and `rostopic hz <color topic>` in parallel.
If you are using compression for image transport you will need a lot more CPU power then without, but are using more network bandwidth. On a local machine you should not use compression. By default the `kinect2_viewer` is not using compression.
`kinect2_bridge` is the bridge between libfreenect2 and ROS. The transition into the ROS world needs a lot of stuff, like converting the images into another format, rectification, compression for network transport, etc. This part is multi-threaded and integrated into `kinect2_bridge`, because with the standard ROS library it's not possible to get 30 fps. And instead of just the libfreenct2s `color to depth` registration, it also adds a `depth to color`, so you could get edge aware interpolated and upscaled depth images in HD resolution.
And regarding the news entry: At that time I added some optimizations to libfreenect2, to make it faster. But they have been integrated to libfreenect2 a long time ago. So that news is outdated.

shakith, 2015-09-17:
Without the viewer, the topic publish rate is 30 fps indeed.  Thanks for the clarification.

closed 2015-09-18 by wiedemeyer