code-iai/iai_kinect2 #144: Jetson with iai_kinect2 : JPEG parameter struct mismatch

vishu2287, 2015-09-16:
After following HOWTO, kinect2 works well with libfreenect2 and now trying to work with iai_kinect2. While compiling the code faced lot of problems and got good suggestions from xlz and compiled well. 
Now while running kinect2_bridge getting error as 

" terminate called after throwing an instance of 'std::runtime_error' what(): JPEG parameter struct mismatch: library thinks size is 488, caller expects 560"

I  tried by turning off ENABLE_TEGRA_JPEG  in cmakelists of protonect , but getting same error
and even changed the cmakelist of kinect2_bridge by keeping freenect2_library at first but still getting same error. 

target_link_libraries(kinect2_bridge
  ${freenect2_LIBRARY}
  ${catkin_LIBRARIES}
  ${OpenCV_LIBRARIES}
)

Can anyone please help me.....

vishu2287, 2015-09-28:
I have removed tegra jpeg dependency in cmakelist.txt, now its working fine but facing another problem.
When I rosrun kinect2_bridge kinect2_bridge
waiting for client, 
once clients get connected kinect2_bridge leads to segmentation fault

wiedemeyer, 2015-10-14:
I can't really help with platform specific issues that are not related to a x64 Ubuntu. Did you figured out to fix the problem?

closed 2015-10-26 by wiedemeyer

vishu2287, 2015-10-28:
its libusb issue. 
the libusb used by libfreenect2 is different with kinect2 libusb. Now its working fine