code-iai/iai_kinect2 #145: error in kinect2_registration compile : could not find CUDA: Found unsuitable version "4.2", but required is exact version "5.5" (found /usr/local/cuda)

YuweiGuo, 2015-09-18:
hallo,
i have this problem, when i try to compile the iai-kinect2 in my rosworkspace.

"
could not find CUDA: Found unsuitable version "4.2", but required is exact version "5.5" (found /usr/local/cuda) 
"

i have tried to install the cuda and toolkit , put the CUDA_TOOLKIT_ROT_DIR and CUDA_BIN_PATH in the .bashrc. but it did not work.
my enviroment is , ros(hydro) ,  ubuntu 14.04.
and the libfreenet2 worked well.

can you help me ? thank you!

ps: i have 2 graphic card in my computer, nvidia and intel Ivybridge Mobile  

yuwei

wiedemeyer, 2015-09-18:
Hi,
are you sure it's an issue of iai_kinect2? Because it is not using CUDA.

YuweiGuo, 2015-09-18:
I think so too,it do not use cuda . But the error happens,while I run catkin make in the workspace .

wiedemeyer, 2015-09-18:
Maybe there is something else that required CUDA.

closed 2015-10-14 by wiedemeyer