code-iai/iai_kinect2 #146: GL_MAX_RECTANGLE_TEXTURE_SIZE is too small: 0

lucbettaieb, 2015-09-18:
Hello,
Most of the time when I try to run `rosrun kinect2_bridge kinect2_bridge` I get the following error:
`[Error] [Texture<FormatT>] GL_MAX_RECTANGLE_TEXTURE_SIZE is too small: 0`

The entire stdout is:
```
luc@atlas4-397:~$ rosrun kinect2_bridge kinect2_bridge _fps_limit:=2
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: 2
       calib_path: /home/luc/indigo/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 12 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:5 with serial 003668550247
[Info] [Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 003668550247 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened

starting kinect2

[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started

device serial: 003668550247
device firmware: 2.3.3913.0.7
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...
[Info] [Freenect2DeviceImpl] stopped

default ir camera parameters: 
fx 366.398, fy 366.398, cx 260.717, cy 209.796
k1 0.0916684, k2 -0.271591, p1 0, p2 0, k3 0.0934514

default color camera parameters: 
fx 1081.37, fy 1081.37, cx 959.5, cy 539.5
using sensor defaults for color intrinsic parameters.
using sensor defaults for ir intrinsic parameters.
using defaults for rotation and translation.
using defaults for depth shift.

camera parameters used:
camera matrix color:
[1081.3720703125, 0, 959.5;
  0, 1081.3720703125, 539.5;
  0, 0, 1]
distortion coefficients color:
[0, 0, 0, 0, 0]
camera matrix ir:
[366.3981018066406, 0, 260.7166137695312;
  0, 366.3981018066406, 209.7958984375;
  0, 0, 1]
distortion coefficients ir:
[0.09166838228702545, -0.2715910077095032, 0, 0, 0.09345144033432007]
camera matrix depth:
[366.3981018066406, 0, 260.7166137695312;
  0, 366.3981018066406, 209.7958984375;
  0, 0, 1]
distortion coefficients depth:
[0.09166838228702545, -0.2715910077095032, 0, 0, 0.09345144033432007]
rotation:
[1, 0, 0;
  0, 1, 0;
  0, 0, 1]
translation:
[0; 0; 0]
depth shift:
0

[DepthRegistration::New] Using CPU registration method!
[DepthRegistration::New] Using CPU registration method!
[kinect2_bridge] waiting for clients to connect

[kinect2_bridge] client connected. starting device...
[Info] [Freenect2DeviceImpl] starting...
[Error] [Texture<FormatT>] GL_MAX_RECTANGLE_TEXTURE_SIZE is too small: 0
```

Any idea as to what's going on?

wiedemeyer, 2015-09-19:
Hi,
that issue is related to OpenGL. It comes from the OpenGL depth processor from libfreenect2. You could try to use OpenCL instead or ask on the libfreenect2 issues for help.

lucbettaieb, 2015-09-22:
Thanks a lot for the info!

closed 2015-09-22 by lucbettaieb