code-iai/iai_kinect2 #147: How to display the depth image or depth pointcloud only?

Cynthia0217, 2015-09-21:
Hi, I am new to Kinect 2.0 and ROS. Now I face a problem, After calibration, using Kinect_viewer I can obtain image and pointcloud images succesfully, the problem is I don't know the  mode: 'qhd', 'hd', 'sd' or 'ir' exactly stands for what kind of image. If I want to obtain the depth image only (not superimposed on the RGB image), how can I get one?

Thanks a lot!

wiedemeyer, 2015-09-22:
The `kinect2_viewer` has no option to only show the depth image. For viewing single image streams you could use `image_view`, but it might not work for 16 bit depth images.
The modes are different resolutions, for details look at the documentation of `kinect2_bridge`. And ir uses the IR image instead of the color image.

Cynthia0217, 2015-09-24:
@wiedemeyer Thank you for explanation. Best Regards~

closed 2015-09-25 by wiedemeyer