code-iai/iai_kinect2 #148: Frame Transforms

dekent, 2015-09-21:
I noticed that the point cloud topics are based in different frames, e.g. the hd and qhd clouds are in the kinect2_rgb_optical_frame and the sd cloud is in kinect2_ir_optical frame.  What's the transformation between these frames?  Is that part of what's calculated in the extrinsic calibration, and otherwise, is there a urdf or something similar that gives the relationship between these frames?

Thanks!

wiedemeyer, 2015-09-22:
The transform between the frames are the extrinsic parameters. After the calibration they are stored in `calib_pose.yaml`. If you run kinect2_bridge with `publish_tf:=true` it will publish the transformation on `/tf`. You could also use the transformation from `calib_pose.yaml` to create a URDF, but you need to convert them to the format URDF needs.

dekent, 2015-09-22:
Great, thanks for the quick response!

closed 2015-09-22 by dekent