code-iai/iai_kinect2 #149: Can we expect any Jetson TK1 Release?

filipetrocadoferreira, 2015-09-22:
I'm facing several problems compiling and running iai-kinect2 with the Jetson TK1 board despite of being able to run Protonect with good results.

Can I expect any release soon?

vishu2287, 2015-09-23:
are you able to compile iai_kinect2?

filipetrocadoferreira, 2015-09-23:
I am. Only after copy the files from registration.h to the source code. I guess actual commit does not have this part because it's not compatible with CUDA

kerolos, 2015-09-25:
which opencv , are you working with? , maybe you installed two openCVs ? ,if so you should deactivate one of them.

filipetrocadoferreira, 2015-09-25:
I'm working with the original from JETPACK- Opencv4tegra. I still have many doubts about which version I should use in this board. 
Which version is the most indicated to this setup?

xlz, 2015-09-29:
I am the author of the Jetson TK1 fork.

The trouble you are having is that libfreenect2 upstream has got many updates and bug fixes. So iai_kinect2 is updated accordingly. But my Jetson fork of libfreenect2 is not updated. So if you use an old libfreenect2 with the new iai_kinect2, problem surely will happen.

The Jetson fork contains a CUDA depth processor and a Tegra JPEG decoder. If you disable any of them there is no point in using my fork. You can just try the upstream one and see how it works (should work, albeit slowly). These two features are waiting on upstream prior work to be finished. libfreenect has to first publish 0.1 release for these to move forward.

wiedemeyer, 2015-10-14:
Since I don't own a Jetson, I can't provide a release, therefore someone else has to do and maintain it. For now the answer would be: no/maybe, if someone volunteers.

closed 2015-10-14 by wiedemeyer

Abbyls, 2016-09-06:
@xlz @wiedemeyer 
Hi, i wonder is there any progress in converting the iai_kinect2 code to cuda? As the jetson can't support the Opencl.

wiedemeyer, 2016-09-06:
No, there is no plan to do so. There is a CPU implementation that should work fine. For the very demanding depth processing in libfreenect2 is a CUDA implementation available, which iai_kinect2 is using.
But you are free to provide a pull request with a CUDA port of the OpenCL code from iai_kinect2.

Abbyls, 2016-09-06:
ok, i will have a try.