code-iai/iai_kinect2 #151: Problem in kinect2 Viewer

kerolos, 2015-09-24:
Hello , 

i have an error , when i run kinect2_viewer "Segmentation fault( core dumped)" 
but when i run this code "rosrun kinect2_bridge osrun kinect2_bridge ",
it appears that :+1: 

#######################################################################
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/kerolos/kinect/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl]   1: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Warning] [OpenCLDepthPacketProcessorImpl] Your ocl-icd has deadlock bugs. Update to 2.2.4+ is recommended.
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:5 with serial 039374243547
[Info] [Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 039374243547 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened

starting kinect2

[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started

device serial: 039374243547
device firmware: 4.3.3913.0.7
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...
[Info] [Freenect2DeviceImpl] stopped

default ir camera parameters: 
fx 365.604, fy 365.604, cx 255.156, cy 207.491
k1 0.0897843, k2 -0.271625, p1 0, p2 0, k3 0.0963501

default color camera parameters: 
fx 1081.37, fy 1081.37, cx 959.5, cy 539.5
using sensor defaults for color intrinsic parameters.
using sensor defaults for ir intrinsic parameters.
using defaults for rotation and translation.
using defaults for depth shift.

camera parameters used:
camera matrix color:
[1081.3720703125, 0, 959.5;
  0, 1081.3720703125, 539.5;
  0, 0, 1]
distortion coefficients color:
[0, 0, 0, 0, 0]
camera matrix ir:
[365.6040954589844, 0, 255.1558074951172;
  0, 365.6040954589844, 207.4907989501953;
  0, 0, 1]
distortion coefficients ir:
[0.08978427946567535, -0.2716245949268341, 0, 0, 0.09635014086961746]
rotation:
[1, 0, 0;
  0, 1, 0;
  0, 0, 1]
translation:
[-0.052;
  0;
  0]
depth shift:
0

[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistrationOpenCL::listDevice]  devices:
  0: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
  1: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
[DepthRegistrationOpenCL::init]  selected device: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
[DepthRegistrationOpenCL::listDevice]  devices:
  0: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
  1: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
[DepthRegistrationOpenCL::init]  selected device: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
[kinect2_bridge] waiting for clients to connect
#######################################################################

do you know what is error comes from ? and how can i fixed it .

 thanks in advance .

 BR,
kerollos

lucbettaieb, 2015-09-24:
Did you set up udev permissions?

wiedemeyer, 2015-09-25:
Hi,
I don't see any errors. kinect2_bridge started and is waiting for clients to start listening to topics.

kerolos, 2015-09-25:
hello,
what about the error appeared when i run(rosrun kinect2_viewer kinect2_viewer ) -------> "Segmentation fault ( core dumped)" ?
how can i fix this error and where it is coming from?
thanks in advance

wiedemeyer, 2015-09-25:
The error message is not very helpful. Could you build it in debug with `catkin_make -DCMAKE_BUILD_TYPE="Debug"`. Then run it with gdb `gdb <catkin_ws>/devel/lib/kinect2_viewer/kinect2_viewer` and paste the stack trace here.

kerolos, 2015-09-25:
this is track: 

Starting program: /home/kerolos/kinect/devel/lib/kinect2_viewer/kinect2_viewer 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007fffe90a0dfc in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
(gdb) backtrace 
#0  0x00007fffe90a0dfc in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#1  0x00007ffff7dea13a in call_init (l=<optimized out>, argc=argc@entry=1, 
    argv=argv@entry=0x7fffffffd908, env=env@entry=0x7fffffffd918)
    at dl-init.c:78
#2  0x00007ffff7dea223 in call_init (env=<optimized out>, 
    argv=<optimized out>, argc=<optimized out>, l=<optimized out>)
    at dl-init.c:36
#3  _dl_init (main_map=0x7ffff7ffe1c8, argc=1, argv=0x7fffffffd908, 
    env=0x7fffffffd918) at dl-init.c:126
#4  0x00007ffff7ddb30a in _dl_start_user () from /lib64/ld-linux-x86-64.so.2
#5  0x0000000000000001 in ?? ()
#6  0x00007fffffffdd3f in ?? ()
#7  0x0000000000000000 in ?? ()

wiedemeyer, 2015-09-28:
Could it be, that you have an custom opencv installed?

kerolos, 2015-09-29:
yes, i have custom openCV .

wiedemeyer, 2015-09-29:
I guess there is something wrong with your custom OpenCV installation, maybe some linking or configuration issues. It should work fine with the OpenCV version from the official repositories. Could you try it out with the official release.

NukeVol22, 2015-10-08:
I am also having a segmentation fault with the viewer, but for a different reason:

```terminal
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Traceback (most recent call last):
  File "/usr/share/gdb/auto-load/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.19-gdb.py", line 63, in <module>
    from libstdcxx.v6.printers import register_libstdcxx_printers
ImportError: No module named 'libstdcxx'
[New Thread 0x7fffc7ea2700 (LWP 30371)]
topic color: /kinect2/qhd/image_color_rect
topic depth: /kinect2/qhd/image_depth_rect
[New Thread 0x7fffc6967700 (LWP 30374)]
[New Thread 0x7fffc6166700 (LWP 30375)]
[New Thread 0x7fffc5965700 (LWP 30376)]
[New Thread 0x7fffc5164700 (LWP 30381)]
starting receiver...
[New Thread 0x7fffb7fff700 (LWP 30408)]
[New Thread 0x7fffaf7fe700 (LWP 30409)]
[New Thread 0x7fffb77fe700 (LWP 30410)]
[New Thread 0x7fffb6ffd700 (LWP 30411)]

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff493c8d7 in vtkXOpenGLRenderWindow::MakeCurrent() () from /usr/lib/libvtkRendering.so.5.8
```

It is clear that I'm still getting a vtk problem (see Issue #161), but the program is at least compiling at this stage. As I understand it, VTK 5.8 is needed for the current stable release of PCL. Any further suggestions for this? Thanks.

kerolos, 2015-10-09:
I reinstalled ubuntu and everything works well.

closed 2015-10-09 by wiedemeyer