code-iai/iai_kinect2 #153: Does the kinect2_bridge publish the actual RGBXYZ point cloud?

soulslicer, 2015-09-25:
It doesn't right? The bridge only publishes the disparity/depth maps and other camera outputs. I noticed that the Kinect 2 Viewer actually performs the computation to generate the point cloud from the RGB and Depth image, using OpenMP.

Does this mean I need to write my won ROS node to output the point cloud, referencing your implementation?

soulslicer, 2015-09-25:
Okay, so looks like there is a kinect2_registration package. But there doesn't seem to be any examples on how to interface with it

wiedemeyer, 2015-09-25:
Hi,
the kinect2_bridge node does not publish point clouds itself, point clouds are published by nodelets from depth_image_proc. They will get started if you start kinect2_bridge with `roslaunch kinect2_bridge kinect2_bridge.launch`.
But if you transferring data over the network, it is recommended to use the compressed images instead of the point clouds. Point clouds are much bigger than the images and you won't get more than 10 fps when using point clouds over a gigabit lan.

soulslicer, 2015-09-28:
Ic. I guess you are talking about this one?
http://wiki.ros.org/depth_image_proc

So I have to specify in a launch file of mine the parameters similar to this example?
https://gist.github.com/bhaskara/2400165 

Is there any example of this?

wiedemeyer, 2015-09-28:
Yes, just launch the provided launchfile and you will get xyzrgba clouds.

shakith, 2015-09-28:
Is it possible to view this point cloud using rviz? rviz is gving error looking for a fixed frame.

wiedemeyer, 2015-09-29:
you need to enable publishing tf by `roslaunch .... publish_tf:=true`

kerolos, 2015-10-09:
Hello Wiedemeyer,
i got an Error, when i displayed Pointcloud2 using Rviz, "fixed frame(map)does not exist". the topic was /Kinect2/qhp/points.i run those lines 1)kinect2_bridge kinect2_bridge.launch publish_tf:=true, 2)rosrun rviz rviz.

wiedemeyer, 2015-10-09:
You have to change the frame in rviz. Default is `map`, change it to `kinect2_link` and it should work.

closed 2015-10-14 by wiedemeyer

ysl208, 2018-02-02:
> you need to enable publishing tf by roslaunch .... publish_tf:=true
could you add this to the documentation if you havent already?