code-iai/iai_kinect2 #154: Install instruction breaks because of libfreenect2's changes

wkentaro, 2015-09-27:
About: https://github.com/code-iai/iai_kinect2#install

no `examples/protonect` in libfreenect2.
https://github.com/OpenKinect/libfreenect2/tree/master/examples

And header files are not found,
```
% mkdir build
% cd build
% cmake ../examples
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26")
-- checking for module 'glfw3'
--   found glfw3, version 3.0.4
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/x86_64-linux-gnu/libX11.so
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so
-- Configuring done
-- Generating done
-- Build files have been written to: /home/wkentaro/tech/libfreenect2/build
% make
Scanning dependencies of target Protonect
[ 33%] Building CXX object CMakeFiles/Protonect.dir/Protonect.cpp.o
/home/wkentaro/tech/libfreenect2/examples/Protonect.cpp:36:33: fatal error: libfreenect2/logger.h: No such file or directory
 #include <libfreenect2/logger.h>
                                 ^
compilation terminated.
make[2]: *** [CMakeFiles/Protonect.dir/Protonect.cpp.o] Error 1
make[1]: *** [CMakeFiles/Protonect.dir/all] Error 2
make: *** [all] Error 2
```

vishu2287, 2015-09-28:
which libfreenect2 you are using?

wkentaro, 2015-09-28:
OpenKinect/libfreenect2

vishu2287, 2015-09-28:
try to rebuild libfreenect2, 
$ <path to libfreenect2>/libfreenect2/examples/build
$ make clean
$ cmake ..
$ make
$ sudo make install

"sudo make install" makes the include folder available to all the packages

wkentaro, 2015-09-28:
actually it fails at building for libfreenect2.
```
cd libfreenect2/examples
mkdir build
cd build
cmake ..
make  # This fails
sudo make install
```

vishu2287, 2015-09-28:
can you check logger.h file in libfreenect2/examples/protonect/include/libfreenect2  is present?

wkentaro, 2015-09-28:
it seems moved to include dir at root.
https://github.com/OpenKinect/libfreenect2/tree/master/include/libfreenect2

JimmyDaSilva, 2015-09-28:
I was just facing the same issue right now.
I managed to compile and make it work for me.
It seems that the new way to make & install now is
```
cd libfreenect2
mkdir build
cmake .. #add -DENABLE_OPENCL or -DENABLE_OPENGL if you want to use your GPU
make 
sudo make install
```
Protonect executable is now in ```libfreenect2/build/bin```

wkentaro, 2015-09-28:
Thanks!
That works and I can see all images by running ./Protonect.

However I have another problem described at here. #104

Anyway this issue has been fixed.

2015年9月28日月曜日、Jimmy DA SILVA<notifications@github.com

closed 2015-09-28 by wkentaro

pallavbakshi, 2017-01-12:
Hi,

I installed the libfreenect2 on 11 Jan 2016 as per the instructions mentioned on the github page of libfreenect2. You can find two bin folders - one is in libfreenect2 and the other one inside of build. The Protonect example is inside the build -> bin. Just make sure to do - ./build/bin/Protonect

Thanks! Just trying to help

commit b0d81f0e (2015-10-12): Follow OpenKinect/libfreenect2 change in README

Closes #154
PR #155 (2015-10-12): Follow OpenKinect/libfreenect2 change in README
Closes #154