code-iai/iai_kinect2 #157: kinect2_bridge segmentation fault

vishu2287, 2015-09-29:
I am able to run ./Protonect and can view color, ir and depth images with out any error.
When I am trying to run kinect2_bridge from the root, it asks for clients. If I provide kinect2/sd/image_color_rect, it produces the messages but if I provide other topics like kinect2/qhd/image_color,  the following error genetares

root@tegra-ubuntu:~/catkin_ws/devel# rosrun kinect2_bridge kinect2_bridge
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/ubuntu/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[Freenect2Impl] enumerating devices...
[Freenect2Impl] 10 usb devices connected
[Freenect2Impl] found valid Kinect v2 @2:5 with serial 017275643547
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 017275643547 (selected)
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

starting kinect2

[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    ............C..A
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    .!3U... ........
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    .......... .....
0x0040:  31 33 00 00 00 04 0a 0d 47 4d 42 30 31 33 2e 31    13......GMB013.1
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                X...........

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  23 26 00 00                                        #&..

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  23 26 00 00                                        #&..

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[... 328 lines cut ...]
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[kinect2_bridge] depth processing: ~nanHz (nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~127.059Hz (7.87038ms) publishing rate: ~14.6242Hz
Segmentation fault

I am using Jetson TK1 board .

cpu ang gpu performance were altered with maximum efficiency

1. Why lots of "skipping rgb packet!" and " [TransferPool::submit] failed to submit transfer " ?

2. why not providing depth data?
[kinect2_bridge] depth processing: ~nanHz (nanms) publishing rate: ~0Hz

3. Why segmentation fault after producing one message to the topics?

wiedemeyer, 2015-09-29:
**First of all: Could you please stop spamming other issues by copy & pasting this message to them?**

1. `[TransferPool::submit] failed to submit transfer`: You are probably linking to a wrong libusb. Jetson CPU may be to slow, therefore skipping packages
2. wrong libusb
3. I don`t know. Run in gdb and check what fails.

vishu2287, 2015-09-29:
Hello wiedemeyer. Thank you for the reply.
I felt the issues related to segmentation fault, so posted in them. Now removed. Thanks for the advise.

1 &2 , If the issue is due to wrong libusb, ./Protonect is working well and I can see rgb, ir and depth images clearly without any error and distraction. 

when I "rosrun kinect2_bridge kinect2_bridge", its performing well till any topic is subscribed for messages and  leads to segmentation fault.

I will post the gdb soon

wiedemeyer, 2015-09-29:
protonect is linked to the right libusb, but kinect2_bridge is not. You should first make sure that the right libusb is linked before investigating the segmentation fault.
Use `ldd <catkin_ws>/devel/lib/kinect2_bridge/kinect2_bridge | grep libusb` and check if it points to the right one. If not make sure it is installed correct and the environment variables are also correct. There are installation instructions in the README.

xlz, 2015-09-29:
OK.

I should issue an announcement that my libfreenect2 fork for Jetson TK1 is in a bug fixing period should not be used right now.

That fork is very old, and I don't know how many bugs have been discovered and resolved upstream. If you report bugs against an old version, it's very hard to debug that and support you.

I think the best course of action is first develop your applications on a PC, and then move it to Jetson TK1.

My Jetson fork contain useful patches, but it has to wait for libfreenect 0.2 because more important work has to be finished first in the upstream.

vishu2287, 2015-09-30:
hello @wiedemeyer 

Both using same libusb file but with different source. whether this could be the reason?
These are the outputs:
ubuntu@tegra-ubuntu:~/libfreenect2/examples/protonect/bin$ ldd Protonect | grep libus
	libusb-1.0.so.0 => /home/ubuntu/libfreenect2/examples/protonect/../../depends/libusb/lib/libusb-1.0.so.0 (0xb69ad000)

ubuntu@tegra-ubuntu:~/catkin_ws/devel/lib/kinect2_bridge$ ldd kinect2_bridge | grep libus
	libusb-1.0.so.0 => /lib/arm-linux-gnueabihf/libusb-1.0.so.0 (0xb4ff2000)

wiedemeyer, 2015-09-30:
They are not the same. The one from libfreenect2 has some patches that are needed. You have to `make install` libfreenect2 to a default location like `/usr/local` where your system searches for dynamically linked libraries (or you have to configure your environment variables to add some search paths.)

vishu2287, 2015-10-22:
thank you @wiedemeyer . I have installed libfreenect2 properly. its working fine.

closed 2015-10-22 by vishu2287